SICKAG/sick_scan_xd #170: TiM781S login failure

Jonny0816, 2023-05-18:
Seems the password is wrong and report error.
[image]
Running with Sick_scan driver is ok, and the using password is different.
[image]

michael1309, 2023-05-19:
I guess you are right. We will fix it during next week (or you can fix it by creating a branch based  on this ticket). See "create a branch for this issue or link a pull request" under "Development" in this ticket report window, if you want to do it by yourself.

michael1309, 2023-06-01:
Just to keep you informed: Due date for a fix is 2023-06-09 after our internal pretest.

rostest, 2023-06-06:
The login issue is fixed in branch https://github.com/SICKAG/sick_scan_xd/tree/feature/tim781s_login. Please checkout this branch and rebuild sick_scan_xd. Thanks!

closed 2023-07-27 by rostest