SICKAG/sick_scan_xd #175: How to get the r(ring) and t(time stamp) of MultiScan

Jonny0816, 2023-06-12:
We can get the xyzi information with the cloud topic, but did not find any information about r and t, please kindly explant how to get it, Thanks.

michael1309, 2023-06-13:
Thank you for your inquiry. You can calculate the elevation from the Cartesian data as follows and then map it to an integer value corresponding to the ring position index.

$$ \text{elevation} =  \text{atan}(z,\sqrt{(x^2+y^2)}$$

For debugging purposes you can also calculate the azimuth angle from the Cartesian data
via 

$$ \text{azimuth} =  \text{atan}(y,x)$$

For the timestamp of a single shot, you can refer to the generation timestamp which corresponds to the first shot of a segment. From the rotation speed and the angle difference from shot to shot you can calculate the exact generation timestamp of each shot.

michael1309, 2023-06-13:
Yes, that is correct. The information is currently not included in the pointcloud in order not to have to transfer too much data.

closed 2023-07-27 by rostest