SICKAG/sick_scan_xd #177: Trying to get encoder ticks for LMS4000

Marco0034, 2023-06-15:
Hardware-Setup: LMS4111R-13000 and Encoder (DBS60E-S4EC02500 --> (HTL/Push Pull))
Used Branch: feature/lms4xxx_encoder

When I launch the code with "sick_lms_4xxx.launch" the code works. 

When I launch the code with "sick_lms_4xxx_encoder.launch" I get the following error: 
[image]

Could you please help me with this matter. Thanks!

rostest, 2023-06-15:
Thanks for reporting. The driver still sends command "sWN DO3And4Fnc", which is not supported by LMS4xxx and is therefore deactivated in the update (branch feature/lms4xxx_encoder). Please make sure that the software is completely rebuild by removing any previous builds using `rm -rf ./build_isolated/ ./devel_isolated/ ./install_isolated/ ./build/ ./devel/ ./install/`.

To enable encoder settings, use sick_lms_4xxx.launch and overwrite parameter `encoder_mode` with value 0, 1, 2, 3 or 4 by commandline (e.g. `encoder_mode:=1`) or by launchfile (e.g. `<arg name="encoder_mode" default="1"/>`). See https://github.com/SICKAG/sick_scan_xd/blob/feature/lms4xxx_encoder/doc/encoder.md#activation-of-encoder-informations for details.

rostest, 2023-06-15:
Just to be on the safe side: If you have not done before, switch to branch feature/lms4xxx_encoder by
```
git fetch
git checkout feature/lms4xxx_encoder
```
or by
```
git clone --branch origin/feature/lms4xxx_encoder https://github.com/SICKAG/sick_scan_xd.git
```

Marco0034, 2023-06-21:
Wen I run the code, I only get the point cloud message. 
[image]

How do I get the output of the radar scan message respectively the encoder ticks?

closed 2023-09-13 by rostest