SICKAG/sick_scan_xd #178: Mulitscan ROS parameters

Navsea, 2023-06-16:
Hi,

I seem to have some issues trying to configure the settings using the sick_multiscan.launch file.
I must say that this is the first time using it, but I think I'm using it right.

I have latest version of the driver (2.9.1), actually using the master, but they are aligned.

I'm using the following command to start the driver:
`ros2 launch sick_scan sick_multiscan.launch.py`

1. I removed the lines 20-23 from sick_mulitscan.launch**.py**
    `node_arguments.append("publish_topic:=/cloud")` is an example of one of these lines.
   If I understand correctly, this will append these settings to whatever is configured in the launch file, is there any particular reason as 
    to why you would want to do this? I assume the last configuration will be kept, overwriting whatever is configured in the launch file

2. I configure the sick_multiscan.launch file according to my needs, I then use colcon to build the ROS package.
    It looks like I can change most of the parameters and the sensor seems to adhere to these.
    However, some of them don't seem to work.
    When I set **host_LFPangleRangeFilter** `"1 -87.0 +87.0 ....` nothing seems to change
    Only when I set **host_set_LFPangleRangeFilter** to True, the driver seems to lock up.
    (lock up= the driver reports: "SickScanServices::sendSpoasCmdCheckResponse(): request: "sMN Run", response: "sAN Run 1", but
    then nothing is reported anymore)
    What is the difference between the enable bit of the first setting and the host_set setting?
    
    I then proceeded by setting the **host_set_LFPangleRangeFilter** parameter back to False, rebuilt and restarted the driver.
    The driver seems to work normally again, but by checking the sensor data in the engineering tool, I can see that the sensor is 
    azimuth range is still limited.
    Also no longer get the cloud_fullframe message, I guess it probably expects to receive the full 360 degrees, and it can no longer
    construct it. But wouldnt it make more sense to still send the limited azimuth constructed pointcloud on that topic?
   
    I then set the enable bit of **host_LFPangleRangeFilter** back to zero.
    The sensor azimuth range is still limited.

3. Any changes to the ROS parameters uing the command line interface do change the ROS parameter, but don't seem to affect the 
sensor, for example:
`ros2 param set sick_scan /publish_frame_id "tractor"`
The received ROS message still uses the original **world** frame id.

So in summary:
- I suspect that the python launch file is overwriting some of the launch file settings, and I don't understand why this is done.
- Some of the launch file parameters don't seem to work.
- When limiting the azimuth range, the cloud_fullframe topic is no longer published, which seems to only limit it's functionality.
- The ROS parameters can be changed at runtime, but don't seem to change the behavior of the sensor.

rostest, 2023-06-19:
Thanks for reporting this issue. The parameters can be set in the launchfile `sick_multiscan.launch`. They can optionally be overwritten in `sick_multiscan.launch.py` (ROS-2 only) or by commandline.

The additional settings in sick_multiscan.launch.py (like "publish_topic:=/cloud") are just default values. Please feel free to customize sick_multiscan.launch.py, i.e. to change, overwrite, add or to remove settings.

> I configure the sick_multiscan.launch file according to my needs, I then use colcon to build the ROS package.

This is perfect and correct, it is the intended way to use the launchfile and to customize settings.

I think you are right according parameter host_set_LFPangleRangeFilter resp. host_LFPangleRangeFilter. If the LFPangleRangeFilter is enabled, the limited angle range must be taken into account when generating the fullframe pointcloud. We will analyse this and provide a fix. 

Apart from settings by sick_scan_xd launchfiles, the multiscan lidar can also be configured by SOPAS Air (i.e. the web GUI provided by multiscan). Therefore the parameter host_set_LFPangleRangeFilter (i.e. set LFPangleRangeFilter true or false) and the 1. argument in host_LFPangleRangeFilter (i.e. filter enabled or disabled) have slightly different effects:
* If host_set_LFPangleRangeFilter is False, the driver will not send any LFPangleRangeFilter commands to the lidar. In this case the settings stored in the lidars eeprom will be used.
* If host_set_LFPangleRangeFilter is True, the driver will set the LFPangleRangeFilter (enable or disable) and therefore overwrite previous lidar settings.

The same logic applies to FREchoFilter and LFPlayerFilter:
* If `host_set_LFP<filtername>` is False, the settings in the lidar eeprom will apply.
* If `host_set_LFP<filtername>` is otherwise True, the settings given by parameters `host_LFP<filtername>` will apply.

Dynamical configuration updates using `ros2 param set sick_scan ...` are currently not supported by the driver. Please use the launchfile for configuration and customization.

rostest, 2023-06-20:
The angle range filter has been adapted. The fullframe pointcloud is published in accordance with angle range filter settings.
Please checkout the driver update in branch [feature/multiscan_anglerangefilter](https://github.com/SICKAG/sick_scan_xd/tree/feature/multiscan_anglerangefilter) and rebuild sick_scan_xd.
Note that angle range filter settings require a firmware 1.2.2 or higher.

closed 2023-07-27 by rostest