SICKAG/sick_scan_xd #179: Cannot Compile library for Windows

adrianfernandeziturcemi, 2023-06-19:
Hello, i am trying to compile this library in order to communicate with a LMS511-22100. Following the commands stated in the tutorial for Windows, displays an error in the step of : 
cmake --build . --clean-first --config Debug
The error is  fatal  error C1017: error c1017 invalid integer constant expression. An it is displayed for every .cpp in all the proyects of the library. 
I followed all the commands as it is shown in the documentation. Any idea how to solve this problem?

rostest, 2023-06-19:
Thank you for reporting this. Unfortunately I cannot reproduce the error, `cmake --build . --clean-first --config Debug` works as expected. Did you build the msgpack library using Visual Studio 2019 first? Can you provide a complete logfile or screenshots of your build?

rostest, 2023-06-19:
Thanks for your quick response. Please open a new console (cmd.exe) in your checkout-folder (i.e. the folder containing the subdirectories msgpack11 and sick_scan_xd) and run the following commands:
```
set _os=x64
set _cmake_string=Visual Studio 17 2022
set _msvc=Visual Studio 2022

mkdir sick_scan_xd\build\msgpack11
pushd sick_scan_xd\build\msgpack11
cmake -DMSGPACK11_BUILD_TESTS=0 -G "%_cmake_string%" ../../../msgpack11
cmake --build . --clean-first --config Debug
cmake --build . --clean-first --config Release
popd

cd sick_scan_xd
pushd build
cmake -DROS_VERSION=0 -G "%_cmake_string%" ..
cmake --build . --clean-first --config Debug
cmake --build . --clean-first --config Release
popd
```

Then just select and copy all console output into a logfile and attach it to this issue.

adrianfernandeziturcemi, 2023-06-23:
Hello, sorry for the delay, i have been out of the office this days working on another thing. I attach the log files.
[logfile.txt](https://github.com/SICKAG/sick_scan_xd/files/11847800/logfile.txt)

rostest, 2023-06-23:
Thank you for the logfile. I do not see any error C1017 (invalid integer constant expression) in this logfile. The build looks good to me, the library and executable should have been created in the build folder (`build/debug` resp. `build/release`):

```
Creando biblioteca C:/Users/adrian.fernandez/Desktop/ProyectosVision/SICK_LIDAR/instalacion/sick_scan_ws/sick_scan_xd/build/Debug/sick_scan_xd_api_test.lib y objeto C:/Users/adrian.fernandez/Desktop/ProyectosVision/SICK_LIDAR/instalacion/sick_scan_ws/sick_scan_xd/build/Debug/sick_scan_xd_api_test.exp
sick_scan_xd_api_test.vcxproj -> C:\Users\adrian.fernandez\Desktop\ProyectosVision\SICK_LIDAR\instalacion\sick_scan_ws\sick_scan_xd\build\Debug\sick_scan_xd_api_test.exe
```

Compiling on Windows currently generates some warnings about implicit conversions (double to float), which can be safely ignored.

closed 2023-06-26 by adrianfernandeziturcemi