SICKAG/sick_scan_xd #181: no data from MultiScan100 lidar sensor using "sick_scan_xd_api_test.exe" application

janvanz62, 2023-06-22:
We use the lidar MRS1000 sensor from gitRepo March 2023 in an older version and with the current version for our project from from "github.com/SICKAG/sick_scan_xdhub"
•	previous and current version using launch file "sick_mrs_1xxx.lauch" are working like expected
•	D:\LiDAR\src\sick_scan_ws\sick_scan_xd\build\Debug>sick_scan_xd_api_test.exe D:\LiDAR\src\sick_scan_ws\sick_scan_xd\launch\sick_mrs_1xxx.launch sw_pll_only_publish:=False
•	We get data from the sensor, see file "mrs_1xxxx_log.txt" in github_issue.zip

We test the MultiScan100 sensor from gitRepo June 2023 for our project from from "github.com/SICKAG/sick_scan_xdhub"
Both Lidar sensors are tested with the same newest build und the correct lauch files
•	current version using launch file "sick_multiscan.lauch" is NOT working like expected
•	D:\LiDAR\src\sick_scan_ws\sick_scan_xd\build\Debug>sick_scan_xd_api_test.exe D:\LiDAR\src\sick_scan_ws\sick_scan_xd\launch\sick_multiscan.launch sw_pll_only_publish:=False
•	Initilization seems to be successfully, NO data from the sensor, see file "MultiScan_log.txt" in github_Issue.zip

We can see visualized scan data from the Lidar sensor MRS1000 and MultiScan100 using the the "SopasET" Tool
[github_Issue.zip](https://github.com/SICKAG/sick_scan_xd/files/11834392/github_Issue.zip)

rostest, 2023-06-23:
Thank you for reporting this. The Multiscan lidar sends its scan data using UDP and therefore needs to know the IP address of the receiver, i.e. the IP address of your local PC. Two IP addresses must be configured for a Multiscan:
* Parameter `hostname`: The  IP address of the Multiscan lidar (e.g. `192.168.7.227`)
* Parameter `udp_receiver_ip`: The IP address of your PC running sick_scan_xd.

According to the logfile, both hostname and udp_receiver_ip are set to 192.168.7.227. Please set the parameter `udp_receiver_ip` to the IP address of your PC via launchfile (e.g. `<arg name="udp_receiver_ip" default="192.168.7.0"/>`) or via commandline  (e.g. `"udp_receiver_ip:=192.168.7.0"`). 

IP addresses are displayed by command `ipconfig /all` on Windows or `ifconfig -a` on Linux.

See https://github.com/SICKAG/sick_scan_xd/blob/master/doc/sick_scan_segment_xd.md#configuration for further information.

janvanz62, 2023-06-23:
if the correct udp ip address is used, the data transfer works. Thx for quick response

rostest, 2023-06-23:
Thanks for quick reply!

closed 2023-06-23 by rostest