SICKAG/sick_scan_xd #183: locked file in Visual Studio : sick_scan_xd\build\sick_scan.sln

bazzou-mohammed, 2023-06-26:
Hi,
It's urgent, I'm stuck
I'm currently using the LMS511-10100 scanner, on Windows and without ROS. I was able to generate executables, now I want to retrieve the data from my scanner in a CSV file. now i want to retrieve the data from my scanner in a CSV file., the problem I'm having is that when I try to load sln solutions such as sick_scan, I find that this solution is locked and I can't debug it.
My question is : will i be able to retrieve the data without using the debeug part?  

Thank you for your help.

rostest, 2023-06-26:
Thanks for your report and screenshot. Target "ALL_BUILD" is a cmake target and cannot be executed. Please select target `sick_scan_xd_api_test` (usually by right mouse click) and click `Set as startup project` or `Debug -> New Instance` in the menu to run sick_scan_xd_api_test.

bazzou-mohammed, 2023-06-26:
Hi;
thank you for the feedback, this is what I get when I launch the run. 
[image]

rostest, 2023-06-26:
Please provide a launchfile like `sick_lms_5xx.launch` by your commandline arguments. Select `sick_scan_xd_api_test`, open the `Properties` menu, select `Debugging` under the `Configuration Properties`, goto `Command Arguments` and insert the arguments e.g. `./launch/sick_lms_5xx.launch hostname:=192.168.0.1` (replace 192.168.0.1 by the IP address of your scanner, if different).

See https://github.com/SICKAG/sick_scan_xd/blob/master/doc/sick_scan_api/sick_scan_api.md for further informations about the API usage.

bazzou-mohammed, 2023-06-27:
good morning, 
Thank you very much for your feedback. Here is what I get :
[image]
Thank you for your help.

rostest, 2023-06-27:
The relative path of a launchfile like `./launch/sick_lms_5xx.launch` depends on your current working directory, i.e. the directory the executable is started from. Please use a path to sick_lms_5xx.launch relative to your working directory, or use an absolute path to the launchfile like `"/_SICK_DEV/sick_scan_ws_1/sick_scan_xd/launch/sick_lms_5xx.launch" hostname:=169.254.74.183`

Before starting the debugger, try to run sick_scan_xd_api_test.exe in a command shell (cmd.exe), e.g.:
```
cd D:\_SICK_DEV\sick_scan_ws_1\sick_scan_xd\build
.\Debug\sick_scan_xd_api_test ../launch/sick_lms_5xx.launch hostname:=169.254.74.183
```
Adapt the paths of the command to your system if required. If sick_scan_xd_api_test works, you can run the same in the debugger by providing the same commandline arguments (just without .`\Debug\sick_scan_xd_api_test`).

bazzou-mohammed, 2023-06-27:
Hello again, 
Thank you so much, it's great. Now I would like to have the CSV file corresponding to my scanner. Could I have some leads. Thank you.
Here is what I get :
[image]

rostest, 2023-06-27:
Great to see that sick_scan_xd_api_test works. You can export the scan data to a csv-file within the callback (i.e. function `apiTestCartesianPointCloudMsgCallback`). File https://github.com/SICKAG/sick_scan_xd/blob/master/test/src/sick_scan_xd_api/sick_scan_xd_api_test.cpp provides an example (function plotPointcloudToJpeg), which converts the scan data to a jpg file. You can e.g. modify function plotPointcloudToJpeg to export the scan data to a csv file, see https://github.com/SICKAG/sick_scan_xd/blob/68259608b11820367bf1bc5b8e0e0c412928a5a0/test/src/sick_scan_xd_api/sick_scan_xd_api_test.cpp#L34C10-L34C10.

See also https://github.com/SICKAG/sick_scan_xd/blob/master/doc/sick_scan_api/sick_scan_api.md for further information about the API examples, their callbacks and the point cloud memory layout.

bazzou-mohammed, 2023-06-27:
Thank you very much, that was my idea too.

closed 2023-06-27 by bazzou-mohammed