SICKAG/sick_scan_xd #185: Continuous point cloud retrieval

bazzou-mohammed, 2023-06-29:
Hi,
Working with the LMS511-10100 scanner, I managed to get the "sick_scan_xd_api_test" tool to work successfully. I was able to retrieve a CSV file containing the data, but I noticed that it only contains data from the first frame, i.e. a single measurement of the scanned object. At the moment, I want to perform continuous acquisitions and I wonder if anyone has an idea for solving this problem.
Thank you for your help.

rostest, 2023-06-30:
Every time the callback `apiTestCartesianPointCloudMsgCallback` in "sick_scan_xd_api_test" is executed, you will receive a message with new scan data. You can see this e.g. in your screenshot https://user-images.githubusercontent.com/103991360/249077637-049f184f-93b0-4837-8a11-dc575ab45e26.png : Each message provides a new scan. If you convert each message into csv, you should continously retrieve the scans from your lidar. Remember to append the data, default file operation might be to overwrite.

bazzou-mohammed, 2023-07-04:
Hi,
you mean I have to put the coordinate printing part here inside : 
[image]
Thank you for your help.

rostest, 2023-07-05:
Hi,
you are right. Use callback `apiTestCartesianPointCloudMsgCallback` to convert and append the scan data to a csv file. You can replace or modify function `plotPointcloudToJpeg` to plot the scan data to csv instead of jpg.

bazzou-mohammed, 2023-07-05:
Hi, 
good morning; 
Thank you for your feedback. I have understood and carried what you advised, but unfortunately I can't see any difference. 
here are the changes made : 
[printPointcloudToCSV.txt](https://github.com/SICKAG/sick_scan_xd/files/11955707/printPointcloudToCSV.txt)
here the call to the printPointcloudToCSV function :
[image]
Thank you for your help.

rostest, 2023-07-05:
Hi,

line 24 opens the csv-file in write modus, i.e. the csv-file will be cleared and overwritten for each call:
```
FILE* foutCsv = fopen(csvFileNameTmp.c_str(), "w"); // open csv file
```
Open the csv-file in append mode instead:
```
FILE* foutCsv = fopen(csvFileNameTmp.c_str(), "a"); // open csv file to append scan data
```
Alternatively, you can create a csv-file with a different filename for each scan.

bazzou-mohammed, 2023-07-05:
Hi, 
that's great, thank you very much.
have a nice day

closed 2023-07-05 by rostest