SICKAG/sick_scan_xd #186: Information about LaunchFiles

adrianfernandeziturcemi, 2023-06-30:
Hello, i am searching for more information about the launch files. I am using a LMS511-22100 in particular. Some of my questions are:

1.  Are the parameters specific to each series of sensors, if so, is there a doc that lists all the available configuration options?
2.  Is there a tool in this library to generate a launch file by connecting to an already configured scanner? 
3.  Is it neccesary to pass a launch file as an argument to connect to the scanner?

Thanks for your time, i am new to this library and i just managed to set it up and running.

rostest, 2023-06-30:
Thank you for your message. Each scanner family requires its specific configuration which is given by a launch file, e.g. sick_lms_5xx.launch for a LMS511. You need to pass this launchfile to the driver from the command line.

The launchfile contains basic default settings. Normally you do not need to modify these settings. If you changed the IP address of the scanner, the parameter `hostname` must be set, e.g. `hostname:=192.168.0.123` on the command line or `<arg name="hostname" default="192.168.0.123"/>` in the launch file. Configuration of parameters `cloud_topic` (the ROS topic for published point clouds) and `frame_id` (the ROS frame for published point clouds) can be useful.

Currently there is no tool to generate resp. modify a launch file. See the comments in the file for more information and the operation manual (telegram listing) for details. Please do not hesitate to ask, if you are in doubt or need to configure specific properties.

rostest, 2023-07-27:
Closed due to inactivity.

closed 2023-07-27 by rostest