SICKAG/sick_scan_xd #191: Error and warning messages when killing a launch node

issambm, 2023-07-08:
I run the `sick_scan_xd` driver to record bag files acquired from a NAV310 LiDAR scanner, using a BeagleBone Black as follows:

`roslaunch sick_scan sick_nav_31x.launch hostname:=<ip-address> scan_cfg_list_entry:=5`

I use `CTRL+C` to kill the launching node and stop the scanner. The scanner does actually stop but after receiving the following error and warning messages.

```
[ WARN] [1688774560.421569083]: ## WARNING in SickScanCommon::loopOnce(): 26 byte message ignored (\x02\x02\x02\x02\x00\x00\x00\x11sEA LMDscandata \x00\x3d)
[ WARN] [1688774561.428274645]: Timeout during waiting for new datagram
[ERROR] [1688774561.429010491]: Read Error when getting datagram: 1
[ WARN] [1688774561.546648360]: Disconnecting TCP-Connection.
[ERROR] [1688774561.550108892]: Tcp::readInputData: Read 0 bytes, connection is lost!
sick_generic_laser: leaving main loop...[ INFO] [1688774561.551902280]: SickThread TcpRecvThread finished (flags: threadShouldRun=0, endThread=0).
sick_scan driver closed.
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

I was wondering if there is anything wrong with the driver.

Thanks.

rostest, 2023-07-10:
Thank you for reporting this. After a shutdown (Ctrl-C or kill) you can safely ignore these warnings. The messages are general notifications after a tcp disconnection. We will make the message more descriptive for an intended disconnection during driver shutdown.

issambm, 2023-07-10:
Thank you for the answer.

Before closing this thread, may I ask—on a separate matter— how to update the `sick_scan_xd` driver of an old existing build to a new release? Or, should one rebuild from scratch?

Thanks again.

rostest, 2023-07-11:
To update the driver from an existing build, run
```
pushd src/sick_scan_xd
git pull
popd
```
To build otherwise from scratch, clone sick_scan_xd by
```
mkdir ./src
pushd ./src
git clone https://github.com/SICKAG/sick_scan_xd.git
popd
```

If not done before, checkout libsick_ldmrs and msgpack11 in the src-folder:
```
pushd ./src
git clone https://github.com/SICKAG/libsick_ldmrs.git
git clone https://github.com/SICKAG/msgpack11.git
popd
```

Then follow the instructions in https://github.com/SICKAG/sick_scan_xd/blob/master/INSTALL-ROS1.md#build-on-linux-ros1 and rebuild sick_scan_xd:
```
# remove any files from a previous build
rm -rf ./build ./build_isolated/ ./devel ./devel_isolated/ ./install ./install_isolated/ ./log/
# build msgpack11
mkdir -p ./build/msgpack11
pushd ./build/msgpack11
cmake -G "Unix Makefiles" -D CMAKE_CXX_FLAGS=-fPIC -D CMAKE_BUILD_TYPE=Release -D MSGPACK11_BUILD_TESTS=0 ../../src/msgpack11
make
sudo make install
popd
# build sick_scan_xd
source /opt/ros/noetic/setup.bash # replace noetic by your ros distro
catkin_make_isolated --install --cmake-args -DROS_VERSION=1 -Wno-dev
source ./devel_isolated/setup.bash
```

closed 2023-07-11 by issambm