SICKAG/sick_scan_xd #192: Unable to change Scan Frequency of Sick Tim 5xx

vignesh-katomaran, 2023-07-10:
I am running on Ubuntu 22.04 and ROS 2 Humble, When i changed the scan_freq parameter from launch file it applied to package. After launch i verified it with Dynamic Reconfigure(RQT reconfigure). Even though it shows the changed value in rqt_reconfigure when i tested the hz of the topic it still coming on 14-15 Hz rate

closed 2023-07-11 by rostest

vignesh-katomaran, 2023-07-11:
When i set scan_freq parameter on launchfile and restarted the node i'm getting the required Frequency at initial but later it gets back to 15hz. It is increasing step step by step for example. i set scan_freq=7.0 it starts from 7.01 then it increased to 15Hz in very soon.

vignesh-katomaran, 2023-07-11:
ERROR] [1689061227.699803686] [sick_scan]: ang_res and scan_freq have to be set, only one param is set skiping scan_fre/ang_res config

rostest, 2023-07-12:
Parameter "scan_freq" and "ang_res" are not supported for the TiM 5xx familie. TiM 5xx lidars have a constant scan frequency and angular resolution. See operation manual and chapter 4.2 in the [telegram listing](https://cdn.sick.com/media/docs/7/27/927/telegram_listing_telegram_listing_ranging_sensors_lms1xx_lms5xx_tim2xx_tim5xx_tim7xx_lms1000_mrs1000_mrs6000_nav310_ld_oem15xx_ld_lrs36xx_lms4000_lrs4000_multiscan100_en_im0045927.pdf) for details.