SICKAG/sick_scan_xd #194: API for 2 scanners

bazzou-mohammed, 2023-07-11:
Hi,
I am currently working on a LIDAR detection project and using a Sick LMS511-10100. By following a tutorial on Windows without ROS, I successfully retrieved data from my scanner. Now, I would like to incorporate another Sick LMS511-10100 and integrate them both into a single API. Is it possible to achieve this? If so, could you please share any tips or tricks you have used to accomplish this?
Thank you for your help.

rostest, 2023-07-12:
Thank you for your feedback. Currently the API does not support concurrent processing of multiple lidars. You may run each lidar in a separate process using the sick_scan_xd API and interprocess communication. See https://github.com/SICKAG/sick_scan_xd/issues/169 for further information.

bazzou-mohammed, 2023-07-12:
Hi,
Thank you for the feedback, the problem we're having today is that we can't use the ROS-1 and ROS-2 processes because they don't suit our environment. do you have another issue.
Thank you for your help.

rostest, 2023-07-14:
Thank you for your feedback. Processing 2 or more lidars concurrently with multiple calls of SickScanApiInit-functions requires threadsafe libraries. The sick_scan_xd library is currently not guaranteed to be fully thread-safe. Processing 2 lidars this way can cause unexpected results incl. core dumps.

Currently you can run 2 lidars in 2 different processes, where each process runs one lidar using the SickScan-API functions. In this case, the integration of both processes in a single application requires interprocess communication and data exchange.

closed 2023-07-26 by bazzou-mohammed

gigades, 2023-08-20:
@rostest  Just curious. I saw that SickScanApiInit() return a handle that have to be passed to other functions. It seems like you have designed it to support multiple scanners in the first place. What made you change your mind?
I have the same requirements and using inter-process communication to communicate with multiple scanners make the development unnecessarily complicate.