SICKAG/sick_scan_xd #195: Error by bulding on Linux ROS2 (Step 3)

Petopp, 2023-07-13:
Hello,
by buldings ROS2 with Linux I become a error message by step **3. Build sick_generic_caller**

In the step "colcon build --packages-select sick_scan --cmake-args " -DROS_VERSION=2" --event-handlers console_direct+"

came this message:
```
In file included from /home/petop/sick_scan_ws/src/sick_scan_xd/driver/src/ldmrs/sick_ldmrs_driver.cpp:42:
/home/petop/sick_scan_ws/src/sick_scan_xd/include/sick_scan/ldmrs/sick_ldmrs_driver.hpp:64:10: fatal error: diagnostic_updater/diagnostic_updater.hpp: Datei oder Verzeichnis nicht gefunden
   64 | #include <diagnostic_updater/diagnostic_updater.hpp> // part of diagnostic_msgs of ROS2, not available on ROS2-Windows until foxy patch 4
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/sick_scan_shared_lib.dir/build.make:664: CMakeFiles/sick_scan_shared_lib.dir/driver/src/ldmrs/sick_ldmrs_driver.cpp.o] Fehler 1
gmake[1]: *** [CMakeFiles/Makefile2:659: CMakeFiles/sick_scan_shared_lib.dir/all] Fehler 2
gmake: *** [Makefile:146: all] Fehler 2
--- stderr: sick_scan                                         
CMake Warning at CMakeLists.txt:215 (find_package):
  By not providing "Finddiagnostic_updater.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "diagnostic_updater", but CMake did not find one.

  Could not find a package configuration file provided by
  "diagnostic_updater" with any of the following names:

    diagnostic_updaterConfig.cmake
    diagnostic_updater-config.cmake

  Add the installation prefix of "diagnostic_updater" to CMAKE_PREFIX_PATH or
  set "diagnostic_updater_DIR" to a directory containing one of the above
  files.  If "diagnostic_updater" provides a separate development package or
  SDK, be sure it has been installed.

In file included from /home/petop/sick_scan_ws/src/sick_scan_xd/driver/src/ldmrs/sick_ldmrs_driver.cpp:42:
/home/petop/sick_scan_ws/src/sick_scan_xd/include/sick_scan/ldmrs/sick_ldmrs_driver.hpp:64:10: fatal error: diagnostic_updater/diagnostic_updater.hpp: Datei oder Verzeichnis nicht gefunden
   64 | #include <diagnostic_updater/diagnostic_updater.hpp> // part of diagnostic_msgs of ROS2, not available on ROS2-Windows until foxy patch 4
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/sick_scan_shared_lib.dir/build.make:664: CMakeFiles/sick_scan_shared_lib.dir/driver/src/ldmrs/sick_ldmrs_driver.cpp.o] Fehler 1
gmake[1]: *** [CMakeFiles/Makefile2:659: CMakeFiles/sick_scan_shared_lib.dir/all] Fehler 2
gmake: *** [Makefile:146: all] Fehler 2
---
Failed   <<< sick_scan [2min 39s, exited with code 2]

Summary: 0 packages finished [2min 39s]
  1 package failed: sick_scan
  1 package had stderr output: sick_scan

```

Have any one a idea to fix this? I am usning Ubuntu 22.04 with ROS2 Humble

Thanks a lot for any help

rostest, 2023-07-14:
Thank you for reporting this. Depending on your ROS installation, package diagnostic_updater can be installed separately. See https://github.com/ros/diagnostics/tree/ros2 for details.
Alternatively, if you do not need to support LDMRS lidars, you can build sick_scan_xd without LDMRS support using option `-DLDMRS=0`:
```
colcon build --packages-select sick_scan --cmake-args " -DROS_VERSION=2" " -DLDMRS=0" --event-handlers console_direct+
```

Petopp, 2023-07-19:
Thanks, that help

closed 2023-07-19 by Petopp