SICKAG/sick_scan_xd #196: field monitoring for mrs1104c

SuhaibAlHindi, 2023-07-17:
In the launch file, its not clear to how to enable field evaluation? 
if you can provide an example for the mrs1104c would be nice

rostest, 2023-07-17:
Currently field evaluation and fields setup can be done with SOPAS ET, but not by launchfile configuration. Please use SOPAS ET for the configuration of field evaluation. 
See https://github.com/SICKAG/sick_scan_xd/blob/master/doc/field_monitoring_extensions.md for ros messages and services. 
The generic API provides field information messages by functions SickScanApiRegisterLFErecMsg and SickScanApiRegisterVisualizationMarkerMsg.

SuhaibAlHindi, 2023-07-17:
I have set the field by SOPAS ET, How can I get the values for the field information without ROS?

SuhaibAlHindi, 2023-07-17:
Can I connect directly to the LiDAR through TCP or UDP and send telegram commands manually without SOPAS or ROS?

rostest, 2023-07-19:
Sorry for the late reply. Field evaluation is currently supported for LMS1xx, LMS5xx, TiM7xx and TiM7xxS, but not for MRS1104. The MRS1104 probably uses different telegrams for the field information. Support in a future release will be provided in consultation with SICK.