SICKAG/sick_scan_xd #197: change of reference point

bazzou-mohammed, 2023-07-26:
Hello,
Reminder: I'm using two identical LMS511-10100 scanners under Windows, without ROS. By running two separate processes, I was able to retrieve the data from my scanners in a CSV file and view it in Excel, as shown in the attachment : 
[image]
The two profiles are currently displayed in two different repositories. Now, the idea I'm working on is to merge these two displays into one, which will be in the middle of the two repositories.
Is it possible to make a modification to sick_scan_xd_test, specifically in the polar or Cartesian coordinates section, in order to implement this transformation?
thank you for your help.

rostest, 2023-07-27:
You can transform the points of a cartesian pointcloud to a reference coordinate system by applying an additional transform. This transform can be configured in the launchfile by parameter "add_transform_xyz_rpy", see https://github.com/SICKAG/sick_scan_xd/blob/master/doc/coordinate_transforms.md for details.

The red pointcloud in your example is rotated against the reference system (i.e. the x- and y-axis) by round about 10 deg counterclockwise and moved by ca. -0.5 m along the x-axis. Thus a configuration
```
<param name="add_transform_xyz_rpy" type="string" value="-0.5,0,0,0,0,0.1745" /> <!-- transform red pointcloud to reference coordinates: x = -0.5 m, y = 0, z = 0, roll = 0, pitch = 0, yaw = 0.1745 rad = 10 deg -->
```
should transform the red points to reference coordinates.

The blue pointcloud is rotated by ca. -20 deg (clockwise) and moved by ca. +0.3 m along the x-axis. Thus the configuration
```
<param name="add_transform_xyz_rpy" type="string" value="0.3,0,0,0,0,-0.3490" /> <!-- transform red pointcloud to reference coordinates: x = 0.3 m, y = 0, z = 0, roll = 0, pitch = 0, yaw = -0.3490 rad = -20 deg -->
```
should transform the blue points to reference coordinates.

The correct values (x, y and yaw) depend on the lidar mounting. The example values are just a guess from the screenshot and need to be adapted to your setup.

bazzou-mohammed, 2023-07-31:
Hello,
@rostest Wonderful. Thank you for your reply.

bazzou-mohammed, 2023-07-31:
hi,
I'd like to ask a quick question about synchronizing two scanners with a computer. During a test, I noticed a time difference between the time recorded in the data messages and the time on the computer I'm working on. My question is as follows: Would it be possible to synchronize the time of the pointscloud messages with the computer time by running a synchronization command?
thank you for your help.

michael1309, 2023-07-31:
The so-called software PLL (see https://github.com/SICKAG/sick_scan_xd/blob/master/doc/software_pll.md ) is used to compare the so-called ticks of the lidar for the scan start against the PC system time. The lidars run asynchronously against each other, so that the timestamp at which a certain angle is reached differs between the lidars. For very fast movements, it is therefore advisable to determine a separate time stamp for each shot (angular direction) by using the start timestamp of the scan and then taking the rotation speed of the lidar into account.

Roughly:
```
Time stamp for a specific angle = timestamp at scan start + (angle difference to start angle)/360 [deg] * 1.0/scanning frequency
```

bazzou-mohammed, 2023-08-01:
good morning,
Thank you for your reply.

closed 2023-08-01 by bazzou-mohammed