SICKAG/sick_scan_xd #199: LMS4xxx Timeout Millisec Startup

manuelrucci7, 2023-07-31:
Hi, I got an issue in using the LMS4xxx in software  mode.
To be precise I can switch on and off the laser using 

```
ros2 service call /scan3/ColaMsg sick_scan/srv/ColaMsgSrv "{request: 'sMN LMCstartmeas'}"
ros2 service call /scan3/ColaMsg sick_scan/srv/ColaMsgSrv "{request: 'sMN LMCstandby'}"
```
However after a certain amount of time the laser ros package (sick_scan_xd) stop working because it enters in a timeout.
The timeout is the **read_timeout_millisec_startup**. 
The issue appears  after having started the laser, then send it to standby mode. From this moment the counter  read_timeout_millisec_startup start counting and the laser ros driver stop.

The software says that there is a timeout 
```
[WARN] [1690808520.011202899] [sick_scan]: Timeout during waiting for new datagram
[ERROR] [1690808520.011333462] [sick_scan]: Read Error when getting datagram: 1
```
that I think is due to the fact that the laser does not send any data because it is in standby mode.

Is there a way to bypass the "read_timeout_millisec_startup" timeout?

I was also thinking to send to the laser a standby message every 10s to keep it alive. Do u know if there is better approach?

I am using the latest master branch.

michael1309, 2023-07-31:
You can set the timeout to a very large value in the launch file. See the attached screenshot for more information.

[image]

manuelrucci7, 2023-08-02:
Ok perfect

closed 2023-08-02 by manuelrucci7