SICKAG/sick_scan_xd #201: LMS511

DanyloTymets, 2023-08-16:
Hello,
Recently I've started to work on the project, which includes usage of scan data received from LMS511.  As a first step, I was using SOPAS ET  to gain .log file with the data. Now I'm trying to implement communication and data retrieving from the LiDAR sensor into my program, using sick_scan_xd.I've succeeded to continuously write data into the CSV file, by modifying function plotPointcloudToJpeg in sick_scan_xd_api_test.cpp. But during tests with sick_scan_xd_api_test.exe, I had several questions regarding principles of work.
1) Setting configuration in the LMS511 launch file. First of all, as I find out by reading existing issues, not all parameterization may be set up in launch file and some of them have to be configured in SOPAS ET. On the other hand, it's not possible to generate launch files using SOPAS ET.  What is more, after several rounds of using sick_scan_xd_api_test I discovered that RSSI was reset to "enable", although I haven't found any mentioned about that parameter in the launch file. My question is, is there a way  to read all parameters from scanner, such as start and stop angles, and avoid their resetting by using a launch file?
2) Is it possible to gain a log file with data, similar to one created by SOPAS ET? If so, is there any information about how to do it?

rostest, 2023-08-16:
Thanks for reporting these questions and issues. It is correct, that not all of the parameters supported in SOPAS ET can be configured by the sick_scan_xd launchfile. We recommend to configure the lidar using SOPAS ET and to store the configuration in its EEPROM. Currently it is not possible to create and export a sick_scan_xd launchfile in SOPAS ET.

You can activate resp. deactivate intensity values in the launchfile by parameter "intensity". Set parameter "intensity" to "false" to disable them:
```
<param name="intensity" type="bool" value="false"/>
<param name="intensity_resolution_16bit" type="bool" value="false"/>
```

Start and stop angles are configured in radians by parameters "min_ang" and "max_ang" the launchfile:
```
<param name="min_ang" type="double" value="-1.658062789" />
<param name="max_ang" type="double" value="1.658062789" />
```
Note that the validity of start and stop angle configuration depends on the lidar. You might e.g. determine suitable start and stop angles with SOPAS ET and use these values in the launchfile. Currently there is no way to set this automatically.

The sick_scan_xd driver provides the scan data and basic settings. Further data processing incl. export and logging has to be done in the application.

DanyloTymets, 2023-08-18:
Thank you for the reply and suggested solution. 
I will try to  disable transport of intensity values.
Regarding validity of start and stop angle, I find out, that for LMS511 the conversion formula from degrees to radians is next:  angle° / 180 * Pi - Pi/2.

rostest, 2023-09-13:
Thanks for your feedback. I'm closing this issue for now. Please do not hesitate to reopen the ticket in case of further issues.

closed 2023-09-13 by rostest