SICKAG/sick_scan_xd #202: Sick_scan docker build fails

ninoweg, 2023-08-17:
I'm having issues building sick_scan inside a docker container. I use the following Dockerfile:

```
FROM ros::noetic-ros-base

RUN apt-get update \
    && apt-get install -y build-essential cmake git \
    && rm -rf /var/lib/apt/lists/{apt,dpkg,cache,log} /tmp/* /var/tmp/*

WORKDIR /sick_scan_ws

RUN /bin/bash -c \
    "mkdir src && \
    cd src && \
    git clone https://github.com/SICKAG/sick_scan_xd.git && \
    cd .. && \
    source /opt/ros/noetic/setup.bash && \ 
    rosdep install --from-paths src --ignore-src -r -y && \
    catkin_make_isolated --install --cmake-args -DROS_VERSION=1 -DLDMRS=0 -DSCANSEGMENT_XD=0 -Wno-dev && \
    source ./devel_isolated/setup.bash"

COPY ./start_ros_node.sh /
# make source script executeable
RUN /bin/bash -c \
    "chmod +x /ros_entrypoint.sh"

# setup entrypoint
ENTRYPOINT ["/ros_entrypoint.sh"]

CMD ["bash"]
```

I'm not able to build it successfully as the build process fails. See error messages: [error_msgs.txt](https://github.com/SICKAG/sick_scan_xd/files/12368516/error_msgs.txt). This happens only when building inside the docker container. I had no issues when building under Ubuntu 20.04.

Has anybody had similar issues or could share a working Dockerfile?

rostest, 2023-08-17:
Thanks for your feedback. Depending on the ROS distribution, MIN and MAX can be multiple defined. You can use sick_scan_xd  branch https://github.com/SICKAG/sick_scan_xd/tree/feature/bloom_pretest which avoids the use of MIN/MAX macros. Replace `git clone https://github.com/SICKAG/sick_scan_xd.git` in the Dockerfile by 
```
git clone -b feature/bloom_pretest https://github.com/SICKAG/sick_scan_xd.git
```
and retry.

Alternatively, you might try a noetic-desktop-full version, e.g. use
```
FROM osrf/ros:noetic-desktop-full
RUN apt-get update && apt-get install -y ros-${ROS_DISTRO}-desktop
```
in the Dockerfile.

ninoweg, 2023-08-22:
Thanks for the quick reply. Switching to branch feature/bloom_pretest worked for me.

rostest, 2023-09-13:
Thanks for your feedback!

closed 2023-09-13 by rostest