SICKAG/sick_scan_xd #203: LMS511_ros

zhshwind, 2023-08-24:
<?xml version="1.0"?>

<launch>
    <arg name="hostname" default="192.168.20.101"/>
    
    <node name="wika_b" pkg="sick_scan" type="sick_generic_caller" respawn="true" output="screen">
        

        
        
        <param name="frame_id" type="str" value="cloud_b"/>
        <param name="scanner_type" type="string" value="sick_lms_5xx"/>
        
        <param name="min_ang" type="double" value="-1.658062789"/>
        
        <param name="max_ang" type="double" value="1.658062789"/>
        <param name="range_max" type="double" value="100.0"/>
        <param name="intensity" type="bool" value="true"/>
        <param name="intensity_resolution_16bit" type="bool" value="false"/>
        <param name="hostname" type="string" value="$(arg hostname)"/>
        <param name="port" type="string" value="2112"/>
        <param name="timelimit" type="int" value="5"/>
        <param name="use_binary_protocol" type="bool" value="true"/>
        
        <param name="start_services" type="bool" value="True"/> 
        <param name="activate_lferec" type="bool" value="True"/> 
        <param name="activate_lidoutputstate" type="bool" value="True"/> 
        <param name="activate_lidinputstate" type="bool" value="True"/> 
	<param name="min_intensity" type="double" value="0.0"/> 
	<remap from="scan" to="scan_b"/>
	<remap from="cloud" to="cloud_b"/>
    </node>

[image]

rostest, 2023-08-24:
Thank you for your message. Does your LMS511 lidar work with the most recent lidar firmware and the latest sick_scan_xd release with default settings and default launchfile?

Please send a complete logfile of the startup sequence, i.e. /home/wika/.ros/*/wika*.log resp. the complete console output from the driver. The startup sequence in the logfile should look like this example:

```
SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /sick_lms_5xx/activate_lferec: True
 * /sick_lms_5xx/activate_lidinputstate: True
 * /sick_lms_5xx/activate_lidoutputstate: True
 * /sick_lms_5xx/add_transform_check_dynamic_updates: False
 * /sick_lms_5xx/add_transform_xyz_rpy: 0,0,0,0,0,0
 * /sick_lms_5xx/client_authorization_pw: F4724744
 * /sick_lms_5xx/cloud_topic: cloud
 * /sick_lms_5xx/encoder_mode: -1
 * /sick_lms_5xx/filter_echos: 2
 * /sick_lms_5xx/frame_id: cloud
 * /sick_lms_5xx/hostname: 127.0.0.1
 * /sick_lms_5xx/intensity: True
 * /sick_lms_5xx/intensity_resolution_16bit: False
 * /sick_lms_5xx/max_ang: 1.658062789
 * /sick_lms_5xx/message_monitoring_enabled: True
 * /sick_lms_5xx/min_ang: -1.658062789
 * /sick_lms_5xx/min_intensity: 0.0
 * /sick_lms_5xx/port: 2112
 * /sick_lms_5xx/range_filter_handling: 0
 * /sick_lms_5xx/range_max: 100.0
 * /sick_lms_5xx/range_min: 0.05
 * /sick_lms_5xx/read_timeout_millisec_default: 5000
 * /sick_lms_5xx/read_timeout_millisec_kill_node: 150000
 * /sick_lms_5xx/read_timeout_millisec_startup: 120000
 * /sick_lms_5xx/ros_qos: -1
 * /sick_lms_5xx/scanner_type: sick_lms_5xx
 * /sick_lms_5xx/start_services: True
 * /sick_lms_5xx/sw_pll_only_publish: True
 * /sick_lms_5xx/timelimit: 5
 * /sick_lms_5xx/use_binary_protocol: True
 * /sick_lms_5xx/use_generation_timestamp: True

NODES
  /
    sick_lms_5xx (sick_scan/sick_generic_caller)

process[sick_lms_5xx-1]: started with pid [12049]
[ INFO] [1692858706.333820664]: sick_generic_caller V. 2.9.1
[ INFO] [1692858706.345408662]: Range filter configuration: range_min=0.05, range_max=100, range_filter_handling=0
[ INFO] [1692858706.345911626]: Found sopas_protocol_type param overwriting default protocol:
[ INFO] [1692858706.345935714]: Binary protocol activated
[ INFO] [1692858706.346596980]: Start initialising scanner [Ip: 192.168.0.1] [Port:2112]
[ INFO] [1692858706.365800643]: Publishing on topic "/sick_lms_5xx/lferec", qos=10
[ INFO] [1692858706.366290187]: Publishing on topic "/sick_lms_5xx/lidoutputstate", qos=10
[ INFO] [1692858706.366637419]: Publishing on topic "/sick_lms_5xx/marker", qos=10
[ INFO] [1692858706.369736604]: SickCloudTransform: add_transform_xyz_rpy = (0,0,0,0,0,0)
[ INFO] [1692858706.369779485]: SickCloudTransform: azimuth_offset = 0 [deg]
[ INFO] [1692858706.369798139]: SickCloudTransform: additional 3x3 rotation matrix = { (1,0,0), (0,1,0), (0,0,1) }
[ INFO] [1692858706.369820581]: SickCloudTransform: apply 3x3 rotation = false
[ INFO] [1692858706.369843970]: SickCloudTransform: additional translation = (0,0,0)
[ INFO] [1692858706.369859431]: SickCloudTransform: check_dynamic_updates = false
[ INFO] [1692858706.369884011]: Publishing lidar pointcloud2 to cloud
[ INFO] [1692858706.370151050]: Publishing on topic "/cloud", qos=10
[ INFO] [1692858706.371800040]: Publishing on topic "/sick_lms_5xx/imu", qos=10
[ INFO] [1692858706.372970094]: Publishing on topic "/sick_lms_5xx/encoder", qos=10
[ INFO] [1692858706.373463655]: Publishing on topic "/sick_lms_5xx/scan", qos=10
[ INFO] [1692858706.373993973]: SickCloudTransform: add_transform_xyz_rpy = (0,0,0,0,0,0)
[ INFO] [1692858706.374022881]: SickCloudTransform: azimuth_offset = 0 [deg]
[ INFO] [1692858706.374048557]: SickCloudTransform: additional 3x3 rotation matrix = { (1,0,0), (0,1,0), (0,0,1) }
[... 122 lines cut ...]
[ INFO] [1692858707.939406387]: Sending  : <STX><STX><STX><STX><Len=0019>sEN LIDinputstate 0x01 CRC:<0x39>
[ INFO] [1692858707.942402254]: Receiving: <STX>sEA LIDinputstate \x01<ETX>
[ INFO] [1692858707.942453813]: Sending  : <STX><STX><STX><STX><Len=0016>sMN LMCstartmeas CRC:<0x68>
[ INFO] [1692858707.943386367]: Receiving: <STX>sAN LMCstartmeas \x00<ETX>
[ INFO] [1692858707.943465791]: Sending  : <STX><STX><STX><STX><Len=0007>sMN Run CRC:<0x19>
[ INFO] [1692858707.943750096]: Receiving: <STX>sAN Run \x01<ETX>
[ INFO] [1692858707.943792056]: Sending  : <STX><STX><STX><STX><Len=0017>sEN LMDscandata 0x01 CRC:<0x33>
[ INFO] [1692858707.944356431]: Receiving: <STX>sEA LMDscandata \x01<ETX>
[ INFO] [1692858707.945336197]: SickScanServices: service "/sick_lms_5xx/ColaMsg" created ("/sick_lms_5xx/ColaMsg")
[ INFO] [1692858707.945707960]: SickScanServices: service "/sick_lms_5xx/ECRChangeArr" created ("/sick_lms_5xx/ECRChangeArr")
[ INFO] [1692858707.946024246]: SickScanServices: service "/sick_lms_5xx/LIDoutputstate" created ("/sick_lms_5xx/LIDoutputstate")
[ INFO] [1692858707.946310825]: SickScanServices: service "/sick_lms_5xx/SCdevicestate" created ("/sick_lms_5xx/SCdevicestate")
[ INFO] [1692858707.946543856]: SickScanServices: service "/sick_lms_5xx/SCreboot" created ("/sick_lms_5xx/SCreboot")
[ INFO] [1692858707.946773229]: SickScanServices: service "/sick_lms_5xx/SCsoftreset" created ("/sick_lms_5xx/SCsoftreset")
[ INFO] [1692858707.946984312]: SickScanServices: service "/sick_lms_5xx/SickScanExit" created ("/sick_lms_5xx/SickScanExit")
[ INFO] [1692858707.947007600]: SickScanServices: ros services initialized
[... cut ...]
[ INFO] [1692858710.279777979]: Software PLL locking started, mapping ticks to system time.
[ INFO] [1692858710.279811918]: 1 / 6 packets dropped. Software PLL not yet locked.
[ INFO] [1692858710.324510009]: 2 / 6 packets dropped. Software PLL not yet locked.
[ INFO] [1692858710.366901123]: 3 / 6 packets dropped. Software PLL not yet locked.
[ INFO] [1692858710.412497939]: 4 / 6 packets dropped. Software PLL not yet locked.
[ INFO] [1692858710.453000749]: 5 / 6 packets dropped. Software PLL not yet locked.
[ INFO] [1692858710.494347991]: Software PLL is ready and locked now!
```

rostest, 2023-09-14:
Closed due to inactivity.

closed 2023-09-14 by rostest