SICKAG/sick_scan_xd #204: Running the driver with verboseLevel 1 causes problems

Kaju-Bubanja, 2023-08-29:
When running the lms5xx.launch file with the argument verboseLevel set to 1 it generates more logs and data files in the /tmp folder. So far so good, but it fills up the complete harddisk in a matter of minutes. Then when I tried to reboot the system it got stuck in following condition: https://serverfault.com/questions/987488/boot-stuck-at-a-start-job-is-running-for-create-volatile-files-and-directories now I can't boot anymore and have to use a recovery stick to delete these heaps of tmp files from the driver.

It would be nice if the driver would maybe store only every x seconds or minutes a data file and/or if it stops before filling the disk completely

rostest, 2023-08-29:
Thank you for this valuable hint. Creating data dump files was meant for low level debugging only. We will remove the argument verboseLevel or restrict the total file size in a future version.

rostest, 2023-09-13:
The number and size of dumpfiles in case of verboseLevel>0 will be restricted in the next release. Thanks again!

closed 2023-09-13 by rostest