SICKAG/sick_scan_xd #206: ROS2 slam test test_005_hector.launch not working

zezami, 2023-08-31:
Hello,

Trying to use the test_005_hector.launch with sick_nav_350 in ROS 2. It seems the launch file gets stuck after launching nav350. Communication with the nav 350 is working as it should but it does not launch hector or the rviz stuff. I've tried launching it in two different ways:

ros run sick_scan sick_scan_generic test_005_hector.launch

And i've written a python wrapper so it works with:

ros launch sick_scan test_005.hector.launch

Same result:
[sick_generic_caller-1] [INFO] [1693461766.148262524] [sick_scan]: SickScanServices: service "/SickScanExit" created ("/SickScanExit")
[sick_generic_caller-1] [INFO] [1693461766.148357365] [sick_scan]: SickScanServices: ros services initialized
[sick_generic_caller-1] [INFO] [1693461766.148383347] [sick_scan]: Setup completed, sick_scan_xd is up and running. Pointcloud is published on topic "cloud"
[sick_generic_caller-1] scanner_type                   string     sick_nav_350
[sick_generic_caller-1] slam_echo                      string     laser_POS_000_DIST1
[sick_generic_caller-1] slam_bundle                    bool       true
[sick_generic_caller-1] min_ang                        double     -2.3998277
[sick_generic_caller-1] max_ang                        double     +2.3998277
[sick_generic_caller-1] intensity_resolution_16bit     bool       False
[sick_generic_caller-1] hostname                       string     192.168.0.1
[sick_generic_caller-1] port                           string     2112
[sick_generic_caller-1] timelimit                      int        5

To me it looks like the launch file is not forking the process and won't continue working through the launch file.

zezami, 2023-09-01:
I've realised that this test is for ROS 1 only. Are there any plans to add examples for ROS 2 with scan matching algoritms without the need for wheel odometry?

Tried the test again with ROS1, scanner, hector and rviz starts up but Global status, Pointcloud and TF are not happy. Complains about:
Global status > Fixed frame > Unknown frame map

Pointcloud2 > Status error > Transform > For frame [cloud]: Fixed frame [map] does not exist

TF: No transform from [xxxx] to frame[map]

Launch file looks like this:

<?xml version="1.0"?>
<launch>
    <arg name="hostname" default="192.168.0.1"/>
    <node name="sick_nav_350" pkg="sick_scan" type="sick_generic_caller" respawn="false" output="screen">
        <param name="scanner_type" type="string" value="sick_nav_350"/>
        <param name="slam_echo" type="string" value="laser_POS_000_DIST1"/>
        <param name="slam_bundle" type="bool" value="true"/>
        <param name="min_ang" type="double" value="-2.3998277"/>
        <param name="max_ang" type="double" value="+2.3998277"/>
        <param name="intensity_resolution_16bit" type="bool" value="False"/>
        <param name="hostname" type="string" value="$(arg hostname)"/>
        <param name="port" type="string" value="2112"/>
        <param name="timelimit" type="int" value="5"/>
    </node>
    
    <node pkg="tf" type="static_transform_publisher" name="laser_baselaser" args="0 0 0 0 0 0 1 base_laser laser 100"/>
    
    <node pkg="tf" type="static_transform_publisher" name="laser_layer_0_first_echo_base_laser"
          args="0 0 0 0 0 0 1 base_laser laser_POS_000_DIST1 100"/>
    <node pkg="tf" type="static_transform_publisher" name="base_laser_to_base_link"
          args="0 0 0 0 0 0 1 base_link base_laser  100"/>
    <node pkg="tf" type="static_transform_publisher" name="base_link_to_base_footprint"
          args="0 0 0 0 0 0 1 base_footprint base_link  100"/>
    <node pkg="tf" type="static_transform_publisher" name="base_footprint_to_odom"
          args="0 0 0 0 0 0 1 odom base_footprint 100"/>
    <node pkg="tf" type="static_transform_publisher" name="odom_to_scanmatcher"
          args="0 0 0 0 0 0 1 scanmatcher_frame odom 100"/>
    <arg name="tf_map_scanmatch_transform_frame_name" default="/scanmatcher_frame"/>
    <arg name="base_frame" default="base_link"/>
    <arg name="odom_frame" default="base_link"/>
    <arg name="pub_map_odom_transform" default="true"/>
    <arg name="scan_subscriber_queue_size" default="5"/>
    <arg name="scan_topic" default="scan"/>
    <arg name="map_size" default="2048"/>
    <node pkg="hector_mapping" type="hector_mapping" name="hector_mapping" output="screen">
        
        <param name="map_frame" value="map"/>
        <param name="base_frame" value="$(arg base_frame)"/>
        <param name="odom_frame" value="$(arg base_frame)"/>
        
        <param name="use_tf_scan_transformation" value="true"/>
        <param name="use_tf_pose_start_estimate" value="false"/>
        <param name="pub_map_odom_transform" value="$(arg pub_map_odom_transform)"/>
        
        <param name="map_resolution" value="0.050"/>
        <param name="map_size" value="$(arg map_size)"/>
        <param name="map_start_x" value="0.5"/>
        <param name="map_start_y" value="0.5"/>
        <param name="map_multi_res_levels" value="2"/>
        
        <param name="update_factor_free" value="0.4"/>
        <param name="update_factor_occupied" value="0.7"/>
        <param name="map_update_distance_thresh" value="0.2"/>
        <param name="map_update_angle_thresh" value="0.9"/>
        <param name="laser_z_min_value" value="-1.0"/>
        <param name="laser_z_max_value" value="1.0"/>
        
        <param name="advertise_map_service" value="true"/>
        <param name="scan_subscriber_queue_size" value="$(arg scan_subscriber_queue_size)"/>
        <param name="scan_topic" value="$(arg scan_topic)"/>
        
        
        <param name="tf_map_scanmatch_transform_frame_name" value="$(arg tf_map_scanmatch_transform_frame_name)"/>
    </node>
    
    <arg name="rviz" default="true"/>
    <arg name="rviz_file" default="$(find sick_scan)/launch/rviz/hector.rviz"/>
    <node pkg="rviz" type="rviz" name="$(anon rviz)" args="-d $(arg rviz_file)" output="screen" if="$(arg rviz)"/>
</launch>

rostest, 2023-09-01:
Thanks for your feedback. We will analyze hector_slam support for NAV-350 on ROS-1 and possibilities for a migration to ROS-2.

rostest, 2023-09-04:
Please find attached launch and rviz configuration files adapted for NAV-350 and SLAM support. Just unzip file sick_scan_xd_slam_launch.zip into folder sick_scan_xd.

**NAV-350 ROS-1 SLAM example**

Build hector_slam and sick_scan_xd:
```
cd src
git clone https://github.com/SICKAG/sick_scan_xd.git
git clone https://github.com/tu-darmstadt-ros-pkg/hector_slam.git
cd ..
catkin_make_isolated --install --cmake-args -DROS_VERSION=1 -DCMAKE_ENABLE_EMULATOR=1 -Wno-dev
```

Run rviz, sick_scan_xd and hector_slam:
```
source /opt/ros/noetic/setup.bash
source ./devel_isolated/setup.bash
rosrun rviz rviz -d ./src/sick_scan_xd/test/emulator/config/rviz_slam_nav350.rviz &
roslaunch sick_scan sick_nav_350.launch hostname:=192.168.0.1 &
roslaunch sick_scan test_200_slam_ros1_hector.launch scan_topic:=/sick_nav_350/scan &
```

The following rviz screenshot shows an example of a NAV-350 pointcloud created by sick_scan_xd and its map generated by hector_slam on ROS1:
[image]

**NAV-350 ROS-2 SLAM example**

Install the ROS2 slam-toolbox with `sudo apt install ros-foxy-navigation2 ros-foxy-nav2-bringup ros-foxy-slam-toolbox` (replace `foxy` by your ros distribution).

Build sick_scan_xd for ROS-2 as described in [INSTALL-ROS2.md](https://github.com/SICKAG/sick_scan_xd/blob/master/INSTALL-ROS2.md#build-on-linux-ros2)

Run rviz2, sick_scan_xd, slam_toolbox and static transforms:
```
source /opt/ros/foxy/setup.bash
source ./install/setup.bash
ros2 run rviz2 rviz2 -d ./src/sick_scan_xd/test/emulator/config/rviz2_slam_nav350.rviz &
ros2 launch sick_scan sick_nav_350.launch.py hostname:=192.168.0.1 &
ros2 run tf2_ros static_transform_publisher 0 0 0 0 0 0 base_link cloud  &
ros2 run tf2_ros static_transform_publisher 0 0 0 0 0 0 base_footprint base_link  &
ros2 run tf2_ros static_transform_publisher 0 0 0 0 0 0 odom base_footprint  &
ros2 launch nav2_bringup navigation_launch.py &
ros2 launch slam_toolbox online_async_launch.py &
```

Note: Laserscan messages need to be remapped to topic `/scan` (default is `/sick_nav_350/scan`). Use `remappings=[ ('/sick_nav_350/scan', '/scan'), ]` in the launchfile, e.g.:
```
node = Node(
    package='sick_scan',
    executable='sick_generic_caller',
    output='screen',
    remappings=[ ('/sick_nav_350/scan', '/scan'), ], # remap laserscan messages to topic /scan
)
```

The following rviz2 screenshot shows an example of a NAV-350 laserscan created by sick_scan_xd and its map generated by slam_toolbox on ROS2:
[image]

[sick_scan_xd_slam_launch.zip](https://github.com/SICKAG/sick_scan_xd/files/12514462/sick_scan_xd_slam_launch.zip)

zezami, 2023-09-05:
Thank you! I will give it a try. 

I see that you are using slam_toolbox for ROS2 instead of Hector slam. I thought it was not possible to use slam_toolbox without wheel odometry feedback?

rostest, 2023-09-05:
You are probably right. According to https://github.com/SteveMacenski/slam_toolbox/issues/221 and https://stackoverflow.com/questions/75864827/slam-toolbox-mapping-issue, you can use https://github.com/AlexKaravaev/ros2_laser_scan_matcher to create an odometry feedback. But I have not tested it (it is a bit beyond the scope of the driver).

Nevertheless, you can use the slam_toolbox to create an initial map. In navigational mode, the NAV-350 lidar generates pose- and landmark data, which are translated to TF messages. Using these TF messages, you can see in rviz, how the lidar moves on the map.

zezami, 2023-09-07:
I tried the ROS 1 example but I get the following when launching hector:
[ INFO] [1694066953.126533259]: lookupTransform base_link to cloud timed out. Could not transform laser scan into base_frame.

rostest, 2023-09-13:
Thanks for your update!

closed 2023-09-13 by rostest

erknl, 2024-02-22:
Can this be done only on nav350, or can it be done on lms-1xx as well?

rostest, 2024-02-22:
@erknl I think the above "NAV-350 ROS-2 SLAM example" might work with humble and lms-1xx, if you swap `foxy` with `humble` and `sick_nav_350` with `sick_lms_1xx`.