SICKAG/sick_scan_xd #208: How to obtain the status of a LiDAR (LMS511)sensor when it fails due to dust contamination?

hoopp, 2023-09-01:
Sopas: devicestatus 35 Device contaminated,warning threshold reached-Please clean front screen
Sending  : sRN SCdevicestate
Receiving: <STX>sRA SCdevicestate 0<ETX>   or <STX>sRA SCdevicestate 1<ETX>

I tried, but the returned status is not the desired dirty state. 
How to obtain the status of a LiDAR (LMS511)sensor when it fails due to dust contamination?

rostest, 2023-09-01:
Thanks for your message. You can use SOPAS command `sRA GetContaminationResult` to read the contamination status. 

Use ros service "ColaMsg" to send the SOPAS command:
```
# ROS-1: 
rosservice call /sick_lms_5xx/ColaMsg "{request: 'sMN SetAccessMode 3 F4724744'}" # switch to authorized client
rosservice call /sick_lms_5xx/ColaMsg "{request: 'sRA GetContaminationResult'}" # read contamination status
# ROS-2: 
ros2 service call /ColaMsg sick_scan/srv/ColaMsgSrv "{request: 'sMN SetAccessMode 3 F4724744'}" # switch to authorized client
ros2 service call /ColaMsg sick_scan/srv/ColaMsgSrv "{request: 'sRA GetContaminationResult'}" # read contamination status
```

An additional service to get the contamination status will be added in the next release.

doubeLin, 2023-09-06:
> Thanks for your message. You can use SOPAS command `sRA GetContaminationResult` to read the contamination status.
>
> Use ros service "ColaMsg" to send the SOPAS command:
>
> ```
> # ROS-1:
> rosservice call [...]

use SOPAS command sRA GetContaminationResult to read the contamination status, I tried several times, when SetAccessMode , the connection will be lost，and Calling the GetContinuationResult command will block.

Partial related logs：
[sick_generic_caller-1] [INFO] [1693971162.409436719] [sick_scan]: SickScanServices: Sending request "sMN SetAccessMode 3 F4724744"
[sick_generic_caller-1] [INFO] [1693971162.409504501] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
[sick_generic_caller-1] [INFO] [1693971162.411948097] [sick_scan]: Receiving: <STX>sAN SetAccessMode \x01<ETX>
[sick_generic_caller-1] [INFO] [1693971162.412006327] [sick_scan]: SickScanServices: Request "sMN SetAccessMode 3 F4724744" successfully sent, received reply "sAN SetAccessMode \x01"
[sick_generic_caller-1] [INFO] [1693971162.412017513] [sick_scan]: SickScanServices: request: "sMN SetAccessMode 3 F4724744"
[sick_generic_caller-1] [INFO] [1693971162.412024155] [sick_scan]: SickScanServices: response: "sAN SetAccessMode \x01"
[sick_generic_caller-1] [WARN] [1693971167.413414562] [sick_scan]: Timeout during waiting for new datagram
[sick_generic_caller-1] [ERROR] [1693971167.413469545] [sick_scan]: Read Error when getting datagram: 1
[sick_generic_caller-1] [ERROR] [1693971167.413487959] [sick_scan]: ## ERROR SickScanMonitor::checkState(): read timeout after 5.00067 sec, timeout (5 sec) exceeded.
[sick_generic_caller-1] [ERROR] [1693971167.413495585] [sick_scan]: ## ERROR in sick_scan_xd: restarting scanner after read timeout
[sick_generic_caller-1] [WARN] [1693971167.413501557] [sick_scan]: Disconnecting TCP-Connection.
[sick_generic_caller-1] [ERROR] [1693971167.414138047] [sick_scan]: Tcp::readInputData: Read 0 bytes, connection is lost!
[sick_generic_caller-1] [INFO] [1693971167.414186124] [sick_scan]: SickThread TcpRecvThread finished (flags: threadShouldRun=0, endThread=0).
[sick_generic_caller-1] [INFO] [1693971172.414508443] [sick_scan]: sick_scan_xd: Tcp::open: connecting to 192.168.6.200:2112 ...
[sick_generic_caller-1] [INFO] [1693971172.415268493] [sick_scan]: sick_scan_xd Tcp::open: connected to 192.168.6.200:2112
[sick_generic_caller-1] [INFO] [1693971172.415272859] [sick_scan]: SickThread TcpRecvThread started.
[sick_generic_caller-1] [INFO] [1693971172.415790217] [sick_scan]: Parameter setting for <active_echo: 0>
[sick_generic_caller-1] [INFO] [1693971172.415993831] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>
[sick_generic_caller-1] [INFO] [1693971172.418186877] [sick_scan]: Receiving: <STX>sRA SCdevicestate \x01<ETX>
[sick_generic_caller-1] [INFO] [1693971172.418596589] [sick_scan]: checkColaDialect: lidar response in configured Cola-dialect Cola-B
......
[sick_generic_caller-1] [INFO] [1693971189.562719623] [sick_scan]: SickScanServices: Sending request "sRA GetContaminationResult"
[sick_generic_caller-1] [INFO] [1693971189.562786264] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0026>sRA GetContaminationResult CRC:<0x7f>
[sick_generic_caller-1] [INFO] [1693971189.563499763] [sick_scan]: Receiving: <STX>sFA\x00\x0b<ETX>
[sick_generic_caller-1] [WARN] [1693971189.563532816] [sick_scan]: Error Sopas answer mismatch: Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0026>sRA GetContaminationResult CRC:<0x..., received answer: "sFA\x00\x0b", expected patterns: "sRA GetContaminationResult"

rostest, 2023-09-06:
Thanks for reporting this. Sorry, my fault - the SOPAS command is `sRN ContaminationResult`, not `sRA GetContaminationResult`.

Please try again with
```
rosservice call /sick_mrs_1xxx/ColaMsg "{request: 'sRN ContaminationResult'}" # ROS-1
ros2 service call /ColaMsg sick_scan/srv/ColaMsgSrv "{request: 'sRA GetContaminationResult'}" # ROS-2
```
Note that this command is only supported by MRS-1000, LMS-1000, multiScan, picoScan.

doubeLin, 2023-09-06:
Does this command (sRN ContaminationResult) not support LMS5xx? 
If not supported, how can I obtain the dirty status of the lms5xx device?

rostest, 2023-09-06:
For LMS-1xx and LMS-5xx the SOPAS command "sRN LCMstate" can be used to query the contamination status:
```
rosservice call /sick_mrs_1xxx/ColaMsg "{request: 'sRN LCMstate'}" # ROS-1
ros2 service call /ColaMsg sick_scan/srv/ColaMsgSrv "{request: 'sRN LCMstate'}" # ROS-2
```
See the telegram listing   https://cdn.sick.com/media/docs/7/27/927/telegram_listing_telegram_listing_ranging_sensors_lms1xx_lms5xx_tim2xx_tim5xx_tim7xx_lms1000_mrs1000_mrs6000_nav310_ld_oem15xx_ld_lrs36xx_lms4000_lrs4000_multiscan100_en_im0045927.pdf for details.

doubeLin, 2023-09-07:
This command works, thank you very much！

rostest, 2023-09-13:
Thanks for your friendly feedback!

closed 2023-09-13 by rostest

mustafaktaas, 2024-07-23:
Does this command (sRN LCMstate) not support sick_tim_7xx?
If not supported, how can I obtain the dirty status of the sick_tim_7xx device?

rostest, 2024-07-23:
@mustafaktaas Thanks for your feedback. Tim7xx devices do not support command `sRN LCMstate` resp. a contamination status according to the telegram listing:

[image]