SICKAG/sick_scan_xd #209: PicoScan driver issues

sjeffreydong, 2023-09-01:
[image]
When customer use our latest PicoScan driver, they met this situation. Would you please check what happened and how to solve it.

rostest, 2023-09-01:
Thank you for reporting this. I can not reproduce that error. To ensure a complete rebuild, please remove any previous build files with `rm -rf ./build ./build_isolated/ ./devel ./devel_isolated/ ./install ./install_isolated/ ./log/`. 

Follow the build instructions on https://github.com/SICKAG/sick_scan_xd/blob/master/INSTALL-ROS1.md#build-on-linux-ros1 to build sick_scan_xd:
```
# Create a workspace folder,
mkdir -p ./sick_scan_ws
cd ./sick_scan_ws
# Clone repositories
mkdir ./src
pushd ./src
git clone https://github.com/SICKAG/libsick_ldmrs.git
git clone https://github.com/SICKAG/msgpack11.git
git clone https://github.com/SICKAG/sick_scan_xd.git
popd
rm -rf ./build ./build_isolated/ ./devel ./devel_isolated/ ./install ./install_isolated/ ./log/
mkdir -p ./build/msgpack11
pushd ./build/msgpack11
cmake -G "Unix Makefiles" -D CMAKE_CXX_FLAGS=-fPIC -D CMAKE_BUILD_TYPE=Release -D MSGPACK11_BUILD_TESTS=0 ../../src/msgpack11
make
sudo make install
popd
# Build msgpack11 library 
mkdir -p ./build/msgpack11
pushd ./build/msgpack11
cmake -G "Unix Makefiles" -D CMAKE_CXX_FLAGS=-fPIC -D CMAKE_BUILD_TYPE=Release -D MSGPACK11_BUILD_TESTS=0 ../../src/msgpack11
make
sudo make install
popd
# Build sick_generic_caller
source /opt/ros/noetic/setup.bash # replace noetic by your ros distro
catkin_make_isolated --install --cmake-args -DROS_VERSION=1 -Wno-dev
source ./devel_isolated/setup.bash
```
If the problem still persists, please post a logfile of the complete build process.

rostest, 2023-09-14:
Closed due to inactivity.

closed 2023-09-14 by rostest