SICKAG/sick_scan_xd #210: ros2 service call falied

zhouerlin, 2023-09-04:
Compile Command：
colcon build --packages-select sick_scan --cmake-args " -DROS_VERSION=2" " -DLDMRS=0" " -DSCANSEGMENT_XD=0" " -DENABLE_EMULATOR=1" --event-handlers console_direct+

local@robot:~$ **ros2 service call /ColaMsg sick_scan/srv/ColaMsgSrv "{request: 'sRN SCdevicestate'}"**
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_generator_py/import_type_support_impl.py", line 46, in import_type_support
    return importlib.import_module(module_name, package=pkg_name)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 657, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 556, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1166, in create_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
ImportError: /home/local/ros2-ws/install/sick_scan/lib/libsick_scan__rosidl_generator_c.so: undefined symbol: sensor_msgs__msg__PointCloud2__copy

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/ros/foxy/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.9.10', 'console_scripts', 'ros2')()
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2cli/cli.py", line 67, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2service/command/service.py", line 41, in main
    return extension.main(args=args)
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2service/verb/call.py", line 58, in main
    return requester(
  File "/opt/ros/foxy/lib/python3.8/site-packages/ros2service/verb/call.py", line 83, in requester
    cli = node.create_client(srv_module, service_name)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/node.py", line 1248, in create_client
    check_for_type_support(srv_type)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/type_support.py", line 29, in check_for_type_support
    msg_type.__class__.__import_type_support__()
  File "/home/local/ros2-ws/install/sick_scan/lib/python3.8/site-packages/sick_scan/srv/_cola_msg_srv.py", line 254, in __import_type_support__
    module = import_type_support('sick_scan')
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_generator_py/import_type_support_impl.py", line 48, in import_type_support
    raise UnsupportedTypeSupport(pkg_name)
rosidl_generator_py.import_type_support_impl.UnsupportedTypeSupport: Could not import 'rosidl_typesupport_c' for package 'sick_scan'

rostest, 2023-09-04:
Thanks for reporting this issue. Unfortunately, I can not reproduce this error. Please give some more information: Did you observe any warnings or errors during the build process, or can you post a logfile of the complete build? Which lidar do you use? Does sick_scan_xd otherwise run successfully except for the service call?

The error message `undefined symbol: sensor_msgs__msg__PointCloud2__copy` indicates a missing link or dependency to the ros standard sensor messages. Did you call `source ./install/setup.bash` before the ros2 service call? Can you successfully call other services like `ros2 service call /rviz/list_parameters rcl_interfaces/srv/ListParameters` (assumed rivz2 is running), or can you build and run ros2 services like e.g. in https://docs.ros.org/en/foxy/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Cpp-Service-And-Client.html? If in doubt, please post a complete logfile of the build plus all sick_scan_xd output incl. the service call.

doubeLin, 2023-09-05:
no warnings or errors information were found during compilation. We are using LMS5xx radar.

ros2 service call /ColaMsg sick_scan/srv/ColaMsgSrv "{request: 'sRN SCdevicestate'}" and ros2 service call /ColaMsg sick_scan/srv/ColaMsgSrv "{request: 'sRA GetContaminationResult'}" call failed, prompt the same error message.

ros2 service call /rviz/list_parameters rcl_interfaces/srv/ListParameters call successfully .

Running the Little Turtle service in this environment can work normally.

doubeLin, 2023-09-05:
[Uploading build_log.txt…]()

doubeLin, 2023-09-05:
[CMakeLists.txt](https://github.com/SICKAG/sick_scan_xd/files/12519982/CMakeLists.txt)
[packagexml.txt](https://github.com/SICKAG/sick_scan_xd/files/12519999/packagexml.txt)

doubeLin, 2023-09-05:
[build_log.txt](https://github.com/SICKAG/sick_scan_xd/files/12520006/build_log.txt)

doubeLin, 2023-09-05:
[build_log.txt](https://github.com/SICKAG/sick_scan_xd/files/12520130/build_log.txt)

doubeLin, 2023-09-05:
[output.txt](https://github.com/SICKAG/sick_scan_xd/files/12520372/output.txt)

rostest, 2023-09-06:
Thanks for your feedback and logfiles. I still can not produce any errors when calling `ros2 service call /ColaMsg sick_scan/srv/ColaMsgSrv "{request: 'sRN SCdevicestate'}"` and I can not see any obvious reason, why `ros2 service` fails to import libsick_scan__rosidl_generator_c.so due to  `undefined symbol sensor_msgs__msg__PointCloud2__copy`.

Loading libraries (like libsick_scan__rosidl_generator_c.so) depend on the environment. Please check and post your environment variables PKG_CONFIG_PATH, AMENT_PREFIX_PATH, CMAKE_PREFIX_PATH, COLCON_PREFIX_PATH, PYTHONPATH, LD_LIBRARY_PATH and PATH after sourcing install/setup.bash:
```
source ./install/setup.bash
echo =$
echo PKG_CONFIG_PATH=$PKG_CONFIG_PATH
echo AMENT_PREFIX_PATH=$AMENT_PREFIX_PATH
echo CMAKE_PREFIX_PATH=$CMAKE_PREFIX_PATH
echo COLCON_PREFIX_PATH=$COLCON_PREFIX_PATH
echo PYTHONPATH=$PYTHONPATH
echo LD_LIBRARY_PATH=$LD_LIBRARY_PATH
echo PATH=$PATH
```

These environment variables should look like this:
```
PKG_CONFIG_PATH=/home/erlin/ros2_ws/install/msgpack11/lib/pkgconfig:/home/erlin/ros2_ws/install/libsick_ldmrs/lib/x86_64-linux-gnu/pkgconfig:/home/erlin/ros2_ws/install/libsick_ldmrs/lib/pkgconfig:/home/erlin/ros2_ws/install/msgpack11/lib/pkgconfig:/home/erlin/ros2_ws/install/libsick_ldmrs/lib/x86_64-linux-gnu/pkgconfig:/home/erlin/ros2_ws/install/libsick_ldmrs/lib/pkgconfig
AMENT_PREFIX_PATH=/home/erlin/ros2_ws/install/sick_scan:/home/erlin/ros2_ws/install/sick_scan:/opt/ros/foxy
CMAKE_PREFIX_PATH=/home/erlin/ros2_ws/install/sick_scan:/home/erlin/ros2_ws/install/libsick_ldmrs:/home/erlin/ros2_ws/install/sick_scan:/home/erlin/ros2_ws/install/libsick_ldmrs
COLCON_PREFIX_PATH=/home/erlin/ros2_ws/install:/home/erlin/ros2_ws/install
PYTHONPATH=/home/erlin/ros2_ws/install/sick_scan/lib/python3.8/site-packages:/home/erlin/ros2_ws/install/sick_scan/lib/python3.8/site-packages:/opt/ros/foxy/lib/python3.8/site-packages
LD_LIBRARY_PATH=/home/erlin/ros2_ws/install/sick_scan/lib:/home/erlin/ros2_ws/install/libsick_ldmrs/lib:/home/erlin/ros2_ws/install/sick_scan/lib:/home/erlin/ros2_ws/install/libsick_ldmrs/lib:/opt/ros/foxy/opt/yaml_cpp_vendor/lib:/opt/ros/foxy/opt/rviz_ogre_vendor/lib:/opt/ros/foxy/lib/x86_64-linux-gnu:/opt/ros/foxy/lib
PATH=/home/erlin/ros2_ws/install/libsick_ldmrs/bin:/home/erlin/ros2_ws/install/libsick_ldmrs/bin:/opt/ros/foxy/bin:/home/erlin/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin
```

According to build_log.txt, sick_scan is installed in folder `/home/erlin/ros2_ws/install`, but ros2 service uses another folder, i.e. `/home/local/ros2-ws/install` according to the ImportError traceback. Please remove any previous build and installation folders to avoid mismatches with previous versions:
```
cd /home/local/ros2-ws
rm -rf ./build ./build_isolated ./devel ./devel_isolated ./install ./install_isolated ./log
cd /home/erlin/ros2-ws
rm -rf ./build ./build_isolated ./devel ./devel_isolated ./install ./install_isolated ./log
```
and rebuild.

There is an additional section in your CMakeLists.txt wich includes CPack and InstallRequiredSystemLibraries:
```
# ament_package()
if(${CMAKE_SYSTEM_NAME} MATCHES "Linux")
    include(InstallRequiredSystemLibraries)
    set(CPACK_PACKAGING_INSTALL_PREFIX "/opt/ros/foxy")
    set(CPACK_DEBIAN_PACKAGE_ARCHITECTURE "amd64")
    set(MAJOR_VERSION "1")
    set(MINOR_VERSION "0")
    set(PATCH_VERSION "0")
    SET(CPACK_GENERATOR "DEB")
    SET(CPACK_PACKAGE_NAME "ros-foxy-sick-scan")
    set(CPACK_PACKAGE_VENDOR "LP-Research Inc. <nanjinglingque>")
    #set(CPACK_PACKAGE_DESCRIPTION_FILE "${CMAKE_CURRENT_SOURCE_DIR}/SUMMARY.txt")
    #set(CPACK_RESOURCE_FILE_LICENSE "${CMAKE_CURRENT_SOURCE_DIR}/LICENSE.txt")
    set(CPACK_PACKAGE_VERSION_MAJOR "${MAJOR_VERSION}")
    set(CPACK_PACKAGE_VERSION_MINOR "${MINOR_VERSION}")
    set(CPACK_PACKAGE_VERSION_PATCH "${PATCH_VERSION}")
    SET(CPACK_DEBIAN_PACKAGE_MAINTAINER "sick scan") #required
    set(CPACK_PACKAGE_DESCRIPTION_SUMMARY "sick scan")
    include(CPack)
endif()
```
To be on the safe side: Can you build without this addon, just to make sure that it has no side effect on the runtime environment?

cmake warns about missing diagnostic_updater. Depending on the ROS2 distribution, diagnostic_updater should be installed by `sudo apt-get install ros-foxy-diagnostic-updater ros-foxy-diagnostic-msgs` (see https://github.com/SICKAG/sick_scan_xd/blob/master/FAQ.md). To further reduce dependencies and potential dependency issues, sick_scan_xd can be built for LMS lidars without LDMRS and SCANSEGMENT_XD support. Please build sick_scan_xd with flags `-DLDMRS=0 -DSCANSEGMENT_XD=0`:
```
colcon build --packages-select sick_scan --cmake-args " -DROS_VERSION=2" " -DLDMRS=0" " -DSCANSEGMENT_XD=0" --event-handlers console_direct+
```

doubeLin, 2023-09-07:
It is indeed caused by a path abnormality. Thank you very much!

closed 2023-09-11 by doubeLin