SICKAG/sick_scan_xd #214: The 90 shift issue happens again in this new repository

Phymin, 2023-09-06:
The linked issue in the archived repository: [#138](https://github.com/SICKAG/sick_scan/issues/138)

unfortunately, this issue happens again in this new repository, please see the pictures below, the first one is from the archived repository, and the second one is from this new repository. Could you explain why this happens again? Thanks.
[image]
[image]

rostest, 2023-09-11:
We checked the angle shift for the TiM-240 with both sick_scan_xd and sick_scan.  rivz showed identical and correct pointcloud and laserscan messages in both versions:

rviz-screenshot using latest version of sick_scan_xd:
[image]

rviz-screenshot using sick_scan:
[image]

Did you use the latest sick_scan_xd release with default launchfile sick_tim_240.launch? Please post a complete logfile or screenshots of all sick_scan_xd messages from start. The output should look like this:
```
rostest@ROS-NB:~/ros_sick_scan_xd/sick_scan_xd_2.x.x_ros1$ roslaunch sick_scan sick_tim_240.launch hostname:=192.168.0.1
... logging to /home/rostest/.ros/log/6a603aaa-5088-11ee-8482-57e76cec8df8/roslaunch-ROS-NB-38975.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/rostest/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://ROS-NB:37761/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.8
 * /sick_tim_240/add_transform_check_dynamic_updates: False
 * /sick_tim_240/add_transform_xyz_rpy: 0,0,0,0,0,0
 * /sick_tim_240/client_authorization_pw: F4724744
 * /sick_tim_240/cloud_topic: cloud
 * /sick_tim_240/frame_id: cloud
 * /sick_tim_240/hostname: 192.168.0.1
 * /sick_tim_240/intensity: True
 * /sick_tim_240/intensity_resolution_16bit: True
 * /sick_tim_240/max_ang: 2.094395102
 * /sick_tim_240/message_monitoring_enabled: True
 * /sick_tim_240/min_ang: -2.094395102
 * /sick_tim_240/min_intensity: 0.0
 * /sick_tim_240/port: 2111
 * /sick_tim_240/range_filter_handling: 0
 * /sick_tim_240/range_max: 100.0
 * /sick_tim_240/range_min: 0.0
 * /sick_tim_240/read_timeout_millisec_default: 5000
 * /sick_tim_240/read_timeout_millisec_kill_node: 150000
 * /sick_tim_240/read_timeout_millisec_startup: 120000
 * /sick_tim_240/ros_qos: -1
 * /sick_tim_240/scanner_type: sick_tim_240
 * /sick_tim_240/start_services: True
 * /sick_tim_240/sw_pll_only_publish: True
 * /sick_tim_240/time_increment: 0.00019290123
 * /sick_tim_240/timelimit: 5
 * /sick_tim_240/use_binary_protocol: True
 * /sick_tim_240/use_generation_timestamp: True

NODES
  /
    sick_tim_240 (sick_scan/sick_generic_caller)

auto-starting new master
process[master]: started with pid [39002]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 6a603aaa-5088-11ee-8482-57e76cec8df8
process[rosout-1]: started with pid [39012]
started core service [/rosout]
process[sick_tim_240-2]: started with pid [39019]
[ INFO] [1694425733.660340853]: sick_generic_caller V. 2.10.0
[ INFO] [1694425733.661035070]: Program argument 1: /home/rostest/ros_sick_scan_xd/sick_scan_xd_2.x.x_ros1/devel_isolated/sick_scan/lib/sick_scan/sick_generic_caller
[ INFO] [1694425733.661059943]: Program argument 2: __log:=/home/rostest/.ros/log/6a603aaa-5088-11ee-8482-57e76cec8df8/sick_tim_240-2.log
[ INFO] [1694425733.661072547]: Program argument 3: __name:=sick_tim_240
[ INFO] [1694425733.661091453]: ==========================================
[... 52 lines cut ...]
[ INFO] [1694425735.131466625]: Sending  : <STX><STX><STX><STX><Len=0032>sWN LMDscandatacfg 0x01 0x00 0x01 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x00 0x01 CRC:<0x43>
[ INFO] [1694425735.133640562]: Receiving: <STX>sWA LMDscandatacfg <ETX>
[ INFO] [1694425735.133678431]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMDscandatacfg CRC:<0x67>
[ INFO] [1694425735.137645196]: Receiving: <STX>sRA LMDscandatacfg \x01\x00\x01\x01\x00\x00\x00\x00\x00\x00\x01\x00\x01<ETX>
[ INFO] [1694425735.138552920]: Sending  : <STX><STX><STX><STX><Len=0007>sMN Run CRC:<0x19>
[ INFO] [1694425735.141668673]: Receiving: <STX>sAN Run \x01<ETX>
[ INFO] [1694425735.141729799]: Sending  : <STX><STX><STX><STX><Len=0017>sEN LMDscandata 0x01 CRC:<0x33>
[ INFO] [1694425735.145625673]: Receiving: <STX>sEA LMDscandata \x01<ETX>
[ INFO] [1694425735.147017563]: SickScanServices: service "/sick_tim_240/ColaMsg" created ("/sick_tim_240/ColaMsg")
[ INFO] [1694425735.147498121]: SickScanServices: service "/sick_tim_240/ECRChangeArr" created ("/sick_tim_240/ECRChangeArr")
[ INFO] [1694425735.148178896]: SickScanServices: service "/sick_tim_240/LIDoutputstate" created ("/sick_tim_240/LIDoutputstate")
[ INFO] [1694425735.149018549]: SickScanServices: service "/sick_tim_240/SCdevicestate" created ("/sick_tim_240/SCdevicestate")
[ INFO] [1694425735.149942335]: SickScanServices: service "/sick_tim_240/SCreboot" created ("/sick_tim_240/SCreboot")
[ INFO] [1694425735.150625834]: SickScanServices: service "/sick_tim_240/SCsoftreset" created ("/sick_tim_240/SCsoftreset")
[ INFO] [1694425735.151368605]: SickScanServices: service "/sick_tim_240/SickScanExit" created ("/sick_tim_240/SickScanExit")
[ INFO] [1694425735.151420515]: SickScanServices: ros services initialized
[ INFO] [1694425735.151452014]: Setup completed, sick_scan_xd is up and running. Pointcloud is published on topic "cloud"
[ INFO] [1694425735.161444691]: Software PLL locking started, mapping ticks to system time.
[ INFO] [1694425735.161486981]: 1 / 6 packets dropped. Software PLL not yet locked.
[ INFO] [1694425735.231417654]: 2 / 6 packets dropped. Software PLL not yet locked.
[ INFO] [1694425735.301167657]: 3 / 6 packets dropped. Software PLL not yet locked.
[ INFO] [1694425735.370767468]: 4 / 6 packets dropped. Software PLL not yet locked.
[ INFO] [1694425735.440234432]: 5 / 6 packets dropped. Software PLL not yet locked.
[ INFO] [1694425735.509916754]: Software PLL is ready and locked now!
```

Phymin, 2023-10-31:
@rostest Could u reopen this issue? 
I think what I'm talking about is the point cloud but not sick scan. please see the pictures below and logs:
old sick_scan driver:
[image]
new sick_scan_xd driver:
[image]
output logs:
```
... logging to /root/.ros/log/837e489c-7789-11ee-884f-00e0672f74a2/roslaunch-MS-91-12473.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://MS-91:40943/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /sick_tim_240/add_transform_check_dynamic_updates: False
 * /sick_tim_240/add_transform_xyz_rpy: 0,0,0,0,0,0
 * /sick_tim_240/client_authorization_pw: F4724744
 * /sick_tim_240/cloud_topic: cloud
 * /sick_tim_240/frame_id: cloud
 * /sick_tim_240/hostname: 192.168.3.30
 * /sick_tim_240/intensity: True
 * /sick_tim_240/intensity_resolution_16bit: True
 * /sick_tim_240/max_ang: 2.094395102
 * /sick_tim_240/message_monitoring_enabled: True
 * /sick_tim_240/min_ang: -2.094395102
 * /sick_tim_240/min_intensity: 0.0
 * /sick_tim_240/port: 2111
 * /sick_tim_240/range_filter_handling: 0
 * /sick_tim_240/range_max: 100.0
 * /sick_tim_240/range_min: 0.0
 * /sick_tim_240/read_timeout_millisec_default: 5000
 * /sick_tim_240/read_timeout_millisec_kill_node: 150000
 * /sick_tim_240/read_timeout_millisec_startup: 120000
 * /sick_tim_240/ros_qos: -1
 * /sick_tim_240/scanner_type: sick_tim_240
 * /sick_tim_240/start_services: True
 * /sick_tim_240/sw_pll_only_publish: True
 * /sick_tim_240/time_increment: 0.00019290123
 * /sick_tim_240/timelimit: 5
 * /sick_tim_240/use_binary_protocol: False
 * /sick_tim_240/use_generation_timestamp: True

NODES
  /
    sick_tim_240 (sick_scan_xd/sick_generic_caller)

auto-starting new master
process[master]: started with pid [12483]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 837e489c-7789-11ee-884f-00e0672f74a2
process[rosout-1]: started with pid [12496]
started core service [/rosout]
process[sick_tim_240-2]: started with pid [12505]
[ INFO] [1698714299.934205211]: sick_generic_caller V. 3.0.0
[ INFO] [1698714299.935392082]: Program argument 1: /root/sick_ws/install/lib/sick_scan_xd/sick_generic_caller
[ INFO] [1698714299.935440264]: Program argument 2: __log:=/root/.ros/log/837e489c-7789-11ee-884f-00e0672f74a2/sick_tim_240-2.log
[ INFO] [1698714299.935469614]: Program argument 3: __name:=sick_tim_240
[ INFO] [1698714299.935495597]: ==========================================
[ INFO] [1698714299.941059390]: Range filter configuration: range_min=0, range_max=100, range_filter_handling=0
[ INFO] [1698714299.942172879]: Found sopas_protocol_type param overwriting default protocol:
[ INFO] [1698714299.942198253]: ASCII protocol activated
[ INFO] [1698714299.943869627]: PointCloudMonitor started.
[... 40 lines cut ...]
[ INFO] [1698714301.403663509]: Receiving: <STX>sWA LMPoutputRange<ETX>
[ INFO] [1698714301.403692142]: Sending  : <STX>sRN LMPoutputRange<ETX>
[ INFO] [1698714301.405558089]: Receiving: <STX>sRA LMPoutputRange 1 2710 FFEDB080 124F80<ETX>
[ INFO] [1698714301.405588458]: Angle resolution of scanner is 1 [deg]  (in 1/10000th deg: 10000)
[ INFO] [1698714301.407485253]: MIN_ANG (after command verification): -2.0944 [rad] -120 [deg]
[ INFO] [1698714301.407508575]: MAX_ANG (after command verification): 2.0944 [rad] 120 [deg]
[ INFO] [1698714301.407526791]: Sending  : <STX>sWN LMDscandatacfg 01 00 1 1 0 0 00 0 0 0 1 1<ETX>
[ INFO] [1698714301.408022983]: Receiving: <STX>sWA LMDscandatacfg<ETX>
[ INFO] [1698714301.408051443]: Sending  : <STX>sRN LMDscandatacfg<ETX>
[ INFO] [1698714301.409515983]: Receiving: <STX>sRA LMDscandatacfg 1 0 1 1 0 0 0 0 0 0 1 1<ETX>
[ INFO] [1698714301.410368879]: Sending  : <STX>sMN Run<ETX>
[ INFO] [1698714301.411517671]: Receiving: <STX>sAN Run 1<ETX>
[ INFO] [1698714301.411552615]: Sending  : <STX>sEN LMDscandata 1<ETX>
[ INFO] [1698714301.414064726]: Receiving: <STX>sEA LMDscandata 1<ETX>
[ INFO] [1698714301.415334699]: SickScanServices: service "/sick_tim_240/ColaMsg" created ("/sick_tim_240/ColaMsg")
[ INFO] [1698714301.415847180]: SickScanServices: service "/sick_tim_240/ECRChangeArr" created ("/sick_tim_240/ECRChangeArr")
[ INFO] [1698714301.416363100]: SickScanServices: service "/sick_tim_240/LIDoutputstate" created ("/sick_tim_240/LIDoutputstate")
[ INFO] [1698714301.416859494]: SickScanServices: service "/sick_tim_240/SCdevicestate" created ("/sick_tim_240/SCdevicestate")
[ INFO] [1698714301.417378077]: SickScanServices: service "/sick_tim_240/SCreboot" created ("/sick_tim_240/SCreboot")
[ INFO] [1698714301.417956607]: SickScanServices: service "/sick_tim_240/SCsoftreset" created ("/sick_tim_240/SCsoftreset")
[ INFO] [1698714301.418475837]: SickScanServices: service "/sick_tim_240/SickScanExit" created ("/sick_tim_240/SickScanExit")
[ INFO] [1698714301.418503090]: SickScanServices: ros services initialized
[ INFO] [1698714301.418522591]: Setup completed, sick_scan_xd is up and running. Pointcloud is published on topic "cloud"

```

rostest, 2023-10-31:
Thanks for further information and logfile. We could not reproduce this error with our previous test (see comment above). It might be related to the configuration of the ascii communication protocol type according to the logfile. Please use the default settings in sick_tim_240.launch incl. the binary protocol type, i.e. `<param name="use_binary_protocol" type="bool" value="True"/>` (default configuration in sick_tim_240.launch). The IP address of the lidar can be set either in the launchfile or by commandline option `hostname:=192.168.3.30`.

Phymin, 2023-11-01:
I will try and provide feedback, but I don't think it has something to do with the binary protocol, because the old sick_scan driver works normally also with ascii protocol.

rostest, 2023-11-01:
Many thanks for following up! Btw: Which of your screenshots shows the correct orientation, the one by sick_scan or by sick_scan_xd?

Phymin, 2023-11-01:
The screen shot by sick_scan is correct. And you are right, after set the protocol to binary, they are the same now, but I just don't understand why will this happen. Why will it has relationship with the protocol format? 
old_sick_scan driver(ASCII):
[image]
sick_scan_xd driver(Binary):
[image]

rostest, 2023-11-01:
Thank you for your feedback and testing. We are clarifying automatically switching into binary mode for the TiM240 (see #226). In the short term, we recommend using SOPAS ET to preset the lidars to binary mode.

Phymin, 2023-12-06:
Ok, will try it, thanks

Phymin, 2023-12-07:
After testing, it's confirmed it's working to change the ascii protocol to binary protocol, but after it's changed to binary mode, if I change `use_binary_protocol` to false, though at the begining from the output log, there is no error out, but there is also no data published. and after about 150s, there is error out which says there is no data published after about 150s and the driver shutdown automatically.

rostest, 2023-12-11:
Thanks for your feedback. Support for the ascii communication protocol is being phased out. Please use the binary communication mode with default setting `<param name="use_binary_protocol" type="bool" value="true"/>`.

closed 2023-12-11 by rostest