SICKAG/sick_scan_xd #216: Cannot build on ROS1 and Ubuntu 20.04

MechatronixX, 2023-09-21:
Following the instructions on https://github.com/SICKAG/sick_scan_xd/blob/master/INSTALL-ROS1.md#build-on-linux-ros1 I cannot build sick_generic_caller`. 

The issues stem from the `MIN, MAX` macros in `src/sick_scan_xd/include/sick_scan/sick_ros_wrapper.h` . They expand at the wrong places, e.g. in ros headers. 

```
/opt/ros/noetic/include/ros/time.h:262:63: note: in expansion of macro ‘MIN’
  262 |   template<> const WallTime TimeBase<WallTime, WallDuration>::MIN;
```

I could work myself by it quiet easily to get it to compile. What about not using these macros anywhere and relying on `std::min` and `std::max` directly instead?

rostest, 2023-09-22:
Thanks for reporting this. Depending on the ROS distribution, MIN and MAX can be multiple defined. 

This error will be corrected in the next release. You can use sick_scan_xd branch https://github.com/SICKAG/sick_scan_xd/tree/feature/bloom_pretest in the meantime. Use `git clone -b feature/bloom_pretest https://github.com/SICKAG/sick_scan_xd.git` to checkout.

Jonny0816, 2023-10-08:
Dear Sir 

We also met this issue under Ununtu 20.04, we want to use this driver with PicoScan LiDAR, the sick_scan_xd brand do not support picoscan, so want to check with you when the new driver version will be released? Look forward it.
Thanks
![image](https://github.com/SICKAG/sick_scan_xd/assets/97651961

rostest, 2023-10-09:
Thanks for your feedback. The next release is currently being tested. In the mean time, please use the new branch https://github.com/SICKAG/sick_scan_xd/tree/feature/min_max_fix to fix this error.

rostest, 2023-10-25:
Release 3.0 fixes this problem.

closed 2023-10-25 by rostest