SICKAG/sick_scan_xd #218: cannot re-init api in native windows application

derivativerobmc, 2023-10-10:
I'm trying to integrate SICK support into our native windows application. I have a simple example working connecting to a sensor and receiving pointcloud data using the callbacks. 

However, I have multiple usability problems if the user wants to change relaunch the sensor in order to change settings or fix a connection problem. I'm currently calling functions in this order:

SickScanApiLoadLibrary
handle = SickScanApiCreate( launchfile )
SickScanApiInitByCli( handle, launchfile  )
SickScanApiRegisterCartesianPointCloudMsg(handle)

* User hits re-init *

SickScanApiDeregisterCartesianPointCloudMsg(handle);
SickScanApiClose(handle);
SickScanApiRelease(handle);

newHandle = SickScanApiCreate( updated launch file )
SickScanApiInitByCli( handle, launch file )
SickScanApiRegisterCartesianPointCloudMsg(handle)

Unfortunately, this never seems to work correctly because of static thread objects that seem to persist between inits. The exact problems I get depend on whether it was able to successfully connect to the sensor in the first pass or not (sometimes it freezes my whole app waiting to join a thread and sometimes it just never updates with the new data).

Am I doing this correctly or is the API even designed for restarting like this? 

Also, a +1 for multi-sensor support in the same process. My initial review of the API made it sound like this was possible and its a major limitation for our user base to not be able to connect to multiple sensors in the same application.

rostest, 2023-10-10:
Thanks for reporting this issue. Your calling order to reinitialize the sensor looks as intended. We will test this scenario and analyze reinitialization problems. Which sensor did you use?

Concurrent use of multiple sensors in one process is currently not supported. We recommend to use multiple processes, such that each process serves one sensor.

derivativerobmc, 2023-10-10:
Thanks for looking into it. I'm happy to get your more specific callstacks or whatever else you need.

We're using a TIM551 sensor.

Re multi-sensors, I think we have a workaround by launching multiple copies of our software, but there is definitely a lot of extra overhead that way. Was it possible to connect to multiple sensors in one process using one of the older APIs? Our client was sure this had been done for them in an older project. However, even if that was true, I'm assuming you don't recommend going back to those APIs at this point?

rostest, 2023-10-18:
Unfortunately, the internal state was not completely reset by function SickScanApiInit. Please find a corrected version in branch https://github.com/SICKAG/sick_scan_xd/tree/feature/api_reinit.

Your re-initialization calls should now work as expected. The usage example [sick_scan_xd_api_test.cpp](https://github.com/SICKAG/sick_scan_xd/blob/feature/api_reinit/test/src/sick_scan_xd_api/sick_scan_xd_api_test.cpp) waits for key 'r' to repeat the lidar initialization in the same way you did. Thanks again for reporting this isssue!

commit 9ea62b3d (2023-10-18): Fix #218 API re-init
closed 2023-12-18 by rostest