SICKAG/sick_scan_xd #219: Request: ROS topic for multiScan IMU data

GXJ-ProfilingSystem, 2023-10-11:
We would like to use multiScan IMU data for robot operating.
(Currently, another IMU is installed in the robot independently. We would like to omit it.)

Are there any plan to output this data via ROS topic?

rostest, 2023-10-11:
Thanks for your feedback. We are planning to support multiScan IMU and publish corresponding ros messages (sensor_msgs/Imu) in one the next releases.

kajiyyuGXJ, 2023-10-13:
Thank you very much for quick response. Our colleagues are planning to try SLAM by only multiScan (with internal IMU). It will be much helpful. Would it be possible to tell me the rough schedule for the next releases?

rostest, 2023-10-16:
The schedule for multiscan IMU support is currently being clarified. We will come back to this question next week with a more concrete timeplan.

kajiyyuGXJ, 2023-10-16:
Many thanks! I look forward to hearing you again.

RobertoGuida, 2023-11-08:
Hi all, same issue here. 
We would like to use our multiScan IMU data too. We were too wondering if there is any update on support libraries realease.

kajiyyuGXJ, 2023-11-28:
Many thanks! My colleague could confirm the output of IMU rostopic, therefore we can close this thread. Thank you again for dedicated support.

closed 2023-11-28 by kajiyyuGXJ