SICKAG/sick_scan_xd #220: Requirement of PicoScan

Jonny0816, 2023-10-11:
Dear sir
For the Picoscan hardware, customer asked for two requirements, please kindly help to check whether can support:
1.Is it possible to create a Laserscan topic like TiM5 variant? Because some customers are using TiM5 and they want to use PicoScan with the same topic.
2.This driver supports all ranging LiDAR and with too much content in it, is it possible to create the driver with series products and delete other irrelevant product content？
thanks

rostest, 2023-10-16:
Thanks for your feedback. Laserscan messages for the picoScan can be generated, if the following parameter is added in launchfile sick_picoscan.launch:
```
<!-- Configuration of laserscan messages (ROS only): -->
<!-- Parameter "laserscan_layer_filter" sets a mask to create laserscan messages for configured layer (0: no laserscan message, 1: create laserscan messages for this layer) -->
<!-- Use "1" to activate resp. "0" to deactivate laserscan messages (picoScan has 1 layer) -->
<param name="laserscan_layer_filter" type="string" value="1" />
```
By adding this parameter, laserscan messages will be published on topic "/sick_picoscan/scan_segment" with frame id "world_0". These laserscan messages are generated for each scan segment. Currently the driver does not support laserscan messages for complete fullframes (-138.0 to +138.0 degree for piocScan lidars). We will discuss optional support of fullframe laserscan messages with SICK.

sick_scan_xd is a generic driver supporting a wide range of lidars. This allows the use of several types of lidars without the need to modify the API or application. Yherefore, product specific drivers are currently not favoured.

Jonny0816, 2023-10-20:
Thanks for the detail explanation.  Customer feedback each segment laserscan is not fulfill their requirement and full frame laserscan is needed, hope can speed up the process and support the full frame laserscan asap.
Thanks

rostest, 2023-10-25:
An update is available in branch https://github.com/SICKAG/sick_scan_xd/tree/feature/issue_update, which publishes fullframe laserscan messages for multiScan and picoScan lidars. Laserscan messages of segments are published on topic `/sick_picoscan/scan_segment`, laserscan messages of the fullframe scans are published on topic `/sick_picoscan/scan_fullframe`. Messages on both topics have frame id `world_1` or -if all echos are acitivated- frame ids `world_1_<echocnt>`. Please checkout this update and rebuild sick_scan_xd.

Phymin, 2023-12-14:
Hi, I have checkout the issue_update branch, but there is still no `/scan_fullframe` topic published, below is the output log:
```
[INFO] [sick_generic_caller-1]: process started with pid [11396]
[sick_generic_caller-1] [INFO] [1702553045.476448840] [sick_scan_xd]: sick_generic_caller V. 3.0.0
[sick_generic_caller-1] [INFO] [1702553045.476499501] [sick_scan_xd]: Program argument 1: /root/PrimeRobot/install/sick_scan_xd/lib/sick_scan_xd/sick_generic_caller
[sick_generic_caller-1] [INFO] [1702553045.476505896] [sick_scan_xd]: Program argument 2: /root/PrimeRobot/install/prime_launch/share/prime_launch/launch/sick_picoscan150.launch
[sick_generic_caller-1] [INFO] [1702553045.476509017] [sick_scan_xd]: Program argument 3: --ros-args
[sick_generic_caller-1] [INFO] [1702553045.476512264] [sick_scan_xd]: ==========================================
[sick_generic_caller-1] [INFO] [1702553045.476607220] [sick_scan_xd]: Try loading launchfile : /root/PrimeRobot/install/prime_launch/share/prime_launch/launch/sick_picoscan150.launch
[sick_generic_caller-1] [INFO] [1702553045.477178855] [sick_scan_xd]: LaunchParser::parseFile(/root/PrimeRobot/install/prime_launch/share/prime_launch/launch/sick_picoscan150.launch): default_args["hostname"]="192.168.3.30"
[sick_generic_caller-1] [INFO] [1702553045.477186404] [sick_scan_xd]: LaunchParser::parseFile(/root/PrimeRobot/install/prime_launch/share/prime_launch/launch/sick_picoscan150.launch): default_args["udp_receiver_ip"]="192.168.3.10"
[sick_generic_caller-1] [INFO] [1702553045.477190456] [sick_scan_xd]: LaunchParser::parseFile(/root/PrimeRobot/install/prime_launch/share/prime_launch/launch/sick_picoscan150.launch): default_args["nodename"]="sick_picoscan"
[sick_generic_caller-1] [INFO] [1702553045.477193833] [sick_scan_xd]: LaunchParser::parseFile(/root/PrimeRobot/install/prime_launch/share/prime_launch/launch/sick_picoscan150.launch): default_args["publish_topic"]="/cloud"
[sick_generic_caller-1] [INFO] [1702553045.477197720] [sick_scan_xd]: LaunchParser::parseFile(/root/PrimeRobot/install/prime_launch/share/prime_launch/launch/sick_picoscan150.launch): default_args["publish_topic_all_segments"]="/cloud_fullframe"
[sick_generic_caller-1] [INFO] [1702553045.477201471] [sick_scan_xd]: LaunchParser::parseFile(/root/PrimeRobot/install/prime_launch/share/prime_launch/launch/sick_picoscan150.launch): default_args["publish_frame_id"]="world"
[sick_generic_caller-1] [INFO] [1702553045.477204772] [sick_scan_xd]: LaunchParser::parseFile(/root/PrimeRobot/install/prime_launch/share/prime_launch/launch/sick_picoscan150.launch): default_args["add_transform_xyz_rpy"]="0,0,0,0,0,0"
[sick_generic_caller-1] [INFO] [1702553045.477208048] [sick_scan_xd]: LaunchParser::parseFile(/root/PrimeRobot/install/prime_launch/share/prime_launch/launch/sick_picoscan150.launch): default_args["scandataformat"]="1"
[sick_generic_caller-1] [INFO] [1702553045.477211545] [sick_scan_xd]: LaunchParser::parseFile(/root/PrimeRobot/install/prime_launch/share/prime_launch/launch/sick_picoscan150.launch): default_args["all_segments_min_deg"]="-138.0"
[sick_generic_caller-1] [INFO] [1702553045.477214783] [sick_scan_xd]: LaunchParser::parseFile(/root/PrimeRobot/install/prime_launch/share/prime_launch/launch/sick_picoscan150.launch): default_args["all_segments_max_deg"]="+138.0"
[sick_generic_caller-1] [INFO] [1702553045.477218305] [sick_scan_xd]: LaunchParser::parseFile(/root/PrimeRobot/install/prime_launch/share/prime_launch/launch/sick_picoscan150.launch): default_args["custom_pointclouds"]="cloud_unstructured_segments cloud_unstructured_fullframe cloud_polar_unstructured_segments cloud_polar_unstructured_fullframe cloud_all_fields_fullframe"
[sick_generic_caller-1] [INFO] [1702553045.477274796] [sick_scan_xd]: LaunchParser::parseFile(/root/PrimeRobot/install/prime_launch/share/prime_launch/launch/sick_picoscan150.launch): name="scanner_type", type="string", value="sick_picoscan"
[sick_generic_caller-1] [INFO] [1702553045.477280797] [sick_scan_xd]: LaunchParser::parseFile(/root/PrimeRobot/install/prime_launch/share/prime_launch/launch/sick_picoscan150.launch): name="hostname", type="string", value="$(arg hostname)" overwritten by default value "192.168.3.30"
[sick_generic_caller-1] [INFO] [1702553045.477284595] [sick_scan_xd]: LaunchParser::parseFile(/root/PrimeRobot/install/prime_launch/share/prime_launch/launch/sick_picoscan150.launch): name="hostname", type="string", value="192.168.3.30"
[sick_generic_caller-1] [INFO] [1702553045.477288019] [sick_scan_xd]: LaunchParser::parseFile(/root/PrimeRobot/install/prime_launch/share/prime_launch/launch/sick_picoscan150.launch): name="port", type="int", value="2115"
[sick_generic_caller-1] [INFO] [1702553045.477291679] [sick_scan_xd]: LaunchParser::parseFile(/root/PrimeRobot/install/prime_launch/share/prime_launch/launch/sick_picoscan150.launch): name="udp_receiver_ip", type="string", value="$(arg udp_receiver_ip)" overwritten by default value "192.168.3.10"
[sick_generic_caller-1] [INFO] [1702553045.477318240] [sick_scan_xd]: LaunchParser::parseFile(/root/PrimeRobot/install/prime_launch/share/prime_launch/launch/sick_picoscan150.launch): name="udp_receiver_ip", type="string", value="192.168.3.10"
[sick_generic_caller-1] [INFO] [1702553045.477322494] [sick_scan_xd]: LaunchParser::parseFile(/root/PrimeRobot/install/prime_launch/share/prime_launch/launch/sick_picoscan150.launch): name="udp_sender", type="string", value=""
[... cut ...]
[... 296 lines cut ...]
root@prime-YL-CLU6L-V1:~# ros2 topic list
/clicked_point
/cloud
/cloud_all_fields_fullframe
/cloud_polar_unstructured_fullframe
/cloud_polar_unstructured_segments
/cloud_unstructured_fullframe
/cloud_unstructured_segments
/diagnostics
/goal_pose
/initialpose
/parameter_events
/rosout
/scan_fullframe
/scan_segment
/sick_multiscan/encoder
/sick_multiscan/imu
/sick_multiscan/scan
/tf
/tf_static
```
no topic data is published:
[image]

Could you give me some clue where would the problem is? Thanks

Phymin, 2023-12-14:
@rostest

Phymin, 2023-12-15:
Nerver mind, it's ok now. thanks

closed 2023-12-18 by rostest