SICKAG/sick_scan_xd #221: No scandata while activated scan range filter

reinbja, 2023-10-13:
If the scan range filter is enabled, no scandata are available. If scan range filter is disabled, scandata are available.

reinbja, 2023-10-16:
I`m using a multiScan136. For your evaluation... the sensor was launched with no activated filters. After launching the multiScan, the `rangeFilter ` was activated. With default setting `-90° ...90°` it was working, but there is a point of adjustment, where the pointcloud is freezing in rviz 

**Working range**
- start angle theta: >= -18°
- stop angle theta: <=75°

There is also s.th. wrong with the layer filter. If Layer 1 or 16 is disabled the pointcloud is freezing as well

rostest, 2023-10-20:
Thanks for additional informations. Compact- and msgpack-format behave different in case of angle range filters. If an angle range filter is activated, fullframe pointclouds are currently published with mspack, but not necessarily when using compact-format. We will adapt the fullframe check for compact data with angle range filters and provide an update.

rostest, 2023-10-25:
An update is available in branch https://github.com/SICKAG/sick_scan_xd/tree/feature/issue_update, which corrects the fullframe pointclouds with activated angle range and layer filter. Please checkout this update and rebuild sick_scan_xd.

closed 2023-12-18 by rostest