SICKAG/sick_scan_xd #223: Compile error on native ubuntu 2204

cblesing, 2023-10-19:
Hi,

i have to use the sick_scan_xd library on a native ubuntu c++ project to read out a LSM 1xxx. When I try to compile the library I get a strange error: 

`[  7%] Building CXX object _deps/sick_scan_xd-build/CMakeFiles/sick_scan_xd_lib.dir/driver/src/abstract_parser.cpp.o
<command-line>: error: token "=" is not valid in preprocessor expressions
/home/vagrant/.cpm-source-cache/sick_scan_xd/9da84fc9d165813149d60f87a1b1a46864af7282/include/sick_scan/sick_scan_base.h:62:31: note: in expansion of macro ‘__ROS_VERSION’
   62 | #if !defined __ROS_VERSION || __ROS_VERSION == 0
      |                               ^~~~~~~~~~~~~
<command-line>: error: token "=" is not valid in preprocessor expressions
/home/vagrant/.cpm-source-cache/sick_scan_xd/9da84fc9d165813149d60f87a1b1a46864af7282/include/sick_scan/sick_ros_wrapper.h:118:5: note: in expansion of macro ‘__ROS_VERSION’
  118 | #if __ROS_VERSION <= 1 // ROS-SIMU (native Linux or Windows) or ROS-1 (Linux only)
      |     ^~~~~~~~~~~~~
<command-line>: error: token "=" is not valid in preprocessor expressions
/home/vagrant/.cpm-source-cache/sick_scan_xd/9da84fc9d165813149d60f87a1b1a46864af7282/include/sick_scan/sick_ros_wrapper.h:242:7: note: in expansion of macro ‘__ROS_VERSION’
  242 | #elif __ROS_VERSION == 2 // ROS-2 (Linux or Windows)
      |       ^~~~~~~~~~~~~
In file included from /home/vagrant/.cpm-source-cache/sick_scan_xd/9da84fc9d165813149d60f87a1b1a46864af7282/include/sick_scan/abstract_parser.h:39,
                 from /home/vagrant/.cpm-source-cache/sick_scan_xd/9da84fc9d165813149d60f87a1b1a46864af7282/driver/src/abstract_parser.cpp:63:
/home/vagrant/.cpm-source-cache/sick_scan_xd/9da84fc9d165813149d60f87a1b1a46864af7282/include/sick_scan/sick_ros_wrapper.h:446:2: error: #error __ROS_VERSION undefined or unsupported, build with __ROS_VERSION 0, 1 or 2
  446 | #error __ROS_VERSION undefined or unsupported, build with __ROS_VERSION 0, 1 or 2
      |  ^~~~~
<command-line>: error: token "=" is not valid in preprocessor expressions
/home/vagrant/.cpm-source-cache/sick_scan_xd/9da84fc9d165813149d60f87a1b1a46864af7282/include/sick_scan/sick_ros_wrapper.h:453:5: note: in expansion of macro ‘__ROS_VERSION’
  453 | #if __ROS_VERSION == 2 // ROS 2
      |     ^~~~~~~~~~~~~
<command-line>: error: token "=" is not valid in preprocessor expressions
/home/vagrant/.cpm-source-cache/sick_scan_xd/9da84fc9d165813149d60f87a1b1a46864af7282/include/sick_scan/sick_ros_wrapper.h:491:7: note: in expansion of macro ‘__ROS_VERSION’
  491 | #elif __ROS_VERSION == 1 // ROS 1
      |       ^~~~~~~~~~~~~
In file included from /home/vagrant/.cpm-source-cache/sick_scan_xd/9da84fc9d165813149d60f87a1b1a46864af7282/include/sick_scan/abstract_parser.h:39,
                 from /home/vagrant/.cpm-source-cache/sick_scan_xd/9da84fc9d165813149d60f87a1b1a46864af7282/driver/src/abstract_parser.cpp:63:
/home/vagrant/.cpm-source-cache/sick_scan_xd/9da84fc9d165813149d60f87a1b1a46864af7282/include/sick_scan/sick_ros_wrapper.h:503:10: fatal error: diagnostic_updater/diagnostic_updater.h: No such file or directory
  503 | #include <diagnostic_updater/diagnostic_updater.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [_deps/sick_scan_xd-build/CMakeFiles/sick_scan_xd_lib.dir/build.make:76: _deps/sick_scan_xd-build/CMakeFiles/sick_scan_xd_lib.dir/driver/src/abstract_parser.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1400: _deps/sick_scan_xd-build/CMakeFiles/sick_scan_xd_lib.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2`

Can anyone help me out?

cblesing, 2023-10-19:
I have answered it to my self. I forgot to set the ROS_VERSION ENV to 0. 
But now I got another error:

`[ 13%] Building CXX object _deps/sick_scan_xd-build/CMakeFiles/sick_scan_xd_lib.dir/driver/src/sick_generic_laser.cpp.o
In file included from /home/vagrant/.cpm-source-cache/sick_scan_xd/9da84fc9d165813149d60f87a1b1a46864af7282/include/sick_scan/ldmrs/sick_ldmrs_node.h:12,
                 from /home/vagrant/.cpm-source-cache/sick_scan_xd/9da84fc9d165813149d60f87a1b1a46864af7282/driver/src/sick_generic_laser.cpp:70:
/home/vagrant/.cpm-source-cache/sick_scan_xd/9da84fc9d165813149d60f87a1b1a46864af7282/include/sick_scan/ldmrs/sick_ldmrs_driver.hpp:58:10: fatal error: sick_ldmrs/manager.hpp: No such file or directory
   58 | #include <sick_ldmrs/manager.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [_deps/sick_scan_xd-build/CMakeFiles/sick_scan_xd_lib.dir/build.make:216: _deps/sick_scan_xd-build/CMakeFiles/sick_scan_xd_lib.dir/driver/src/sick_generic_laser.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1400: _deps/sick_scan_xd-build/CMakeFiles/sick_scan_xd_lib.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2`

How exactly is it possible to disable the LDMRS? I have set this:

`option(BUILD_WITH_LDMRS_SUPPORT "Build ${PROJECT_NAME} with LDMRS support (ON, libsick_ldmrs from https://github.com/SICKAG/libsick_ldmrs required) or without LDMRS support (OFF)" OFF)
if(WIN32)
    set(BUILD_WITH_LDMRS_SUPPORT OFF CACHE BOOL "Build ${PROJECT_NAME} with LDMRS support (ON, libsick_ldmrs from https://github.com/SICKAG/libsick_ldmrs required) or without LDMRS support (OFF)" FORCE)
elseif(LDMRS EQUAL 0)
    set(BUILD_WITH_LDMRS_SUPPORT OFF CACHE BOOL "Build ${PROJECT_NAME} with LDMRS support (ON, libsick_ldmrs from https://github.com/SICKAG/libsick_ldmrs required) or without LDMRS support (OFF)" FORCE)
else()
    set(BUILD_WITH_LDMRS_SUPPORT ON CACHE BOOL "Build ${PROJECT_NAME} with LDMRS support (ON, libsick_ldmrs from https://github.com/SICKAG/libsick_ldmrs required) or without LDMRS support (OFF)" FORCE)
endif()`

rostest, 2023-10-20:
Thanks for your feedback. Use cmake flags `-DLDMRS=0 -DSCANSEGMENT_XD=0` to build sick_scan_xd without LDMRS and scansegment_xd support, which are not required for LMS 1xxx lidars. See https://github.com/SICKAG/sick_scan_xd/blob/master/INSTALL-GENERIC.md#build-on-linux-generic-without-ros for further options. Please remove all temporary build and cmake files from the build folder to ensure a complete rebuild.

cblesing, 2023-10-20:
Thanks for the quick reply. That worked perfectly fine. But is it possible to use the sick_scan_xd without installing it? When I use the CPMAddPackage I ran into the error mentioned here https://github.com/SICKAG/sick_scan_xd/issues/223#issuecomment-1770726130

For some reason in that case the compiler wants to have the sick_ldmrs/manager.hpp even if I have set BUILD_WITH_LDMRS_SUPPORT and BUILD_WITH_SCANSEGMENT_XD_SUPPORT = OFF in the CPMAddPackage options section.

Any idea?

rostest, 2023-10-20:
It is possible to use sick_scan_xd without installation. Just make sure that the library can be found, i.e. environment variables `PATH` and `LD_LIBRARY_PATH` contain the sick_scan_xd build folder.

sick_generic_laser.cpp includes ldmrs-header, if LDMRS_SUPPORT is defined and not equal 0. It looks like LDMRS_SUPPORT is still defined within the CPM environment; maybe the CPM cache folder (`/home/vagrant/.cpm-source-cache`) needs to be cleared or removed to ensure a complete rebuild.

closed 2023-12-18 by rostest