SICKAG/sick_scan_xd #224: Error Sopas answer mismatch

lhy0709, 2023-10-19:
Hi,
Recently, I started to using sick lidar (TIM781S-2174104). After build on Linux generic without ROS, and running simple example. I received follow messages.

Before running in linux, I setted my sick lidar IP in window using SOPAS ET PC software.

[Info]: Try loading launchfile : ./sick_scan_xd/launch/sick_tim_7xxS.launch
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): default_args["hostname"]="192.168.0.10"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): default_args["cloud_topic"]="cloud"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): default_args["frame_id"]="cloud"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): default_args["sw_pll_only_publish"]="true"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): default_args["nodename"]="sick_tim_7xxS"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): default_args["port"]="2112"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): default_args["skip"]="0"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): default_args["add_transform_xyz_rpy"]="0,0,0,0,0,0"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): default_args["add_transform_check_dynamic_updates"]="false"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="scanner_type", type="string", value="sick_tim_7xxS"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="nodename", type="string", value="$(arg nodename)" overwritten by default value "sick_tim_7xxS"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="nodename", type="string", value="sick_tim_7xxS"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="min_ang", type="double", value="-2.35619449"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="max_ang", type="double", value="2.35619449"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="use_binary_protocol", type="bool", value="true"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="range_min", type="double", value="0.0"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="range_max", type="double", value="100.0"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="range_filter_handling", type="int", value="0"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="intensity", type="bool", value="True"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="hostname", type="string", value="$(arg hostname)" overwritten by default value "192.168.0.10"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="hostname", type="string", value="192.168.0.10"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="cloud_topic", type="string", value="$(arg cloud_topic)" overwritten by default value "cloud"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="cloud_topic", type="string", value="cloud"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="frame_id", type="str", value="$(arg frame_id)" overwritten by default value "cloud"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="frame_id", type="str", value="cloud"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="port", type="string", value="$(arg port)" overwritten by default value "2112"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="port", type="string", value="2112"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="skip", type="int", value="$(arg skip)" overwritten by default value "0"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="skip", type="int", value="0"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="timelimit", type="int", value="5"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="sw_pll_only_publish", type="bool", value="$(arg sw_pll_only_publish)" overwritten by default value "true"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="sw_pll_only_publish", type="bool", value="true"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="use_generation_timestamp", type="bool", value="true"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="start_services", type="bool", value="True"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="activate_lferec", type="bool", value="True"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="activate_lidoutputstate", type="bool", value="True"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="activate_lidinputstate", type="bool", value="True"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="min_intensity", type="double", value="0.0"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="add_transform_xyz_rpy", type="string", value="$(arg add_transform_xyz_rpy)" overwritten by default value "0,0,0,0,0,0"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="add_transform_xyz_rpy", type="string", value="0,0,0,0,0,0"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="add_transform_check_dynamic_updates", type="bool", value="$(arg add_transform_check_dynamic_updates)" overwritten by default value "false"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="add_transform_check_dynamic_updates", type="bool", value="false"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="message_monitoring_enabled", type="bool", value="True"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="read_timeout_millisec_default", type="int", value="5000"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="read_timeout_millisec_startup", type="int", value="120000"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="read_timeout_millisec_kill_node", type="int", value="150000"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="client_authorization_pw", type="string", value="F4724744"
[Info]: LaunchParser::parseFile(./sick_scan_xd/launch/sick_tim_7xxS.launch): name="ros_qos", type="int", value="-1"
scanner_type                   string     sick_tim_7xxS       
nodename                       string     sick_tim_7xxS       
min_ang                        double     -2.35619449         
max_ang                        double     2.35619449          
use_binary_protocol            bool       true                
range_min                      double     0.0                 
range_max                      double     100.0               
range_filter_handling          int        0                   
intensity                      bool       True                
hostname                       string     192.168.0.10        
cloud_topic                    string     cloud               
frame_id                       str        cloud               
port                           string     2112                
skip                           int        0                   
timelimit                      int        5                   
sw_pll_only_publish            bool       true                
[... 35 lines cut ...]
[Info]: SickCloudTransform: apply 3x3 rotation = false
[Info]: SickCloudTransform: additional translation = (0,0,0)
[Info]: SickCloudTransform: check_dynamic_updates = false
[Info]: sick_scan_xd: Tcp::open: connecting to 192.168.0.10:2112 ...
[Info]: sick_scan_xd Tcp::open: connected to 192.168.0.10:2112
[Info]: Parameter setting for <active_echo: 0>
[Info]: SickThread TcpRecvThread started.
[Info]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>
[Info]: Receiving: <STX>sRA SCdevicestate \x01<ETX>
[... cut ...] "sAN EIHstCola"

1. [Info]: Receiving: <STX>sAN SetAccessMode \x00<ETX>
=> Why the received SetAccessMode is 0?. In the telegram listing, it means error. I searched some previous issues, and they said the password will affect this sypmtoms. However, I didn't changed any password.

2. <STX>sFA\x00\x0a<ETX>
=> Lidar is setted the Cola binary mode. But it shows x0a. In the telegram listing, it means Sopas_Error_VARIABLE_WRITE_ACCESSDENIED.

Could you get some ideas? or need more log?

Best regards.

rostest, 2023-10-20:
Thanks for your feedback. The SOPAS command `sMN SetAccessMode 0x03 0x6f 0xd6 0x2c 0x05` is correct and uses the default password hash for the TiM781S. Currently I do not see a reason for a response with error code 0x0A.

Please reset the lidar to factory defaults and check the firmware version using SOPAS ET.

If the error still remains, you may try using launchfile `sick_tim_7xx.launch`, which uses the default authorization for the Tim7xx series (safety series TIM781S uses different defaults than TIM781). Just for a test in case that the default passwords had been changed for some reason.

lhy0709, 2023-10-20:
Thank you for your fast reply.

Device version is v5.0-30.06.22 and firmware version is v5.0.
Actually, I tired to reset the lidar to factory defaults using SOPAS ET. 
However, the password reset panel is not activated.

1. Is it possible that some lidar products could not reset factory defaults? (I'm using TIM781S)
I apologize for the question regarding SOPAS ET.

2. Is there any another way to reset it?

3. I tried to run sick_tim_7xx.launch. And checked the SetAccessMode send different value. Unfortunately, it also doesn't work.

4. Without communicate SetAccessMode, could I get the lidar data? That means, without knowing password, could I access lidar data using sick_scan_xd c++ library?

Best regards.

rostest, 2023-10-23:
Thanks for your informations. It is not possible to initialize the lidar and to use the sick_scan_xd library without successfull SetAccessMode command. Please contact SICK support for resetting your TiM781S to the factory defaults.

closed 2023-10-23 by rostest