SICKAG/sick_scan_xd #226: Could we change the protocol type  through commands?

Phymin, 2023-10-26:
Hi,

We're facing continuous data interruption issue with ASCII protocol on TIM-240(see the picture below), and it's suggested to change to binary protocol to prevent this issue, but now it's burdensome for us to change the protocol type through SOPAS ET, so I'm wondering if there is a way to change the protocol through commands? thanks.

[image]

rostest, 2023-10-26:
Thanks for your feedback. You can change the protocol type to Cola binary using SOPAS command `sWN EIHstCola 1` and write the settings to the EEPROM with command `sMN mEEwriteall`. See telegram listing 
https://cdn.sick.com/media/docs/7/27/927/telegram_listing_telegram_listing_ranging_sensors_lms1xx_lms5xx_tim2xx_tim5xx_tim7xx_lms1000_mrs1000_mrs6000_nav310_ld_oem15xx_ld_lrs36xx_lms4000_lrs4000_multiscan100_picoscan100_en_im0045927.pdf for details.

The sick_scan_xd driver should automatically switch to Cola binary and reconnect to the lidar. Did you use the lastest sick_scan_xd release?

According to the messages in the screenshot, the lidar does not respond to the initial SetAccessMode command. Can you check the network settings and tcp ports? If in doubt, you might use wireshark to see the network traffic with the TiM240. If there is no response at all from the lidar, the problem is more likely related to network settings (ip address or tcp port).

Phymin, 2023-10-30:
Yes, I tried with sick_scan_xd driver too, but without change the protocol to binary, the driver can't communicate with the lidar, that's the driver is not initialized successfully. Did I miss something, do you mean the sick_scan_xd driver will automatically change the protocol type if it's not the same with the configure file? 

The issue on the screen shot is not happened at the beginning, but running for a while or for a long time. And under this situation, we can ping through the ip of the lidar. most of the time, after power cicle it, the lidar will work again.

rostest, 2023-10-30:
Thanks for your reply. Yes, the sick_scan_xd driver should automatically switch to the binary protocol after start, using SOPAS command `sWN EIHstCola 1`. This causes a short delay after startup (a few seconds or less), which can be avoided by storing the binary mode in the lidars EEPROM. Otherwise protocol type switching should be done automatically. Can you post a complete logfile or screenshots from the whole initialization sequence, in case the protocol switching fails?

If the lidar fails to send its scandata after a long time and needs a power down and up cycle to restart, it might be a lidar problem. Please kill and restart sick_scan_xd driver in this error situation. Does the lidar respond after sick_scan_xd restart, or is a power cycle of the lidar (power down and up) required?

Phymin, 2023-10-30:
Hi thanks for your timely reply. Sorry, currently, we're still using sick_scan driver for our products. I want to use sick_scan_xd, but from my previous trying, I can't use the binary protocol if it's current ASCII format. I will try your method again to see if it works. BTW, how should I send the command `sWN EIHstCola 1`, should I modify the driver code to send it or can I send it through command line? 

Currently, we haven't tried with restart the driver, every time this issue happens, we either reboot the computer(I think this is similar with restart the driver) or power cycle the whole robot.

rostest, 2023-10-30:
Thanks for further information. Note that sick_scan is not supported any longer. Please migrate to sick_scan_xd. The sick_scan_xd monitors the lidar communication and reconnects automatically in case of network or communication problems, incl. switching to the binary protocol. 

If you use sick_scan_xd on ROS, you can send SOPAS commands like `sWN EIHstCola 1` using ros services, but this should not be necessary in normal operation.

Please do not hesitate to reopen this issue, if the problem remains with the latest sick_scan_xd release.

Phymin, 2023-10-31:
This is log from the latest release of sick_scan_xd:
```
... logging to /root/.ros/log/8192766a-7785-11ee-884f-00e0672f74a2/roslaunch-MS-91-5519.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://MS-91:37367/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /sick_tim_240/add_transform_check_dynamic_updates: False
 * /sick_tim_240/add_transform_xyz_rpy: 0,0,0,0,0,0
 * /sick_tim_240/client_authorization_pw: F4724744
 * /sick_tim_240/cloud_topic: cloud
 * /sick_tim_240/frame_id: cloud
 * /sick_tim_240/hostname: 192.168.3.30
 * /sick_tim_240/intensity: True
 * /sick_tim_240/intensity_resolution_16bit: True
 * /sick_tim_240/max_ang: 2.094395102
 * /sick_tim_240/message_monitoring_enabled: True
 * /sick_tim_240/min_ang: -2.094395102
 * /sick_tim_240/min_intensity: 0.0
 * /sick_tim_240/port: 2111
 * /sick_tim_240/range_filter_handling: 0
 * /sick_tim_240/range_max: 100.0
 * /sick_tim_240/range_min: 0.0
 * /sick_tim_240/read_timeout_millisec_default: 5000
 * /sick_tim_240/read_timeout_millisec_kill_node: 150000
 * /sick_tim_240/read_timeout_millisec_startup: 120000
 * /sick_tim_240/ros_qos: -1
 * /sick_tim_240/scanner_type: sick_tim_240
 * /sick_tim_240/start_services: True
 * /sick_tim_240/sw_pll_only_publish: True
 * /sick_tim_240/time_increment: 0.00019290123
 * /sick_tim_240/timelimit: 5
 * /sick_tim_240/use_binary_protocol: True
 * /sick_tim_240/use_generation_timestamp: True

NODES
  /
    sick_tim_240 (sick_scan_xd/sick_generic_caller)

ROS_MASTER_URI=http://localhost:11311

process[sick_tim_240-1]: started with pid [5536]
[ INFO] [1698712979.004624023]: sick_generic_caller V. 3.0.0
[ INFO] [1698712979.005621772]: Program argument 1: /root/sick_ws/install/lib/sick_scan_xd/sick_generic_caller
[ INFO] [1698712979.005641893]: Program argument 2: __log:=/root/.ros/log/8192766a-7785-11ee-884f-00e0672f74a2/sick_tim_240-1.log
[ INFO] [1698712979.005654445]: Program argument 3: __name:=sick_tim_240
[ INFO] [1698712979.005665449]: ==========================================
[ INFO] [1698712979.010854874]: Range filter configuration: range_min=0, range_max=100, range_filter_handling=0
[ INFO] [1698712979.012126725]: Found sopas_protocol_type param overwriting default protocol:
[ INFO] [1698712979.012156328]: Binary protocol activated
[ INFO] [1698712979.013912717]: PointCloudMonitor started.
[ INFO] [1698712979.013974814]: Start initialising scanner [Ip: 192.168.3.30] [Port:2111]
[ INFO] [1698712979.070594760]: Publishing lidar pointcloud2 to cloud
[ INFO] [1698712979.071099756]: Publishing on topic "/cloud", qos=10
[ INFO] [1698712979.072077162]: Publishing on topic "/sick_tim_240/imu", qos=10
[ INFO] [1698712979.073033331]: Publishing on topic "/sick_tim_240/encoder", qos=10
[ INFO] [1698712979.074023690]: Publishing on topic "/sick_tim_240/scan", qos=10
[ INFO] [1698712979.075438354]: SickCloudTransform: add_transform_xyz_rpy = (0,0,0,0,0,0)
[ INFO] [1698712979.075464503]: SickCloudTransform: azimuth_offset = 0 [deg]
[ INFO] [1698712979.075486611]: SickCloudTransform: additional 3x3 rotation matrix = { (1,0,0), (0,1,0), (0,0,1) }
[ INFO] [1698712979.075503025]: SickCloudTransform: apply 3x3 rotation = false
[ INFO] [1698712979.075532704]: SickCloudTransform: additional translation = (0,0,0)
[... 102 lines cut ...]
[ WARN] [1698713004.620741785]: Timeout during waiting for new datagram
[ INFO] [1698713004.620844090]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1698713004.621006375]: Receiving: <STX><ETX>
[ INFO] [1698713004.621077054]: SOPAS Communication -Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>

[ WARN] [1698713004.621159386]: ## ERROR SickScanCommon: sendSopasAndCheckAnswer("sRN SCdevicestate") failed
[ WARN] [1698713004.621206443]: checkColaDialect: lidar response not in configured Cola-dialect Cola-B, trying different Cola configuration
[ INFO] [1698713004.621274311]: Sending  : <STX>sRN SCdevicestate<ETX>
[ WARN] [1698713009.621500098]: Timeout during waiting for new datagram
[ INFO] [1698713009.621631284]: sendSOPASCommand: no full reply available for read after 5000 ms
[ INFO] [1698713009.621756987]: Receiving: <STX><ETX>
[ INFO] [1698713009.621828437]: SOPAS Communication -Error unexpected Sopas answer for request <STX>sRN SCdevicestate<ETX>

[ WARN] [1698713009.621910040]: ## ERROR SickScanCommon: sendSopasAndCheckAnswer("sRN SCdevicestate") failed
[ WARN] [1698713009.622001573]: checkColaDialect: no lidar response in any cola configuration, check lidar and network!
[ WARN] [1698713009.622062147]: SickScanCommon::init_scanner() failed, aborting.
[ WARN] [1698713009.622115312]: SickScanCommon::init_scanner(): checkColaDialect failed, restarting.
[ WARN] [1698713009.622157203]: It is recommended to use the binary communication (Cola-B) by:
[ WARN] [1698713009.622202500]: 1. setting parameter use_binary_protocol true in the launch file, and
[ WARN] [1698713009.622252763]: 2. setting the lidar communication mode with the SOPAS ET software to binary and save this setting in the scanner's EEPROM.
[ INFO] [1698713009.622365950]: Failed to init scanner Error Code: 1
Waiting for timeout...

```
What do you think the problem is?

Phymin, 2023-10-31:
@rostest could you reopen this issue, because I can't do it, thanks

rostest, 2023-10-31:
Thanks for further information and logfile. sick_scan_xd establishes a tcp connection to the lidar, but the lidar does not respond to any requests. This error might be related to a network/firewall problem or to the communication protocol type. Please check the two following settings:
* Is the lidar working with SOPAS ET in both binary and ascii communication? Can you switch between both protocol types in SOPAS ET without restart?
* Please set binary communication mode in SOPAS ET and start sick_scan_xd in binary mode (default settings or `<param name="use_binary_protocol" type="bool" value="True"/>` in the launchfile). Does sick_scan_xd receive now any answers from the lidar? Does sick_scan_xd receive any answers, if the lidar has been set to ascii communication mode by SOPAS ET before?

Phymin, 2023-11-01:
> Is the lidar working with SOPAS ET in both binary and ascii communication? Can you switch between both protocol types in SOPAS ET without restart?

A: yes, it working in both binary and ascii communication, and can switch both protocol types in SOPAS ET without restart. But the default format is ascii, as the screen shot shows. 
[image]

> Please set binary communication mode in SOPAS ET and start sick_scan_xd in binary mode (default settings or <param name="use_binary_protocol" type="bool" value="True"/> in the launchfile). Does [...]

A: After set to binary protocol without restart the lidar, the launch file with `use_binary_protocol` to True works.
```
... logging to /root/.ros/log/26f33f82-7898-11ee-bfc7-ede1e364f3ba/roslaunch-mp-104-4654.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://mp-104:42563/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /sick_tim_240/add_transform_check_dynamic_updates: False
 * /sick_tim_240/add_transform_xyz_rpy: 0,0,0,0,0,0
 * /sick_tim_240/client_authorization_pw: F4724744
 * /sick_tim_240/cloud_topic: cloud
 * /sick_tim_240/frame_id: cloud
 * /sick_tim_240/hostname: 192.168.3.30
 * /sick_tim_240/intensity: True
 * /sick_tim_240/intensity_resolution_16bit: True
 * /sick_tim_240/max_ang: 2.094395102
 * /sick_tim_240/message_monitoring_enabled: True
 * /sick_tim_240/min_ang: -2.094395102
 * /sick_tim_240/min_intensity: 0.0
 * /sick_tim_240/port: 2111
 * /sick_tim_240/range_filter_handling: 0
 * /sick_tim_240/range_max: 100.0
 * /sick_tim_240/range_min: 0.0
 * /sick_tim_240/read_timeout_millisec_default: 5000
 * /sick_tim_240/read_timeout_millisec_kill_node: 150000
 * /sick_tim_240/read_timeout_millisec_startup: 120000
 * /sick_tim_240/ros_qos: -1
 * /sick_tim_240/scanner_type: sick_tim_240
 * /sick_tim_240/start_services: True
 * /sick_tim_240/sw_pll_only_publish: True
 * /sick_tim_240/time_increment: 0.00019290123
 * /sick_tim_240/timelimit: 5
 * /sick_tim_240/use_binary_protocol: True
 * /sick_tim_240/use_generation_timestamp: True

NODES
  /
    sick_tim_240 (sick_scan_xd/sick_generic_caller)

auto-starting new master
process[master]: started with pid [4664]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 26f33f82-7898-11ee-bfc7-ede1e364f3ba
process[rosout-1]: started with pid [4677]
started core service [/rosout]
process[sick_tim_240-2]: started with pid [4684]
[ INFO] [1698830538.340981049]: sick_generic_caller V. 3.0.0
[ INFO] [1698830538.342381547]: Program argument 1: /root/sick_ws/install/lib/sick_scan_xd/sick_generic_caller
[ INFO] [1698830538.342411733]: Program argument 2: __log:=/root/.ros/log/26f33f82-7898-11ee-bfc7-ede1e364f3ba/sick_tim_240-2.log
[ INFO] [1698830538.342436268]: Program argument 3: __name:=sick_tim_240
[ INFO] [1698830538.342459114]: ==========================================
[ INFO] [1698830538.349571250]: Range filter configuration: range_min=0, range_max=100, range_filter_handling=0
[ INFO] [1698830538.351236990]: Found sopas_protocol_type param overwriting default protocol:
[ INFO] [1698830538.351271222]: Binary protocol activated
[... 50 lines cut ...]
[ INFO] [1698830539.866634229]: Receiving: <STX>sRA LMDscandatacfg \x01\x00\x01\x01\x00\x00\x00\x00\x00\x00\x01\x00\x01<ETX>
[ INFO] [1698830539.870656587]: Sending  : <STX><STX><STX><STX><Len=0007>sMN Run CRC:<0x19>
[ INFO] [1698830539.871680613]: Receiving: <STX>sAN Run \x01<ETX>
[ INFO] [1698830539.871861210]: Sending  : <STX><STX><STX><STX><Len=0017>sEN LMDscandata 0x01 CRC:<0x33>
[ INFO] [1698830539.874138545]: Receiving: <STX>sEA LMDscandata \x01<ETX>
[ INFO] [1698830539.880348911]: SickScanServices: service "/sick_tim_240/ColaMsg" created ("/sick_tim_240/ColaMsg")
[ INFO] [1698830539.882927021]: SickScanServices: service "/sick_tim_240/ECRChangeArr" created ("/sick_tim_240/ECRChangeArr")
[ INFO] [1698830539.885329215]: SickScanServices: service "/sick_tim_240/LIDoutputstate" created ("/sick_tim_240/LIDoutputstate")
[ INFO] [1698830539.887887376]: SickScanServices: service "/sick_tim_240/SCdevicestate" created ("/sick_tim_240/SCdevicestate")
[ INFO] [1698830539.890329540]: SickScanServices: service "/sick_tim_240/SCreboot" created ("/sick_tim_240/SCreboot")
[ INFO] [1698830539.892484887]: SickScanServices: service "/sick_tim_240/SCsoftreset" created ("/sick_tim_240/SCsoftreset")
[ INFO] [1698830539.894443919]: SickScanServices: service "/sick_tim_240/SickScanExit" created ("/sick_tim_240/SickScanExit")
[ INFO] [1698830539.894546478]: SickScanServices: ros services initialized
[ INFO] [1698830539.894627718]: Setup completed, sick_scan_xd is up and running. Pointcloud is published on topic "cloud"
[ INFO] [1698830539.917652413]: Software PLL locking started, mapping ticks to system time.
[ INFO] [1698830539.917767934]: 1 / 6 packets dropped. Software PLL not yet locked.
[ INFO] [1698830539.987180704]: 2 / 6 packets dropped. Software PLL not yet locked.
[ INFO] [1698830540.057017939]: 3 / 6 packets dropped. Software PLL not yet locked.
[ INFO] [1698830540.126656651]: 4 / 6 packets dropped. Software PLL not yet locked.
[ INFO] [1698830540.196409274]: 5 / 6 packets dropped. Software PLL not yet locked.
[ INFO] [1698830540.266048638]: Software PLL is ready and locked now!
```
Actually, we don't set the communication intentionally, because it's default communication protocol is ascii (if binary is better, why the default is ascii?). And if it's in ascii communication, the 'use_binary_protocol` set to True doesn't work. 

What we want is don't use SOPAS ET to set the communication type, because it's burdonsome to do this in the customer side, do you have any suggestions? Thanks

rostest, 2023-11-01:
Thank you for your feedback and testing. The TiM240 seems to use a slightly different logic compared to other lidars for switching  from ASCII mode to binary mode. We are clarifying with SICK support and management how to proceed. In the short term, we recommend using SOPAS ET to preset the lidars to binary mode.

rostest, 2023-11-30:
We updated the protocol switching for the TiM-240 in branch feature/tim240_protocol_switch. Please checkout https://github.com/SICKAG/sick_scan_xd/tree/feature/tim240_protocol_switch, rebuild and try again using binary protocol. If the TiM-240 is in Cola-Ascii mode, it will be switched to Cola-B (binary protocol) and restarted after a few seconds.

Phymin, 2023-12-08:
After testing, it's confirmed it's working to change the ascii protocol to binary protocol, but after it's changed to binary mode, if I change use_binary_protocol to false, though at the begining from the output log, there is no error out, but there is also no data published. and after about 150s, there is error out which says there is no data published after about 150s and the driver shutdown automatically.

rostest, 2023-12-11:
Thanks for your feedback. Support for the ascii communication protocol is being phased out. Please use the binary communication mode with default setting `<param name="use_binary_protocol" type="bool" value="true"/>`.

closed 2023-12-11 by rostest