SICKAG/sick_scan_xd #227: sick lms1xx driver error

kevinpretell, 2023-10-26:
I'm having problems starting the driver for the LMS111-10100 lidars
when I launch the driver I get the following error:

**`[WARN] [1698162918.267058959] [sick_scan_xd]: Error Sopas answer mismatch: Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 ..., received answer: "sFA\x00\x0a", expected patterns: "sWA LMPoutputRange","sAN LMPoutputRange"`**

here is the complete output (the error is on the last line):
```
ros2 run sick_scan_xd sick_generic_caller ./src/sick_scan_xd/launch/sick_lms_1xx.launch 
[INFO] [1698162916.840202314] [sick_scan_xd]: sick_generic_caller V. 3.0.0
[INFO] [1698162916.840237364] [sick_scan_xd]: Program argument 1: /home/usad2/usad2_ws/install/sick_scan_xd/lib/sick_scan_xd/sick_generic_caller
[INFO] [1698162916.840242885] [sick_scan_xd]: Program argument 2: ./src/sick_scan_xd/launch/sick_lms_1xx.launch
[INFO] [1698162916.840247025] [sick_scan_xd]: ==========================================
[INFO] [1698162916.840349575] [sick_scan_xd]: Try loading launchfile : ./src/sick_scan_xd/launch/sick_lms_1xx.launch
[INFO] [1698162916.840548007] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): default_args["hostname"]="192.168.1.3"
[INFO] [1698162916.840556377] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): default_args["cloud_topic"]="cloud"
[INFO] [1698162916.840562137] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): default_args["frame_id"]="cloud"
[INFO] [1698162916.840567937] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): default_args["sw_pll_only_publish"]="true"
[INFO] [1698162916.840573667] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): default_args["nodename"]="sick_lms_1xx"
[INFO] [1698162916.840579027] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): default_args["add_transform_xyz_rpy"]="0,0,0,0,0,0"
[INFO] [1698162916.840584377] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): default_args["add_transform_check_dynamic_updates"]="false"
[INFO] [1698162916.840590127] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): default_args["encoder_mode"]="-1"
[INFO] [1698162916.840617037] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="intensity", type="bool", value="False"
[INFO] [1698162916.840622967] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="intensity_resolution_16bit", type="bool", value="false"
[INFO] [1698162916.840628627] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="min_ang", type="double", value="-2.35619"
[INFO] [1698162916.840633837] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="max_ang", type="double", value="2.35619"
[INFO] [1698162916.840639597] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="frame_id", type="str", value="$(arg frame_id)" overwritten by default value "cloud"
[INFO] [1698162916.840645048] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="frame_id", type="str", value="cloud"
[INFO] [1698162916.840650218] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="use_binary_protocol", type="bool", value="true"
[INFO] [1698162916.840655428] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="scanner_type", type="string", value="sick_lms_1xx"
[INFO] [1698162916.840660688] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="range_min", type="double", value="0.05"
[INFO] [1698162916.840665788] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="range_max", type="double", value="25.0"
[INFO] [1698162916.840671128] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="range_filter_handling", type="int", value="0"
[INFO] [1698162916.840676478] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="hostname", type="string", value="$(arg hostname)" overwritten by default value "192.168.1.3"
[INFO] [1698162916.840682288] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="hostname", type="string", value="192.168.1.3"
[INFO] [1698162916.840687738] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="cloud_topic", type="string", value="$(arg cloud_topic)" overwritten by default value "cloud"
[INFO] [1698162916.840798379] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="cloud_topic", type="string", value="cloud"
[INFO] [1698162916.840804729] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="port", type="string", value="2112"
[INFO] [1698162916.840808489] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="timelimit", type="int", value="5"
[INFO] [1698162916.840813319] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="sw_pll_only_publish", type="bool", value="$(arg sw_pll_only_publish)" overwritten by default value "true"
[INFO] [1698162916.840817979] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="sw_pll_only_publish", type="bool", value="true"
[INFO] [1698162916.840822539] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="use_generation_timestamp", type="bool", value="true"
[INFO] [1698162916.840827259] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="start_services", type="bool", value="True"
[INFO] [1698162916.840833199] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="activate_lferec", type="bool", value="True"
[INFO] [1698162916.840840589] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="activate_lidoutputstate", type="bool", value="True"
[INFO] [1698162916.840845779] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="activate_lidinputstate", type="bool", value="True"
[INFO] [1698162916.840849999] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="min_intensity", type="double", value="0.0"
[INFO] [1698162916.840854539] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="encoder_mode", type="int", value="$(arg encoder_mode)" overwritten by default value "-1"
[INFO] [1698162916.840859099] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="encoder_mode", type="int", value="-1"
[INFO] [1698162916.840863329] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="ang_res", type="double", value="0.5"
[INFO] [1698162916.840868279] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="scan_freq", type="double", value="50"
[INFO] [1698162916.840872429] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="add_transform_xyz_rpy", type="string", value="$(arg add_transform_xyz_rpy)" overwritten by default value "0,0,0,0,0,0"
[INFO] [1698162916.840877209] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="add_transform_xyz_rpy", type="string", value="0,0,0,0,0,0"
[INFO] [1698162916.840882139] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="add_transform_check_dynamic_updates", type="bool", value="$(arg add_transform_check_dynamic_updates)" overwritten by default value "false"
[INFO] [1698162916.840887359] [sick_scan_xd]: LaunchParser::parseFile(./src/sick_scan_xd/launch/sick_lms_1xx.launch): name="add_transform_check_dynamic_updates", type="bool", value="false"
[... cut ...]
[WARN] [1698162918.267058959] [sick_scan_xd]: Error Sopas answer mismatch: Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 ..., received answer: "sFA\x00\x0a", expected patterns: "sWA LMPoutputRange","sAN LMPoutputRange"
```
then I kill the process because it keeps restarting

I saw that the error `sFA\x00\x0a` means `Sopas_Error_VARIABLE_WRITE_ACCESSDENIED`.
but I didn't change any variables in the launch file, I just changed the lidar IP.
maybe there is some value I should change in the launch file for the LMS111-10100 lidar?

rostest, 2023-10-26:
Thanks for your feedback. LMPoutputRange should be writable (since AccessMode has been set successfully); we will check this error. You may set parameter `ang_res` and `scan_freq` both to 0 in the launchfile to use the defaults and to avoid overwriting these values by command `sWN LMPoutputRange`.

rostest, 2023-10-30:
The firmware version of your lidar is 1.23. Please update the most recent firmware using SOPAS ET and retry.

kevinpretell, 2023-10-31:
To update the firmware I need to log in in “Service” mode from sopas right?

cause in 'Maintenance' and 'Authorized client' mode it doesn't make me click on Download Firmware

[image]

rostest, 2023-11-01:
Yes, please do so. Please contact SICK support in case of problems with SOPAS ET or firmware updates.

closed 2023-11-01 by rostest

kevinpretell, 2023-11-02:
Thanks for the help, but in the lidar manual there is no default password for the "service" user level but only for the "Maintenance personal" and "Authorized client" user levels with which I cannot update the firmware
(we have never updated the firmware before)

[image]

Do you know the default password for the "Service" user level?
Thanks in advance for the help

rostest, 2023-11-02:
@kevinpretell I'm sorry, but we can't publish SICK passwords. Please contact SICK support for a firmware update.

kevinpretell, 2023-11-02:
@rostest Perfect, I'll do it!
Thanks so much for your help!