SICKAG/sick_scan_xd #229: Error when using the NAV350 launchfile with a reflectors mapping

thiagodefreitas, 2023-11-09:
Hi, I am having issues running the nav350 launchfile with either a pre-recorded map on the device or forcing a reflectors mapping .lmk file. 

More precisely, the 3 scenarios occur:

1. If running with mapping on initialization, everything works fine
2. If disabling mapping on initialization and relying on a saved mapping from SOPAS ET, the node runs but the position gets stuck on a single value,  despite moving the robot.
3.  If trying to load a .lmk file, it gets stuck in the mNLAYAddLandmark message and the node hangs.

[sick_generic_caller.EXE-3] [INFO] [1699474502.903993900] [sick_scan_xd]: LaunchParser::parseFile(C:\opt\sick_ws\install\sick_scan_xd\share\sick_scan_xd\launch/sick_nav_350.launch): name="nav_set_landmark_layout_by_imk_file", type="string", value="$(arg nav_set_landmark_layout_by_imk_file)" overwritten by default value "C:/opt/sick_ws/install/sick_scan_xd/share/sick_scan_xd/launch/sopas/layout_humble.lmk"
[sick_generic_caller.EXE-3] [INFO] [1699474502.905604100] [sick_scan_xd]: LaunchParser::parseFile(C:\opt\sick_ws\install\sick_scan_xd\share\sick_scan_xd\launch/sick_nav_350.launch): name="nav_set_landmark_layout_by_imk_file", type="string", value="C:/opt/sick_ws/install/sick_scan_xd/share/sick_scan_xd/launch/sopas/layout_humble.lmk"
[sick_generic_caller.EXE-3] [INFO] [1699474502.907062700] [sick_scan_xd]: LaunchParser::parseFile(C:\opt\sick_ws\install\sick_scan_xd\share\sick_scan_xd\launch/sick_nav_350.launch): name="nav_do_initial_mapping", type="bool", value="$(arg nav_do_initial_mapping)" overwritten by default value "false"
[sick_generic_caller.EXE-3] [INFO] [1699474502.908843900] [sick_scan_xd]: LaunchParser::parseFile(C:\opt\sick_ws\install\sick_scan_xd\share\sick_scan_xd\launch/sick_nav_350.launch): name="nav_do_initial_mapping", type="bool", value="false"
[sick_generic_caller.EXE-3] [INFO] [1699474502.909700800] [sick_scan_xd]: LaunchParser::parseFile(C:\opt\sick_ws\install\sick_scan_xd\share\sick_scan_xd\launch/sick_nav_350.launch): name="nav_map_cfg_mean", type="int", value="50"
[sick_generic_caller.EXE-3] [INFO] [1699474502.910508600] [sick_scan_xd]: LaunchParser::parseFile(C:\opt\sick_ws\install\sick_scan_xd\share\sick_scan_xd\launch/sick_nav_350.launch): name="nav_map_cfg_neg", type="int", value="0"
[sick_generic_caller.EXE-3] [INFO] [1699474502.911577200] [sick_scan_xd]: LaunchParser::parseFile(C:\opt\sick_ws\install\sick_scan_xd\share\sick_scan_xd\launch/sick_nav_350.launch): name="nav_map_cfg_x", type="int", value="0"
[sick_generic_caller.EXE-3] [INFO] [1699474503.153232400] [sick_scan_xd]: LaunchParser::parseFile(C:\opt\sick_ws\install\sick_scan_xd\share\sick_scan_xd\launch/sick_nav_350.launch): name="nav_map_cfg_y", type="int", value="0"
[sick_generic_caller.EXE-3] [INFO] [1699474503.153855400] [sick_scan_xd]: LaunchParser::parseFile(C:\opt\sick_ws\install\sick_scan_xd\share\sick_scan_xd\launch/sick_nav_350.launch): name="nav_map_cfg_phi", type="int", value="0"
[sick_generic_caller.EXE-3] [INFO] [1699474503.154527500] [sick_scan_xd]: LaunchParser::parseFile(C:\opt\sick_ws\install\sick_scan_xd\share\sick_scan_xd\launch/sick_nav_350.launch): name="nav_map_cfg_reflector_size", type="int", value="80"
[sick_generic_caller.EXE-3] [INFO] [1699474503.155168600] [sick_scan_xd]: LaunchParser::parseFile(C:\opt\sick_ws\install\sick_scan_xd\share\sick_scan_xd\launch/sick_nav_350.launch): name="add_transform_xyz_rpy", type="string", value="$(arg add_transform_xyz_rpy)" overwritten by default value "0,0,0,0,0,0"
[sick_generic_caller.EXE-3] [INFO] [1699474503.156033000] [sick_scan_xd]: LaunchParser::parseFile(C:\opt\sick_ws\install\sick_scan_xd\share\sick_scan_xd\launch/sick_nav_350.launch): name="add_transform_xyz_rpy", type="string", value="0,0,0,0,0,0"
[sick_generic_caller.EXE-3] [INFO] [1699474503.156746200] [sick_scan_xd]: LaunchParser::parseFile(C:\opt\sick_ws\install\sick_scan_xd\share\sick_scan_xd\launch/sick_nav_350.launch): name="add_transform_check_dynamic_updates", type="bool", value="$(arg add_transform_check_dynamic_updates)" overwritten by default value "false"
[sick_generic_caller.EXE-3] [INFO] [1699474503.157670600] [sick_scan_xd]: LaunchParser::parseFile(C:\opt\sick_ws\install\sick_scan_xd\share\sick_scan_xd\launch/sick_nav_350.launch): name="add_transform_check_dynamic_updates", type="bool", value="false"
[sick_generic_caller.EXE-3] [INFO] [1699474503.158379500] [sick_scan_xd]: LaunchParser::parseFile(C:\opt\sick_ws\install\sick_scan_xd\share\sick_scan_xd\launch/sick_nav_350.launch): name="start_services", type="bool", value="True"
[sick_generic_caller.EXE-3] [INFO] [1699474503.159083700] [sick_scan_xd]: LaunchParser::parseFile(C:\opt\sick_ws\install\sick_scan_xd\share\sick_scan_xd\launch/sick_nav_350.launch): name="message_monitoring_enabled", type="bool", value="True"
[sick_generic_caller.EXE-3] [INFO] [1699474503.159738700] [sick_scan_xd]: LaunchParser::parseFile(C:\opt\sick_ws\install\sick_scan_xd\share\sick_scan_xd\launch/sick_nav_350.launch): name="read_timeout_millisec_default", type="int", value="5000"
[sick_generic_caller.EXE-3] [INFO] [1699474503.160459900] [sick_scan_xd]: LaunchParser::parseFile(C:\opt\sick_ws\install\sick_scan_xd\share\sick_scan_xd\launch/sick_nav_350.launch): name="read_timeout_millisec_startup", type="int", value="120000"
[sick_generic_caller.EXE-3] [INFO] [1699474503.161275200] [sick_scan_xd]: LaunchParser::parseFile(C:\opt\sick_ws\install\sick_scan_xd\share\sick_scan_xd\launch/sick_nav_350.launch): name="read_timeout_millisec_kill_node", type="int", value="150000"
[sick_generic_caller.EXE-3] [INFO] [1699474503.162114600] [sick_scan_xd]: LaunchParser::parseFile(C:\opt\sick_ws\install\sick_scan_xd\share\sick_scan_xd\launch/sick_nav_350.launch): name="client_authorization_pw", type="string", value="F4724744"
[sick_generic_caller.EXE-3] [INFO] [1699474503.169038600] [sick_scan_xd]: Range filter configuration: range_min=0.05, range_max=100, range_filter_handling=0
[sick_generic_caller.EXE-3] [INFO] [1699474503.169884300] [sick_scan_xd]: Found emul_sensor overwriting default settings. Emulation:False
[sick_generic_caller.EXE-3] [INFO] [1699474503.170338400] [sick_scan_xd]: Found sopas_protocol_type param overwriting default protocol:
[sick_generic_caller.EXE-3] [INFO] [1699474503.170793000] [sick_scan_xd]: Binary protocol activated
[sick_generic_caller.EXE-3] [INFO] [1699474503.171307000] [sick_scan_xd]: PointCloudMonitor started.
[sick_generic_caller.EXE-3] [INFO] [1699474503.171645700] [sick_scan_xd]: Start initialising scanner [Ip: 192.168.1.10] [Port:2112]
[sick_generic_caller.EXE-3] [INFO] [1699474503.174736500] [sick_scan_xd]: Publishing on topic "sick_nav_350/nav_pose", qos=0
[sick_generic_caller.EXE-3] [INFO] [1699474503.210591800] [sick_scan_xd]: PointCloudMonitor: subscribing to topic cloud, qos=0
[sick_generic_caller.EXE-3] [INFO] [1699474503.237136700] [sick_scan_xd]: Publishing on topic "sick_nav_350/nav_landmark", qos=0
[sick_generic_caller.EXE-3] [INFO] [1699474503.251121500] [sick_scan_xd]: Publishing on topic "sick_nav_350/nav_reflectors", qos=0
[sick_generic_caller.EXE-3] [INFO] [1699474503.342432800] [sick_scan_xd]: Publishing lidar pointcloud2 to cloud
[sick_generic_caller.EXE-3] [INFO] [1699474503.342835100] [sick_scan_xd]: Publishing on topic "cloud", qos=0
[sick_generic_caller.EXE-3] [INFO] [1699474503.354746500] [sick_scan_xd]: Publishing on topic "sick_nav_350/imu", qos=0
[sick_generic_caller.EXE-3] [INFO] [1699474503.366436100] [sick_scan_xd]: Publishing on topic "sick_nav_350/encoder", qos=0
[sick_generic_caller.EXE-3] [INFO] [1699474503.379824200] [sick_scan_xd]: Publishing on topic "sick_nav_350/scan", qos=0
[sick_generic_caller.EXE-3] [INFO] [1699474503.402774400] [sick_scan_xd]: SickCloudTransform: add_transform_xyz_rpy = (0,0,0,0,0,0)
[sick_generic_caller.EXE-3] [INFO] [1699474503.403715300] [sick_scan_xd]: SickCloudTransform: azimuth_offset = 0 [deg]
[sick_generic_caller.EXE-3] [INFO] [1699474503.404254400] [sick_scan_xd]: SickCloudTransform: additional 3x3 rotation matrix = { (1,0,0), (0,1,0), (0,0,1) }
[sick_generic_caller.EXE-3] [INFO] [1699474503.404686000] [sick_scan_xd]: SickCloudTransform: apply 3x3 rotation = false
[sick_generic_caller.EXE-3] [INFO] [1699474503.405079200] [sick_scan_xd]: SickCloudTransform: additional translation = (0,0,0)
[sick_generic_caller.EXE-3] [INFO] [1699474503.405387500] [sick_scan_xd]: SickCloudTransform: check_dynamic_updates = false
[... cut ...] 0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x5a> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x01> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x7e>

thiagodefreitas, 2023-11-09:
As additional info, it is running on Windows 10.0 with ROS 2.0.

rostest, 2023-11-13:
Thanks for reporting this issue! We will analyze the error and will come back to this at the end of next week.

thiagodefreitas, 2023-11-15:
As an update, the same behavior happens on Ubuntu 22.04, with ROS Humble.

thiagodefreitas, 2023-11-24:
> Thanks for reporting this issue! We will analyze the error and will come back to this at the end of next week.

Hi, Would you have any update on this?

p0rys, 2023-11-27:
Hi, I have the same error. Loading an IMK / LMK file (which is actually correct? SOPAS saves LMK and manual/driver say IMK) does not seem to work via the driver. After sending the first landmark the driver seems to get no response ("Sending : sMN mNLAYAddLandmark..." without ever getting a "Receiving: ..."). 

Also a permanent saving of the layout using SOPAS does not seem to be possible. After the device has been de-energised, the layout is gone again. Is this just an operator error? We need to get the device up and running quickly and, if it is not possible via the driver, we would simply save a map permanently via SOPAS and only use the driver for navigation.

rostest, 2023-11-27:
We will analyze the problem. Can you post your lmk-file for testing? Thanks in advance.

p0rys, 2023-11-27:
Sure! This is the content of my imk-file:

```
#SICK AG
#NAV Layout data
#FileFormat: 1.00
#Sopas NAV350 V1.22.2-12.05.2023
#Used Layers:
#LayerID #landmarks
#  000         9
#
globID     x[mm]     y[mm] type subtype size[mm] layer1 layer2 layer3
000001      1736     -2989    1       2       60      0              
000002     -1353     -2188    1       2       60      0              
000003     -8341     -5928    1       2       60      0              
000004    -11655     -5192    1       2       60      0              
000005    -15019     -4464    1       2       60      0              
000006     -6443      1835    1       2       60      0              
000007     -6147      3807    1       2       60      0              
000008     -6035      4393    1       2       60      0              
000009     -5689      6292    1       2       60      0              

```

rostest, 2023-11-30:
The error is fixed in the new branch feature/nav350_imk on https://github.com/SICKAG/sick_scan_xd/tree/feature/nav350_imk. Please update, rebuild and run again. Thanks!

p0rys, 2023-11-30:
Thank you for the quick update! I have tested the branch and unfortunately I have to report that the problem is still the same. According to the new log messages it reads the file without problems, but when sending the landmarks a lot of CRC messages are displayed and the scanner does not respond. Here is an extract from the log:

```
[sick_generic_caller-1] [INFO] [1701347222.779111219] [sick_scan_xd]: sick_scan_xd::readNAVIMKfile("/config/store/layouts/tvz_20231130.imk"): 29 landmarks:
[sick_generic_caller-1] [INFO] [1701347222.779223361] [sick_scan_xd]: landmark 1: 1 14740 -4211 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.779291988] [sick_scan_xd]: landmark 2: 2 10507 -4343 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.779530743] [sick_scan_xd]: landmark 3: 3 2563 -3788 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.779553831] [sick_scan_xd]: landmark 4: 4 -1982 -3819 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.779572905] [sick_scan_xd]: landmark 5: 5 -4619 -8397 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.779591401] [sick_scan_xd]: landmark 6: 6 -8977 -11527 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.779610192] [sick_scan_xd]: landmark 7: 7 -10019 -6854 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.779627450] [sick_scan_xd]: landmark 8: 8 -19252 -3664 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.779645386] [sick_scan_xd]: landmark 9: 9 -10097 2326 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.779665024] [sick_scan_xd]: landmark 10: 10 -8708 3152 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.779695609] [sick_scan_xd]: landmark 11: 11 -3721 2111 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.779721355] [sick_scan_xd]: landmark 12: 12 5903 5088 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.779739159] [sick_scan_xd]: landmark 13: 13 10342 -7756 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.779761224] [sick_scan_xd]: landmark 14: 14 10191 -11184 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.779780610] [sick_scan_xd]: landmark 15: 15 6662 -9005 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.779797965] [sick_scan_xd]: landmark 16: 16 -1124 -11004 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.779816360] [sick_scan_xd]: landmark 17: 17 -1517 -10992 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.779835364] [sick_scan_xd]: landmark 18: 18 535 -3842 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.779866730] [sick_scan_xd]: landmark 19: 19 -63 -3836 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.779889964] [sick_scan_xd]: landmark 20: 20 -9135 -3015 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.779910737] [sick_scan_xd]: landmark 21: 21 -16215 2640 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.779931468] [sick_scan_xd]: landmark 22: 22 5625 1897 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.779952443] [sick_scan_xd]: landmark 23: 23 59 2288 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.779973401] [sick_scan_xd]: landmark 24: 24 2466 -8878 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.779994716] [sick_scan_xd]: landmark 25: 25 1069 -11071 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.780015975] [sick_scan_xd]: landmark 26: 26 -26393 3190 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.780037056] [sick_scan_xd]: landmark 27: 27 -13866 -4105 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.780058573] [sick_scan_xd]: landmark 28: 28 -19388 2828 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.780079775] [sick_scan_xd]: landmark 29: 29 -1726 2537 1 2 60 0
[sick_generic_caller-1] [INFO] [1701347222.780155250] [sick_scan_xd]: Sending  : <STX><STX><STX><STX><Len=0022>sMN mNLAYEraseLayout 0x01 CRC:<0x7c>
[sick_generic_caller-1] [INFO] [1701347225.056797613] [sick_scan_xd]: Receiving: <STX>sAN mNLAYEraseLayout \x00<ETX>
[sick_generic_caller-1] [INFO] [1701347225.056899607] [sick_scan_xd]: Sending  : <STX><STX><STX><STX><Len=0022>sMN mNEVAChangeState 0x02 CRC:<0x72>
[sick_generic_caller-1] [INFO] [1701347225.058543649] [sick_scan_xd]: Receiving: <STX>sMA mNEVAChangeState <ETX>
[sick_generic_caller-1] [INFO] [1701347225.059619299] [sick_scan_xd]: Sending  : <STX><STX><STX><STX><Len=0231>sMN mNLAYAddLandmark 0x00 0x1d 0x00 0x00 0x39 0x94 0xff 0xff 0xef 0x8d 0x01 0x02 0x00 0x3c 0x00 0x01 0x00 0x00 0x00 0x00 0x29 0x0b 0xff 0xff 0xef 0x09 0x01 0x02 0x00 0x3c 0x00 0x01 0x00 0x00 0x00 0x00 0x0a 0x03 0xff 0xff 0xf1 0x34 0x01 0x02 0x00 0x3c 0x00 0x01 0x00 0x00 0xff 0xff 0xf8 0x42 0xff 0xff 0xf1 0x15 0x01 0x02 0x00 0x3c 0x00 0x01 0x00 0x00 0xff 0xff 0xed 0xf5 0xff 0xff 0xdf 0x33 0x01 0x02 0x00 0x3c 0x00 0x01 0x00 0x00 0xff 0xff 0xdc 0xef 0xff 0xff 0xd2 0xf9 0x01 0x02 0x00 0x3c 0x00 0x01 0x00 0x00 0xff 0xff 0xd8 0xdd 0xff 0xff 0xe5 0x3a 0x01 0x02 0x00 0x3c 0x00 0x01 0x00 0x00 0xff 0xff 0xb4 0xcc 0xff 0xff 0xf1 0xb0 0x01 0x02 0x00 0x3c 0x00 0x01 0x00 0x00 0xff 0xff 0xd8 0x8f 0x00 0x00 0x09 0x16 0x01 0x02 0x00 0x3c 0x00 0x01 0x00 0x00 0xff 0xff 0xdd 0xfc 0x00 0x00 0x0c 0x50 0x01 0x02 0x00 0x3c 0x00 0x01 0x00 0x00 0xff 0xff 0xf1 0x77 0x00 0x00 0x08 0x3f 0x01 0x02 0x00 0x3c 0x00 0x01 0x00 0x00 0x00 0x00 0x17 0x0f 0x00 0x00 0x13 0xe0 0x01 0x02 0x00 0x3c 0x00 0x01 0x00 0x00 0x00 0x00 0x28 0x66 0xff 0xff 0xe1 0xb4 0x01 0x02 0x00 0x3c 0x00 0x01 0x00 0x00 CRC:<0x00> CRC:<0x00> CRC:<0x27> CRC:<0xcf> CRC:<0xff> CRC:<0xff> CRC:<0xd4> CRC:<0x50> CRC:<0x01> CRC:<0x02> CRC:<0x00> CRC:<0x3c> CRC:<0x00> CRC:<0x01> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x00> CRC:<0x1a> CRC:<0x06> CRC:< [... cut ...] > CRC:<0xff> CRC:<0xff> CRC:<0xef> CRC:<0xf7> CRC:<0x01> CRC:<0x02> CRC:<0x00> CRC:<0x3c> CRC:<0x00> CRC:<0x01> CRC:<0x00> CRC:<0x00> CRC:<0xff> CRC:<0xff> CRC:<0xb4> CRC:<0x44> CRC:<0x00> CRC:<0x00> CRC:<0x0b> CRC:<0x0c> CRC:<0x01> CRC:<0x02> CRC:<0x00> CRC:<0x3c> CRC:<0x00> CRC:<0x01> CRC:<0x00> CRC:<0x00> CRC:<0xff> CRC:<0xff> CRC:<0xf9> CRC:<0x42> CRC:<0x00> CRC:<0x00> CRC:<0x09> CRC:<0xe9> CRC:<0x01> CRC:<0x02> CRC:<0x00> CRC:<0x3c> CRC:<0x00> CRC:<0x01> CRC:<0x00> CRC:<0x00> CRC:<0x04>
```

thiagodefreitas, 2023-11-30:
For me, it loads correctly.
```
[sick_generic_caller-1] [INFO] [1701346898.376085533] [sick_scan_xd]: Receiving: <STX>sWA NAVScanDataFormat <ETX>
[sick_generic_caller-1] [INFO] [1701346898.376218520] [sick_scan_xd]: Sending  : <STX><STX><STX><STX><Len=0025>sWN NPOSPoseDataFormat 0x01 0x01 CRC:<0x52>
[sick_generic_caller-1] [INFO] [1701346898.377556240] [sick_scan_xd]: Receiving: <STX>sWA NPOSPoseDataFormat <ETX>
[sick_generic_caller-1] [INFO] [1701346898.377904590] [sick_scan_xd]: sick_scan_xd::readNAVIMKfile("/home/tuenkers/belgo.lmk"): 6 landmarks:
[sick_generic_caller-1] [INFO] [1701346898.377944216] [sick_scan_xd]: landmark 1: 1 5279 -2291 1 2 90 1
[sick_generic_caller-1] [INFO] [1701346898.377965335] [sick_scan_xd]: landmark 2: 2 -5115 6554 1 2 90 1
[sick_generic_caller-1] [INFO] [1701346898.377983563] [sick_scan_xd]: landmark 3: 3 2318 -10222 1 2 90 1
[sick_generic_caller-1] [INFO] [1701346898.378001629] [sick_scan_xd]: landmark 4: 4 -2889 -12768 1 2 90 1
[sick_generic_caller-1] [INFO] [1701346898.378019435] [sick_scan_xd]: landmark 5: 5 -12877 -12319 1 2 90 1
[sick_generic_caller-1] [INFO] [1701346898.378037096] [sick_scan_xd]: landmark 6: 6 -21981 -2244 1 2 90 1
[sick_generic_caller-1] [INFO] [1701346898.378118128] [sick_scan_xd]: Sending  : <STX><STX><STX><STX><Len=0022>sMN mNLAYEraseLayout 0x01 CRC:<0x7c>
[sick_generic_caller-1] [INFO] [1701346901.826243160] [sick_scan_xd]: Receiving: <STX>sAN mNLAYEraseLayout \x00<ETX>
[sick_generic_caller-1] [INFO] [1701346901.826386716] [sick_scan_xd]: Sending  : <STX><STX><STX><STX><Len=0022>sMN mNEVAChangeState 0x02 CRC:<0x72>
[sick_generic_caller-1] [INFO] [1701346901.828041158] [sick_scan_xd]: Receiving: <STX>sMA mNEVAChangeState <ETX>
[sick_generic_caller-1] [INFO] [1701346901.829001497] [sick_scan_xd]: Sending  : <STX><STX><STX><STX><Len=0119>sMN mNLAYAddLandmark 0x00 0x06 0x00 0x00 0x14 0x9f 0xff 0xff 0xf7 0x0d 0x01 0x02 0x00 0x5a 0x00 0x01 0x00 0x01 0xff 0xff 0xec 0x05 0x00 0x00 0x19 0x9a 0x01 0x02 0x00 0x5a 0x00 0x01 0x00 0x01 0x00 0x00 0x09 0x0e 0xff 0xff 0xd8 0x12 0x01 0x02 0x00 0x5a 0x00 0x01 0x00 0x01 0xff 0xff 0xf4 0xb7 0xff 0xff 0xce 0x20 0x01 0x02 0x00 0x5a 0x00 0x01 0x00 0x01 0xff 0xff 0xcd 0xb3 0xff 0xff 0xcf 0xe1 0x01 0x02 0x00 0x5a 0x00 0x01 0x00 0x01 0xff 0xff 0xaa 0x23 0xff 0xff 0xf7 0x3c 0x01 0x02 0x00 0x5a 0x00 0x01 0x00 0x01 CRC:<0x1b>
[sick_generic_caller-1] [INFO] [1701346901.831854096] [sick_scan_xd]: Receiving: <STX>sAN mNLAYAddLandmark \x00\x00\x06\x00\x00\x00\x01\x00\x02\x00\x03\x00\x04\x00\x05<ETX>
[sick_generic_caller-1] [INFO] [1701346901.831973662] [sick_scan_xd]: Sending  : <STX><STX><STX><STX><Len=0020>sMN mNLAYStoreLayout CRC:<0x42>
[sick_generic_caller-1] [INFO] [1701346905.944435290] [sick_scan_xd]: Receiving: <STX>sAN mNLAYStoreLayout \x00<ETX>
[sick_generic_caller-1] [INFO] [1701346905.944598221] [sick_scan_xd]: Sending  : <STX><STX><STX><STX><Len=0022>sMN mNEVAChangeState 0x04 CRC:<0x74>
[sick_generic_caller-1] [INFO] [1701346905.946516863] [sick_scan_xd]: Receiving: <STX>sMA mNEVAChangeState <ETX>
[sick_generic_caller-1] [INFO] [1701346905.947202958] [sick_scan_xd]: NAV350: Sending: "sMN mNPOSGetData 1 2"
[sick_generic_caller-1] [INFO] [1701346905.947349998] [sick_scan_xd]: NAV350: sMN mNPOSGetData request was sent
[sick_generic_caller-1] [INFO] [1701346905.949125540] [sick_scan_xd]: SickScanServices: service "/ColaMsg" created ("/ColaMsg")
[sick_generic_caller-1] [INFO] [1701346905.949954249] [sick_scan_xd]: SickScanServices: service "/SCdevicestate" created ("/SCdevicestate")
[sick_generic_caller-1] [INFO] [1701346905.950592759] [sick_scan_xd]: SickScanServices: service "/SickScanExit" created ("/SickScanExit")
[sick_generic_caller-1] [INFO] [1701346905.950641937] [sick_scan_xd]: SickScanServices: ros services initialized
[sick_generic_caller-1] [INFO] [1701346905.950676973] [sick_scan_xd]: Setup completed, sick_scan_xd is up and running. Pointcloud is published on topic "cloud"
```
But the position is never updated:

```
---
header:
  stamp:
    sec: 1701348153
    nanosec: 586431980
  frame_id: cloud
x: -27058294
y: -27058294
phi: 0
opt_pose_data_valid: 1
output_mode: 1
timestamp: 1345305
mean_dev: 0
nav_mode: 2
info_state: 4285351183
quant_used_reflectors: 0
pose_valid: 1
pose_x: 27058.294921875
pose_y: 27058.294921875
pose_yaw: -3.1415927410125732

```
The reflectors seem to be published correctly, according to the position and RViz.

rostest, 2023-11-30:
@p0rys Thanks for following up! Can you send the imk-file tvz_20231130.imk for testing?

p0rys, 2023-11-30:
[tvz_20231130.zip](https://github.com/SICKAG/sick_scan_xd/files/13512999/tvz_20231130.zip)

This time as a zip-file to preserve any Windows/Linux characters/formatting etc.

thiagodefreitas, 2023-11-30:
> For me, now it loads correctly.
> 
> [Block (28 lines)](#)
> ```
> [sick_generic_caller-1] [INFO] [1701346898.376085533] [sick_scan_xd]: Receiving: <STX>sWA NAVScanDataFormat <ETX>
> [sick_generic_caller-1] [INFO] [1701346898.376218520] [sick_scan_xd]: Sending  : <STX><STX><STX><STX><Len=0025>sWN NPOSPoseDataFormat 0x01 0x01 CRC:<0x52>
> [sick_generic_caller-1] [INFO] [1701346898.377556240] [sick_scan_xd]: Receiving: <STX>sWA NPOSPoseDataFormat <ETX>
> [sick_generic_caller-1] [INFO] [1701346898.377904590] [sick_scan_xd]: sick_scan_xd::readNAVIMKfile("/home/robot/layout.lmk"): 6 landmarks:
> [sick_generic_caller-1] [INFO] [1701346898.377944216] [sick_scan_xd]: landmark 1: 1 5279 -2291 1 2 90 1
> [sick_generic_caller-1] [INFO] [1701346898.377965335] [sick_scan_xd]: landmark 2: 2 -5115 6554 1 2 90 1
> [sick_generic_caller-1] [INFO] [1701346898.377983563] [sick_scan_xd]: landmark 3: 3 2318 -10222 1 2 90 1
> [sick_generic_caller-1] [INFO] [1701346898.378001629] [sick_scan_xd]: landmark 4: 4 -2889 -12768 1 2 90 1
> [sick_generic_caller-1] [INFO] [1701346898.378019435] [sick_scan_xd]: landmark 5: 5 -12877 -12319 1 2 90 1
> [sick_generic_caller-1] [INFO] [1701346898.378037096] [sick_scan_xd]: landmark 6: 6 -21981 -2244 1 2 90 1
> [sick_generic_caller-1] [INFO] [1701346898.378118128] [sick_scan_xd]: Sending  : <STX><STX><STX><STX><Len=0022>sMN mNLAYEraseLayout 0x01 CRC:<0x7c>
> [sick_generic_caller-1] [INFO] [1701346901.826243160] [sick_scan_xd]: Receiving: <STX>sAN mNLAYEraseLayout \x00<ETX>
> [sick_generic_caller-1] [INFO] [1701346901.826386716] [sick_scan_xd]: Sending  : <STX><STX><STX><STX><Len=0022>sMN mNEVAChangeState 0x02 CRC:<0x72>
> [sick_generic_caller-1] [INFO] [1701346901.828041158] [sick_scan_xd]: Receiving: <STX>sMA mNEVAChangeState <ETX>
> [sick_generic_caller-1] [INFO] [1701346901.829001497] [sick_scan_xd]: Sending  : <STX><STX><STX><STX><Len=0119>sMN mNLAYAddLandmark 0x00 0x06 0x00 0x00 0x14 0x9f 0xff 0xff 0xf7 0x0d 0x01 0x02 0x00 0x5a 0x00 0x01 0x00 0x01 0xff 0xff 0xec 0x05 0x00 0x00 0x19 0x9a 0x01 0x02 0x00 0x5a 0x00 0x01 0x00 0x01 0x00 0x00 0x09 0x0e 0xff 0xff 0xd8 0x12 0x01 0x02 0x00 0x5a 0x00 0x01 0x00 0x01 0xff 0xff 0xf4 0xb7 0xff 0xff 0xce 0x20 0x01 0x02 0x00 0x5a 0x00 0x01 0x00 0x01 0xff 0xff 0xcd 0xb3 0xff 0xff 0xcf 0xe1 0x01 0x02 0x00 0x5a 0x00 0x01 0x00 0x01 0xff 0xff 0xaa 0x23 0xff 0xff 0xf7 0x3c 0x01 0x02 0x00 0x5a 0x00 0x01 0x00 0x01 CRC:<0x1b>
> [sick_generic_caller-1] [INFO] [1701346901.831854096] [sick_scan_xd]: Receiving: <STX>sAN mNLAYAddLandmark \x00\x00\x06\x00\x00\x00\x01\x00\x02\x00\x03\x00\x04\x00\x05<ETX>
> [sick_generic_caller-1] [INFO] [1701346901.831973662] [sick_scan_xd]: Sending  : <STX><STX><STX><STX><Len=0020>sMN mNLAYStoreLayout CRC:<0x42>
> [sick_generic_caller-1] [INFO] [1701346905.944435290] [sick_scan_xd]: Receiving: <STX>sAN mNLAYStoreLayout \x00<ETX>
> [sick_generic_caller-1] [INFO] [1701346905.944598221] [sick_scan_xd]: Sending  : <STX><STX><STX><STX><Len=0022>sMN mNEVAChangeState 0x04 CRC:<0x74>
> [sick_generic_caller-1] [INFO] [1701346905.946516863] [sick_scan_xd]: Receiving: <STX>sMA mNEVAChangeState <ETX>
> [sick_generic_caller-1] [INFO] [1701346905.947202958] [sick_scan_xd]: NAV350: Sending: "sMN mNPOSGetData 1 2"
> [sick_generic_caller-1] [INFO] [1701346905.947349998] [sick_scan_xd]: NAV350: sMN mNPOSGetData request was sent
> [sick_generic_caller-1] [INFO] [1701346905.949125540] [sick_scan_xd]: SickScanServices: service "/ColaMsg" created ("/ColaMsg")
> [sick_generic_caller-1] [INFO] [1701346905.949954249] [sick_scan_xd]: SickScanServices: service "/SCdevicestate" created ("/SCdevicestate")
> [sick_generic_caller-1] [INFO] [1701346905.950592759] [sick_scan_xd]: SickScanServices: service "/SickScanExit" created ("/SickScanExit")
> [sick_generic_caller-1] [INFO] [1701346905.950641937] [sick_scan_xd]: SickScanServices: ros services initialized
> [sick_generic_caller-1] [INFO] [1701346905.950676973] [sick_scan_xd]: Setup completed, sick_scan_xd is up and running. Pointcloud is published on topic "cloud"
> ```
> 
> But the position is never updated, despite moving the robot it remains constant at:
> 
> [Block (22 lines)](#)
> ```
> ---
> header:
>   stamp:
>     sec: 1701348153
>     nanosec: 586431980
>   frame_id: cloud
> x: -27058294
> y: -27058294
> phi: 0
> opt_pose_data_valid: 1
> output_mode: 1
> timestamp: 1345305
> mean_dev: 0
> nav_mode: 2
> info_state: 4285351183
> quant_used_reflectors: 0
> pose_valid: 1
> pose_x: 27058.294921875
> pose_y: 27058.294921875
> pose_yaw: -3.1415927410125732
> ```
> 
> The reflectors seem to be published correctly, according to the position and RViz.

As a reference, I had some landmarks on layer 1. By setting them to layer 0, the position is updated continuously. I will verify the behaviour with the changes. But for the moment, it seems to be fixed for my use case.

closed 2023-12-18 by rostest