SICKAG/sick_scan_xd #233: LMS511 msg packages dropped，how to slove it please?

hoopp, 2023-11-20:
INFO] [1700376704.584747433] [sick_scan]: SickScanServices: Request "sRN LCMstate" successfully sent, received reply "sRA LCMstate \x00"
[INFO] [1700376704.584756011] [sick_scan]: SickScanServices: request: "sRN LCMstate"
[INFO] [1700376704.584761805] [sick_scan]: SickScanServices: response: "sRA LCMstate \x00"
[INFO] [1700376709.591023984] [sick_scan]: SickScanServices: Sending request "sRN LCMstate"
[INFO] [1700376709.591088662] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0012>sRN LCMstate CRC:<0x7a>
[WARN] [1700376711.643457865] [sick_scan]: 8 of 5474 packets dropped (0.1 perc.), maxAbsDeltaTime=0.000
[WARN] [1700376711.643872990] [sick_scan]: More packages than expected were dropped!!
Check the network connection.
Check if the system time has been changed in a leap.
If the problems can persist, disable the software PLL with the option sw_pll_only_publish=False  !
[WARN] [1700376711.653279925] [sick_scan]: 9 of 5496 packets dropped (0.2 perc.), maxAbsDeltaTime=0.055
[WARN] [1700376711.653325790] [sick_scan]: More packages than expected were dropped!!
Check the network connection.
Check if the system time has been changed in a leap.
If the problems can persist, disable the software PLL with the option sw_pll_only_publish=False  !
[WARN] [1700376711.653900704] [sick_scan]: 10 of 5497 packets dropped (0.2 perc.), maxAbsDeltaTime=0.650
[WARN] [1700376711.653926820] [sick_scan]: More packages than expected were dropped!!
Check the network connection.
Check if the system time has been changed in a leap.
If the problems can persist, disable the software PLL with the option sw_pll_only_publish=False  !
[WARN] [1700376711.654388644] [sick_scan]: 11 of 5498 packets dropped (0.2 perc.), maxAbsDeltaTime=0.639
[WARN] [1700376711.654411008] [sick_scan]: More packages than expected were dropped!!
Check the network connection.
Check if the system time has been changed in a leap.
If the problems can persist, disable the software PLL with the option sw_pll_only_publish=False  !
[INFO] [1700376711.655676431] [sick_scan]: Receiving: <STX>sRA LCMstate \x00<ETX>
[WARN] [1700376711.655690942] [sick_scan]: 12 of 5499 packets dropped (0.2 perc.), maxAbsDeltaTime=0.922
[WARN] [1700376711.655735524] [sick_scan]: More packages than expected were dropped!!
Check the network connection.
Check if the system time has been changed in a leap.
If the problems can persist, disable the software PLL with the option sw_pll_only_publish=False  !
[INFO] [1700376711.655720159] [sick_scan]: SickScanServices: Request "sRN LCMstate" successfully sent, received reply "sRA LCMstate \x00"
[INFO] [1700376711.656192390] [sick_scan]: SickScanServices: request: "sRN LCMstate"
[INFO] [1700376711.656199598] [sick_scan]: SickScanServices: response: "sRA LCMstate \x00"
[INFO] [1700376716.661467659] [sick_scan]: SickScanServices: Sending request "sRN LCMstate"
[INFO] [1700376716.661531619] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0012>sRN LCMstate CRC:<0x7a>
[INFO] [1700376716.662364512] [sick_scan]: Receiving: <STX>sRA LCMstate \x00<ETX>
[INFO] [1700376716.662398962] [sick_scan]: SickScanServices: Request "sRN LCMstate" successfully sent, received reply "sRA LCMstate \x00"
[INFO] [1700376716.662407486] [sick_scan]: SickScanServices: request: "sRN LCMstate"
[INFO] [1700376716.662413374] [sick_scan]: SickScanServices: response: "sRA LCMstate \x00"
[INFO] [1700376721.667231338] [sick_scan]: SickScanServices: Sending request "sRN LCMstate"
[INFO] [1700376721.667279960] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0012>sRN LCMstate CRC:<0x7a>

hoopp, 2023-11-20:
[INFO] [1700376486.435826388] [sick_scan]: sick_generic_caller V. 2.9.1
[INFO] [1700376486.435908035] [sick_scan]: Program argument 1: /opt/ros/foxy/lib/sick_scan/sick_generic_caller
[INFO] [1700376486.435917230] [sick_scan]: Program argument 2: /opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch
[INFO] [1700376486.435922389] [sick_scan]: Program argument 3: --ros-args
[INFO] [1700376486.435927305] [sick_scan]: ==========================================
[INFO] [1700376486.436420856] [sick_scan]: Try loading launchfile : /opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch
[INFO] [1700376486.436970658] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): default_args["hostname"]="192.168.6.200"
[INFO] [1700376486.436986665] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): default_args["cloud_topic"]="/scan"
[INFO] [1700376486.436992673] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): default_args["frame_id"]="base_scan"
[INFO] [1700376486.437001257] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): default_args["sw_pll_only_publish"]="true"
[INFO] [1700376486.437006572] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): default_args["nodename"]="sick_lms_5xx"
[INFO] [1700376486.437011783] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): default_args["add_transform_xyz_rpy"]="0,0,0,0,0,0"
[INFO] [1700376486.437017575] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): default_args["add_transform_check_dynamic_updates"]="false"
[INFO] [1700376486.437044715] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): name="min_ang", type="double", value="-1.658062789"
[INFO] [1700376486.437051497] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): name="max_ang", type="double", value="1.658062789"
[INFO] [1700376486.437057507] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): name="cloud_topic", type="string", value="$(arg cloud_topic)" overwritten by default value "/scan"
[INFO] [1700376486.437062662] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): name="cloud_topic", type="string", value="/scan"
[INFO] [1700376486.437067622] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): name="frame_id", type="str", value="$(arg frame_id)" overwritten by default value "base_scan"
[INFO] [1700376486.437072079] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): name="frame_id", type="str", value="base_scan"
[INFO] [1700376486.437077133] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): name="scanner_type", type="string", value="sick_lms_5xx"
[INFO] [1700376486.437081705] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): name="range_min", type="double", value="0.05"
[INFO] [1700376486.437086076] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): name="range_max", type="double", value="100.0"
[INFO] [1700376486.437092131] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): name="range_filter_handling", type="int", value="0"
[INFO] [1700376486.437097480] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): name="intensity", type="bool", value="true"
[INFO] [1700376486.437102036] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): name="intensity_resolution_16bit", type="bool", value="false"
[INFO] [1700376486.437106985] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): name="hostname", type="string", value="$(arg hostname)" overwritten by default value "192.168.6.200"
[INFO] [1700376486.437126533] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): name="hostname", type="string", value="192.168.6.200"
[INFO] [1700376486.437132787] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): name="port", type="string", value="2112"
[INFO] [1700376486.437143925] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): name="timelimit", type="int", value="5"
[INFO] [1700376486.437149245] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): name="use_binary_protocol", type="bool", value="true"
[INFO] [1700376486.437153845] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): name="filter_echos", type="int", value="2"
[INFO] [1700376486.437158782] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): name="sw_pll_only_publish", type="bool", value="$(arg sw_pll_only_publish)" overwritten by default value "true"
[INFO] [1700376486.437163785] [sick_scan]: LaunchParser::parseFile(/opt/ros/foxy/share/sick_scan/launch/sick_lms_5xx.launch): name="sw_pll_only_publish", type="bool", value="true"
[... cut ...]
[INFO] [1700376487.892302767] [sick_scan]: sick_lms_5xx: activating field monitoring by lferec messages
[INFO] [1700376487.892313680] [sick_scan]: sick_lms_5xx: activating field monitoring by lidoutputstate messages
[INFO] [1700376487.892333337] [sick_scan]: sick_lms_5xx: activating field monitoring by lidinputstate messages
[INFO] [1700376487.892346943] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0012>sEN LFErec 0x01 CRC:<0x42>
[INFO] [1700376487.893040001] [sick_scan]: Receiving: <STX>sEA LFErec \x01<ETX>

hoopp, 2023-11-20:
The network cable connection is strong.There were more than 10 disconnections yesterday.

hoopp, 2023-11-21:
The IPC is connected to SICK lms511 through the enp1s0 network port. The kernel logs of the IPC are as follows:
[image]
After the network is disconnected（ enp1so link is down）, it will also appear that "[sick_scan]: More packages than expected were dropped!!
Check the network connection.
Check if the system time has been changed in a leap".
I repeatedly checked that the network cable connection between the two devices is secure and there is OK。
-- 1.Please ask that it is IPC's issue or sicklms511's issue? 
-- 2.How to analysis and solve it？

rostest, 2023-11-22:
Thanks for your feedback. sick_scan_xd converts the lidar timestamp to the ROS system time using a Software PLL. The difference between the lidar timestamp and the ros timestamp when receiving the scan data is computed and a linear regression between lidar and ros time is continously calculated. See https://github.com/SICKAG/sick_scan_xd/blob/master/doc/software_pll.md for details.

If a TCP packet is delayed due to network traffic, the distance between lidar and ros timestamp can differ from the values calculated so far. In this case, the packet is not used to adjust the Software PLL and a warning is printed. If this happens occasionally like `12 of 5499 packets dropped (0.2 perc.)`, it will have little impact on the timestamps generated. If this happens often, this indicates too high network traffic and may cause inaccurate timestamps.

Network disconnections like `enp1so link is down` are not caused by sick_scan_xd. sick_scan_xd may close and reconnect a tcp socket after network or other unrecoverable errors, but this will not shutdown the entire network.

closed 2023-12-18 by rostest

hoopp, 2023-12-18:
Thanks a lot!

rostest, 2023-12-18:
@hoopp You're welcome!