SICKAG/sick_scan_xd #234: Build error with msgpack11 and without LDMRS on ROS2 humble under ubuntu22.04

Phymin, 2023-11-20:
I followed the below commands to build sick_scan_xd package with msgpack11 and without LDMRS: 
```
colcon build --packages-select msgpack11 --cmake-args " -DMSGPACK11_BUILD_TESTS=0" --event-handlers console_direct+
source ./install/setup.bash
colcon build --packages-select sick_scan_xd --cmake-args " -DROS_VERSION=2" " -DLDMRS=0" --event-handlers console_direct+
```
but I got the following build error:
```
[ 88%] Building CXX object CMakeFiles/sick_scan_xd_lib.dir/driver/src/sick_scansegment_xd/msgpack_parser.cpp.o
/home/phymin/prime_slam_ws/src/primeslambot/third_party/sick_scan_xd/driver/src/sick_scansegment_xd/msgpack_parser.cpp:80:10: fatal error: msgpack11.hpp: No such file or directory
   80 | #include <msgpack11.hpp>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/sick_scan_xd_lib.dir/build.make:706: CMakeFiles/sick_scan_xd_lib.dir/driver/src/sick_scansegment_xd/msgpack_parser.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
[ 88%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/sick_scan_messages.cpp.o
[ 88%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/sick_scan_parse_util.cpp.o
[ 89%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/sick_scan_services.cpp.o
[ 89%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/sick_scan_xd_api/api_impl.cpp.o
[ 89%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/sick_scan_xd_api/sick_scan_api_converter.cpp.o
[ 90%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/softwarePLL.cpp.o
[ 90%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/tcp/Mutex.cpp.o
[ 90%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/tcp/SickThread.cpp.o
[ 90%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/tcp/Time.cpp.o
[ 91%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/tcp/colaa.cpp.o
[ 91%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/tcp/colab.cpp.o
[ 91%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/tcp/errorhandler.cpp.o
[ 92%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/tcp/tcp.cpp.o
[ 92%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/tcp/toolbox.cpp.o
[ 92%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/tcp/wsa_init.cpp.o
[ 92%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/tinyxml/tinystr.cpp.o
[ 93%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/tinyxml/tinyxml.cpp.o
[ 93%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/tinyxml/tinyxmlerror.cpp.o
[ 93%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/tinyxml/tinyxmlparser.cpp.o
[ 94%] Building C object CMakeFiles/sick_scan_xd_shared_lib.dir/roswrap/src/getopt/getopt.c.o
[ 94%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/roswrap/src/launchparser/launchparser.cpp.o
[ 94%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/roswrap/src/toojpeg/toojpeg.cpp.o
[ 94%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/sick_scansegment_xd/compact_parser.cpp.o
[ 95%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/sick_scansegment_xd/config.cpp.o
[ 95%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/sick_scansegment_xd/msgpack_converter.cpp.o
[ 95%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/sick_scansegment_xd/msgpack_exporter.cpp.o
[ 96%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/sick_scansegment_xd/msgpack_parser.cpp.o
[ 96%] Building CXX object CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/sick_scansegment_xd/msgpack_validator.cpp.o
/home/phymin/prime_slam_ws/src/primeslambot/third_party/sick_scan_xd/driver/src/sick_scansegment_xd/msgpack_parser.cpp:80:10: fatal error: msgpack11.hpp: No such file or directory
   80 | #include <msgpack11.hpp>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/sick_scan_xd_shared_lib.dir/build.make:706: CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/sick_scansegment_xd/msgpack_parser.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
[Processing: sick_scan_xd]                                     
gmake[1]: *** [CMakeFiles/Makefile2:601: CMakeFiles/sick_scan_xd_lib.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:628: CMakeFiles/sick_scan_xd_shared_lib.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
--- stderr: sick_scan_xd                                       
/home/phymin/prime_slam_ws/src/primeslambot/third_party/sick_scan_xd/driver/src/sick_scansegment_xd/msgpack_parser.cpp:80:10: fatal error: msgpack11.hpp: No such file or directory
   80 | #include <msgpack11.hpp>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/sick_scan_xd_lib.dir/build.make:706: CMakeFiles/sick_scan_xd_lib.dir/driver/src/sick_scansegment_xd/msgpack_parser.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
/home/phymin/prime_slam_ws/src/primeslambot/third_party/sick_scan_xd/driver/src/sick_scansegment_xd/msgpack_parser.cpp:80:10: fatal error: msgpack11.hpp: No such file or directory
   80 | #include <msgpack11.hpp>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/sick_scan_xd_shared_lib.dir/build.make:706: CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/sick_scansegment_xd/msgpack_parser.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:601: CMakeFiles/sick_scan_xd_lib.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:628: CMakeFiles/sick_scan_xd_shared_lib.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< sick_scan_xd [1min 7s, exited with code 2]

Summary: 0 packages finished [1min 7s]
  1 package failed: sick_scan_xd
  1 package had stderr output: sick_scan_xd
```
Could someone explain why and how can I solve it, thanks.

rostest, 2023-11-22:
Thanks for your feedback. To build sick_scan_xd with scansegment support (multiScan and picoScan), you need to checkout msgpack11, too:

```
pushd ./src
git clone https://github.com/SICKAG/msgpack11.git
git clone https://github.com/SICKAG/sick_scan_xd.git
popd
rm -rf ./build ./build_isolated/ ./devel ./devel_isolated/ ./install ./install_isolated/ ./log/
colcon build --packages-select msgpack11 --cmake-args " -DMSGPACK11_BUILD_TESTS=0" --event-handlers console_direct+
source ./install/setup.bash
colcon build --packages-select sick_scan_xd --cmake-args " -DROS_VERSION=2" " -DLDMRS=0" --event-handlers console_direct+
source ./install/setup.bash
```

See https://github.com/SICKAG/sick_scan_xd/blob/master/INSTALL-ROS2.md#build-on-linux-ros2 for further details and build options.

Alternatively, use the latest release 3.1.0. It contains msgpack11 without dependencies.

Phymin, 2023-11-22:
Please kindly check my build commands again, the first commands is build the msgpack11 and without error, the error is from build the sick_scan_xd after the msgpack11 build.

rostest, 2023-11-22:
Thanks for following up. Building sick_scan_xd fails because msgpack11.hpp is missing. This headerfile should be available after a successful msgpack build.

Which version of sick_scan_xd do you use? If possible, please update to the latest release 3.1.0, which avoids the msgpack11 dependencies of previous versions.

closed 2023-12-18 by rostest