SICKAG/sick_scan_xd #235: Failure to launch ROS driver for MRS6224

SICK-AliciaTee, 2023-11-22:
Hi I need to record rosbag point cloud files using ROS1 or ROS2 for a customer project. 

I have successfully installed ROS2 and the SICK_SCAN_XD ROS driver, at least I think I did (please see attached text file)

But the driver doesn't seem to launch, could you help see what is the issue?
Ubuntu 22.04
ROS version: ROS2 Humble

Thank you.  
[image]
[teealRECONDITEV20231117(1).txt](https://github.com/SICKAG/sick_scan_xd/files/13432738/teealRECONDITEV20231117.1.txt)
[teealRECONDITEV20231117(1).txt](https://github.com/SICKAG/sick_scan_xd/files/13432739/teealRECONDITEV20231117.1.txt)

rostest, 2023-11-22:
Thanks for your feedback and logfiles. Your build looks fine, you probably just missed `source ./install/setup.bash` before calling `ros2 run sick_scan_xd sick_generic_caller ./src/sick_scan_xd/launch/sick_mrs_6xxx.launch` or `ros2 launch sick_scan_xd sick_mrs_6xxx.launch.py`

rostest, 2023-11-24:
Thanks for following up. According to your screenshot and logfile, you have successfully built and installed sick_scan_xd in folder `~/sick_scan_ws`. Run `source ./install/setup.bash` and `ros2 run sick_scan_xd sick_generic_caller ./src/sick_scan_xd/launch/sick_mrs_6xxx.launch` in this directory:
```
cd ~/sick_scan_ws
source ./install/setup.bash
ros2 run sick_scan_xd sick_generic_caller ./src/sick_scan_xd/launch/sick_mrs_6xxx.launch`
```

If the problem remains, please rebuild and run sick_scan_xd by the following command:
```
rm -rf ~/sick_scan_ws
mkdir -p ~/sick_scan_ws
cd ~/sick_scan_ws
mkdir ./src
pushd ./src
git clone https://github.com/SICKAG/libsick_ldmrs.git
git clone https://github.com/SICKAG/sick_scan_xd.git
popd
rm -rf ./build ./build_isolated/ ./devel ./devel_isolated/ ./install ./install_isolated/ ./log/
source /opt/ros/humble/setup.bash
colcon build --packages-select libsick_ldmrs --event-handlers console_direct+
source ./install/setup.bash
colcon build --packages-select sick_scan_xd --cmake-args " -DROS_VERSION=2" --event-handlers console_direct+
source ./install/setup.bash
ros2 run sick_scan_xd sick_generic_caller ./src/sick_scan_xd/launch/sick_mrs_6xxx.launch
```

See https://github.com/SICKAG/sick_scan_xd/blob/master/INSTALL-ROS2.md#build-on-linux-ros2 for further details and build options and https://github.com/SICKAG/sick_scan_xd/blob/master/USAGE.md for sick_scan_xd run options.

Please send a complete logfile of the commands given above if the problem still remains.

closed 2023-12-18 by rostest