SICKAG/sick_scan_xd #236: Failure to launch ROS1 driver in ROS1Noetic in MRS6224

SICK-AliciaTee, 2023-11-22:
Hi, after not successful using ROS2, I installed Ubuntu 22.04 and ROS1 Noetic, then install the ROS driver, I did not encounter any issues this time. 

Still the same issue of unable to find package. 

Kindly advise?
[image]

rostest, 2023-11-22:
Thanks for your feedback. I guess you just missed `source ./install/setup.bash` before calling `roslaunch sick_scan_xd sick_mrs_6xxx.launch`. Please send a complete logfile of build and run if the problem remains.

SICK-AliciaTee, 2023-11-22:
[image]

Sorry but I do not think it worked. Any advise?

SICK-AliciaTee, 2023-11-22:
[2023-05-30 214648 startup packages .txt](https://github.com/SICKAG/sick_scan_xd/files/13435601/2023-05-30.214648.startup.packages.txt)

Hi I do not know where to get a log, as the files is installed two days ago, but this is the closest I can find, thank you.

SICK-AliciaTee, 2023-11-22:
I made some progress, please see logfile, still no point cloud but at least I got the launch file working
[teeal@RECONDITE~$ rostopic echo clo1.txt](https://github.com/SICKAG/sick_scan_xd/files/13438116/teeal%40RECONDITE.rostopic.echo.clo1.txt)
[teeal@RECONDITE~$ rostopic echo clo.txt](https://github.com/SICKAG/sick_scan_xd/files/13438118/teeal%40RECONDITE.rostopic.echo.clo.txt)
[teeal@RECONDITE~$ source optrosnoet.txt](https://github.com/SICKAG/sick_scan_xd/files/13438119/teeal%40RECONDITE.source.optrosnoet.txt)

Kindly advise how to get the point cloud display in rviz? Thank you.

SICK-AliciaTee, 2023-11-22:
[image]

Attached RVIZ screenshot, no point cloud, looks like there is some TF error

rostest, 2023-11-22:
Thanks for following up. The logfile looks fine, sick_scan_xd is running without failures. 
To add a pointcloud in rviz, just click the Add button and select pointcloud topic "cloud":
[image]
Select frame id "cloud" as "Fixed Frame" in "Global Options". 
On the commandline, you can use `rostopic echo cloud` to check that sick_scan_xd provides pointcloud data.

SICK-AliciaTee, 2023-11-23:
It works, thank you.

closed 2023-11-23 by SICK-AliciaTee