SICKAG/sick_scan_xd #237: How to launch the SICK Generic Driver without ROS

SICK-AliciaTee, 2023-11-22:
Hi, we have a multiScan user having issues with the installation of the SICK Generic Driver without ROS. 

Kindly look at the attached log and let us know how to troubleshoot, I think likely the computer does not have connection with the multiScan itself?

Another thing is that after installation, how do you obtain the point clouds and visualize the point cloud?

Thank you. 
[Error.txt](https://github.com/SICKAG/sick_scan_xd/files/13444221/Error.txt)

rostest, 2023-11-24:
Thanks for your feedback. You are right, sick_scan_xd has no connection to the lidar:
```
[Error]: Tcp::open: Failed to open TCP connection to 192.168.0.1:2111
```
Please check the network connection between the PC running sick_scan_xd and the lidar. Try to `ping 192.168.0.1` and open `//192.168.0.1` in your browser to use SOPAS Air. If SOPAS Air can connect to the multiScan, try again with sick_scan_xd.

rostest, 2023-11-24:
Without ROS, callbacks are used to provide scan data and point clouds. I.e. an application registers a callback, which is called from the sick_scan_xd library after receiving scan data from the lidar. See https://github.com/SICKAG/sick_scan_xd/blob/master/doc/sick_scan_api/sick_scan_api.md for details and examples.

Visualization is not part of sick_scan_xd and has to be implemented by the application if required. Examples in https://github.com/SICKAG/sick_scan_xd/blob/master/doc/sick_scan_api/sick_scan_api.md#complete-usage-example-in-c and https://github.com/SICKAG/sick_scan_xd/blob/master/doc/sick_scan_api/sick_scan_api.md#complete-usage-example-in-python show a minimalistic visualization by converting the point cloud to an image.

closed 2023-12-18 by rostest