SICKAG/sick_scan_xd #238: Unable to see point cloud after launching multiScan

SICK-AliciaTee, 2023-11-23:
Hi I run the launch file for multiScan (please see attached log). 

Looks like the launch file is running but I am not able to see any point cloud in RVIZ and did not see any point cloud either when playing the bag file in Foxglove. 

Bag File info below: 

teeal@RECONDITE:~/sick_scan_ws/devel_isolated$ rosbag info 2023-11-22-16-54-59.bag
path:        2023-11-22-16-54-59.bag
version:     2.0
duration:    0.2s
start:       Nov 22 2023 16:54:59.65 (1700700899.65)
end:         Nov 22 2023 16:54:59.82 (1700700899.82)
size:        21.1 KB
messages:    18
compression: none [1/1 chunks]
types:       dynamic_reconfigure/Config            [958f16a05573709014982821e6822580]
             dynamic_reconfigure/ConfigDescription [757ce9d44ba8ddd801bb30bc456f946f]
             rosgraph_msgs/Log                     [acffd30cd6b6de30f120938c17c593fb]
topics:      /multiScan/parameter_descriptions    1 msg     : dynamic_reconfigure/ConfigDescription
             /multiScan/parameter_updates         1 msg     : dynamic_reconfigure/Config
             /rosout                             16 msgs    : rosgraph_msgs/Log

Do you think there are some data missing here? Thank you. 

[image]
[image]

[teeal@RECONDITE~multiScan20231122t.txt](https://github.com/SICKAG/sick_scan_xd/files/13445438/teeal%40RECONDITE.multiScan20231122t.txt)

SICK-AliciaTee, 2023-11-23:
[image]

Attached with the SOPAS Air setup, what would be the host name and what would be the udp receive ip?

multiScan IP: 192.168.0.1

rostest, 2023-11-24:
Thanks for your feedback. hostname is the IP address of the lidar, i.e. 192.168.0.1. UDP receiver ip is the ip address of the PC running sick_scan_xd, i.e. 192.168.0.100 (the "Destination IP address" in SOPAS Air). 

The command `roslaunch sick_scan_xd sick_multiscan.launch hostname:=192.168.0.1 udp_receiver_ip:=192.168.0.100` in your logfile is correct and sick_scan_xd is working as expected. For segment lidars, i.e. multiScan and picoScan, sick_scan_xd publishes several pointclouds with different properties on different topics. You can see these pointclouds in the rviz dialog. Select one of them, e.g. "cloud_unstructured_fullframe" and you should see the pointcloud in rviz. 

See https://github.com/SICKAG/sick_scan_xd/blob/master/doc/sick_scan_segment_xd.md for details about the multiScan pointclouds.

closed 2023-12-18 by rostest