SICKAG/sick_scan_xd #239: [NAV350] NAVPoseData yaw incorrect

p0rys, 2023-11-24:
Hello folks,

I am in the process of writing a ROS2 node that combines the localisation data from 2 NAV350s. I have already successfully started two instances of the driver and get scan and reflector data displayed correctly in RViz (including transformation via my robot URDF).

For the combiner node I now wanted to use the "/sick_nav350/nav_pose" topic to query additional things like "pose_valid" or "infostate". I noticed that "pose_yaw" is apparently not sent in the ROS coordinate system but in the scanner coordinate system (0° backwards instead of forwards). Can anyone confirm this behaviour? If so, is this the intended behaviour? All other data (LaserScan etc) use the "correct" ROS coordinate system, which ensures that you can use either the scan data (0° forwards) or the pose data (0° backwards) with your URDF.

Thanks in advance!

rostest, 2023-11-27:
Thanks for your feedback. Topic "/sick_nav350/nav_pose" publishes the original data from the NAV-350 without transforms (just converted to a ros message). This is intended. ROS Pointcloud- and Laserscan- messages are transformed to ROS coordinates.

closed 2023-11-27 by rostest

p0rys, 2023-11-27:
Thanks for the answer! Then I will rotate the data in the downstream node. 

One suggestions: I would rephrase the comment [here](https://github.com/SICKAG/sick_scan_xd/blob/e5f72060256a61a0aaf7d1f17a78529a6e951c34/msg/NAVPoseData.msg#L17C1-L17C26) to "pose in ros units" instead of " pose in ros coordinates". This would avoid confusion in the future.

rostest, 2023-11-27:
@p0rys Thanks for the hint! We will correct the comment.