SICKAG/sick_scan_xd #240: Field evaluation of MRS1104C scanner not working with ROS driver

shreyashpalande, 2023-11-24:
Description:

I am attempting to use the field evaluation feature of the Sick MRS1104C scanner to activate an output when an object is detected within the evaluation zone. However, I am unable to get the field evaluation to work properly with the ROS driver.

Setup:

 Lidar Firmware version: 2.3.0.0R
 ROS noetic [sick_scan_xd Release v3.1.0](https://github.com/SICKAG/sick_scan_xd/releases/tag/3.1.0)
 ROS launch file used [sick_mrs_1xxx.launch](https://github.com/SICKAG/sick_scan_xd/blob/master/launch/sick_mrs_1xxx.launch)

Steps Followed:

1. Factory reset the scanner to default settings.
2. Rebooted the scanner.
3. Changed the IP address of the scanner from the default IP address.
4. Added the field evaluation by drawing it.
5. Saved the changes permanently to the system.
6. Rebooted the scanner.
7. Tested the field evaluation zone in SOPAS (it works: gives yellow indication, indicating that fields are calculated).
8. Connected to the ROS driver with the latest ROS release v3.1.1 (also tried with 3.1.0).
9. Checked if the field evaluation works (it does not work; gives red indication, indicating that field evaluation is deactivated).

Additional Attempts:
I tried calling the service `rosservice call /sick_mrs_1xxx/LIDoutputstate "active: true"`.

The following output was received:

```
[ INFO] [1700571541.213823083]: SickScanServices: Sending request "sEN LIDoutputstate 1"
[ INFO] [1700571541.214122450]: Sending  : <STX><STX><STX><STX><Len=0020>sEN LIDoutputstate 0x01 CRC:<0x50>
[ INFO] [1700571541.215397224]: Receiving: <STX>sEA LIDoutputstate \x01<ETX>
[ INFO] [1700571541.215613308]: SickScanServices: Request "sEN LIDoutputstate 1" successfully sent, received reply "sEA LIDoutputstate \x01"
[ INFO] [1700571541.215700477]: SickScanServices: request: "sEN LIDoutputstate 1"
[ INFO] [1700571541.215786485]: SickScanServices: response: "sEA LIDoutputstate \x01"
```

Expected Behavior:

The field evaluation should activate the output when an object is detected within the evaluation zone.

Actual Behavior:

The field evaluation does not activate the output and gives red indication in SOPAS, indicating that field evaluation is deactivated (you can see it in the images below)

[image]
[image]

Request:
I would appreciate it if you could investigate this issue and provide a solution.

Thank you

rostest, 2023-11-27:
Thanks for your feedback. sick_scan_xd currently supports field evaluation only for LMS1xx, LMS5xx, TiM7xx and TiM7xxS, not for MRS1104 (see https://github.com/SICKAG/sick_scan_xd/blob/master/doc/field_monitoring_extensions.md for details). Field evaluation for MRS1104 will be followed up by SICK feature requests.

rostest, 2023-11-29:
@shreyashpalande A sick_scan_xd update is available in branch feature/mrs1xxx_init on https://github.com/SICKAG/sick_scan_xd/tree/feature/mrs1xxx_init. The driver now preserves the field evaluation settings after start, i.e. field evaluation settings remain unchanged by sick_scan_xd.
Please update sick_scan_xd, rebuild and try again.