SICKAG/sick_scan_xd #243: Unable to get pointcloud successfully from windows

treeandcat, 2023-11-28:
Hello, I am currently using sick multiscan136 lidar to obtain point cloud data under window. I have modified the corresponding hostname and udp_receiver_ip in the sick_multiscan.launch file, and used a timer to obtain real-time point cloud data from the minimum_sick_scan_api_client.cpp file. In the first few times, real-time point clouds can be obtained and saved, but during the running of the program, memory-related errors will be reported and cause the program to crash.  Later, I have also modified the corresponding udp_receiver_ip and recompiled the program several times, but I was unable to obtain the point cloud data. The prompt message was:[Info]: SickScanServices: Request "sMN Run" successfully sent, received reply "sAN Run 1"
[Info]: SickScanServices::sendSopasCmdCheckResponse(): request: "sMN Run", response: "sAN Run 1". 
Could you please tell me how to modify it to correctly obtain the point cloud data and solve the memory error? Thank you very much! ! !

[image]
[image]

rostest, 2023-12-01:
Thanks for your feedback. Note that minimum_sick_scan_api_client.cpp is a tiny API example. It just uses a sleep for some seconds as a placeholder for an application. Increase or replace this placeholder sleep, e.g.
[image]

I can not reproduce any memory errors. Can you give some more information about your environment? Do you use Visual Studio 2019 for both sick_scan_xd library and minimum_sick_scan_api_client.exe? Did you build and run minimum_sick_scan_api_client.cpp as described in https://github.com/SICKAG/sick_scan_xd/blob/master/doc/sick_scan_api/sick_scan_api.md#minimalistic-usage-example-in-c-1, i.e. with just the print-statement in the callback?

rostest, 2023-12-04:
Thanks for your reply. Without having tested your example, I would recommend a few modifications:

* Line 54 `uint8_t* img_pixel = (uint8_t*)calloc(3 * img_width * img_height, sizeof(uint8_t));` allocates a new memory block of 3 MByte in each callback without freeing it. This can cause the bad allocation exception later. Since `img_pixel` is never used, the allocation can safely be removed.

* Line 74 `mycloud->push_back(point)` pushes each point separately to the end of the pointcloud memory. This may cause bottlenecks due to reallocations required by the push. Memory can be allocated once and outside the loop, e.g. by using `pcl::PointCloud::reserve(size_t n)` with n = msg->width * msg->height before line 55. If you collect points over 5 seconds, n should be (msg->width * msg->height * 240 * CATTIME / CLOCKS_PER_SEC) or greater.

* Line 72 `mylock.lock();` locks and unlocks the mutex for each point. Locking/unlocking each point of a pointcloud can be timeconsuming. This is not necessary here, since `mycloud` is used exclusively within the callback. If `mycloud` is shared, it is normally better to lock once before copying the scan data and to unlock afterwards.

* Line 111-114 might overwrite memory, if main is called with argc < 4. Allocate the new argv either explicitely or use a string array like `char* args[] = { "E:/work/code/sick_scan_ws/build/Release/minimum_sick_scan_api_client.exe", "E:/work/code/sick_scan_ws/sick_scan_xd/launch/sick_multiscan.launch", "hostname:=192.168.10.230", "udp_receiver_ip:=192.168.10.99" };`.

We recommend to copy the scan data to a threadsafe FIFO within the callback and to perform all data processing (like saving scans to disk) in a separate thread or the main loop. This way the callback will not be blocked during data processing.

If the last line printed on the console is `sAN Run 1` (as it is on your screenshot), sick_scan_xd most likely does not receive any scan data from the lidar via UDP. Make sure that `udp_receiver_ip` is correct (i.e. the ip address of your PC running minimum_sick_scan_api_client.exe). Open //192.168.10.230 in a browser and use SOPAS Air to check the pointcloud and settings. Deactivating and activating UDP scan data can help. If the problem remains, restart both minimum_sick_scan_api_client.exe and the lidar.

treeandcat, 2023-12-18:
Thank you very much for your careful and detailed reply! ! Sorry for the late reply as I have been busy recently.
Regarding the variable `img_pixel`, I'm sorry to find that I forgot to comment it. The subsequent code correction also gave me a lot of inspiration. I have made the correction according to the method you said, and opened another thread to process the point cloud in the callback. Very few people read the code I wrote so carefully. Thank you again for your patience and help;
Regarding the inability to obtain the point cloud data of the radar, I found that it may be because after pressing `ctrl+c` to interrupt the program, the subsequent release-related code was not executed, causing the port to be occupied the next time the program was executed, and the point cloud data could not be obtained by udp. I added code to capture keyboard signals to ensure subsequent release operations work properly. However, I found that after adding the relevant code, the point cloud was still unable to be obtained. Using the `netstat -ano | findstr` command in the cmd window, I found that ports `2111` and `2115` were not occupied, but the point cloud was still unable to be obtained.
```
scanner_type                   string     sick_multiscan
hostname                       string     192.168.0.237
udp_receiver_ip                string     192.168.0.236
udp_sender                     string
udp_port                       int        2115
segment_count                  int        12
publish_frame_id               string     world
udp_input_fifolength           int        20
msgpack_output_fifolength      int        20
verbose_level                  int        1
measure_timing                 bool       True
export_csv                     bool       False
export_udp_msg                 bool       False
logfolder                      string
udp_timeout_ms                 int        60000
scandataformat                 int        2
range_min                      double     0.0
range_max                      double     100.0
range_filter_handling          int        0
add_transform_xyz_rpy          string     0,0,0,0,0,0
add_transform_check_dynamic_updates bool       false
sopas_tcp_port                 string     2111
start_sopas_service            bool       True
send_sopas_start_stop_cmd      bool       True
sopas_cola_binary              bool       False
sopas_timeout_ms               int        5000
client_authorization_pw        string     F4724744
host_read_filtersettings       bool       True
host_FREchoFilter              int        2
host_set_FREchoFilter          bool       True
host_LFPangleRangeFilter       string     0 -180.0 +179.0 -90.0 +90.0 1
host_set_LFPangleRangeFilter   bool       False
host_LFPlayerFilter            string     0 0 0 0 0 0 1 0 0 0 0 0 0 0 1 0 0
host_set_LFPlayerFilter        bool       False
msgpack_validator_enabled      bool       False
msgpack_validator_verbose      int        1
msgpack_validator_discard_msgpacks_out_of_bounds bool       True
msgpack_validator_check_missing_scandata_interval int        12
msgpack_validator_required_echos string     0
msgpack_validator_azimuth_start double     -180.0
msgpack_validator_azimuth_end  double     +180.0
msgpack_validator_elevation_start double     -90.0
msgpack_validator_elevation_end double     +90.0
msgpack_validator_valid_segments string     0 1 2 3 4 5 6 7 8 9 10 11
msgpack_validator_layer_filter string     0 0 0 0 0 1 0 0 0 0 0 0 0 1 0 0
ros_qos                        int        -1
laserscan_layer_filter         string     0 0 0 0 0 1 0 0 0 0 0 0 0 1 0 0
custom_pointclouds             string     cloud_unstructured_segments cloud_unstructured_fullframe cloud_polar_unstructured_segments cloud_polar_unstructured_fullframe cloud_all_fields_fullframe
cloud_unstructured_segments    string     coordinateNotation=0 updateMethod=1 echos=0,1,2 layers=6,14 reflectors=0,1 infringed=0,1 rangeFilter=0.05,999,1 topic=/cloud_unstructured_segments frameid=world publish=1
cloud_unstructured_fullframe   string     coordinateNotation=0 updateMethod=0 echos=0,1,2 layers=6,14 reflectors=0,1 infringed=0,1 rangeFilter=0.05,999,1 topic=/cloud_unstructured_fullframe frameid=world publish=1
cloud_polar_unstructured_segments string     coordinateNotation=1 updateMethod=1 echos=0,1,2 layers=6,14 reflectors=0,1 infringed=0,1 rangeFilter=0.05,999,1 topic=/cloud_polar_unstructured_segments frameid=world publish=1
cloud_polar_unstructured_fullframe string     coordinateNotation=1 updateMethod=0 echos=0,1,2 layers=6,14 reflectors=0,1 infringed=0,1 rangeFilter=0.05,999,1 topic=/cloud_polar_unstructured_fullframe frameid=world publish=1
cloud_unstructured_echo1       string     coordinateNotation=0 updateMethod=0 echos=0 layers=6,14 reflectors=0,1 infringed=0,1 rangeFilter=0.05,999,1 topic=/cloud_unstructured_echo1 frameid=world publish=1cloud_unstructured_echo1_segments string     coordinateNotation=0 updateMethod=1 echos=0 layers=6,14 reflectors=0,1 infringed=0,1 rangeFilter=0.05,999,1 topic=/cloud_unstructured_echo1_segments frameid=world publish=1
cloud_unstructured_echo2       string     coordinateNotation=0 updateMethod=0 echos=1 layers=6,14 reflectors=0,1 infringed=0,1 rangeFilter=0.05,999,1 topic=/cloud_unstructured_echo2 frameid=world publish=1cloud_unstructured_echo2_segments string     coordinateNotation=0 updateMethod=1 echos=1 layers=6,14 reflectors=0,1 infringed=0,1 rangeFilter=0.05,999,1 topic=/cloud_unstructured_echo2_segments frameid=world publ [... cut ...]
[Info]: SickScanServices: Request "sMN LMCstartmeas" successfully sent, received reply "sAN LMCstartmeas 0"
[Info]: SickScanServices::sendSopasCmdCheckResponse(): request: "sMN LMCstartmeas", response: "sAN LMCstartmeas 0"
[Info]: SickScanServices: Sending request "sMN Run"
[Info]: Sending  : <STX>sMN Run<ETX>
[Info]: Receiving: <STX>sAN Run 1<ETX>
[Info]: SickScanServices: Request "sMN Run" successfully sent, received reply "sAN Run 1"
[Info]: SickScanServices::sendSopasCmdCheckResponse(): request: "sMN Run", response: "sAN Run 1"
```

rostest, 2023-12-18:
Thanks for following up. Since the last line printed is `sAN Run 1`, sick_scan_xd most likely does not receive any scan data from the lidar via UDP. Please open `//192.168.0.237` (i.e. the ip address of your multiScan lidar) in a browser and use SOPAS Air to check the pointcloud and settings. If you can view the pointcloud in SOPAS Air, please check that the configuration of parameter `udp_receiver_ip:=192.168.0.236` (i.e. the ip address of your PC) is correct and identical to the receiver ip address displayed in SOPAS Air. If SOPAS Air does not display the expected pointcloud, check the UDP network settings in SOPAS Air.

After SOPAS Air displays the expected pointcloud, make sure that parameter `udp_receiver_ip` (i.e. `192.168.0.236` in your logfile) is identical to the configuration in SOPAS Air and retry.

If you still do not receive any pointcloud messages, but you can see the pointcloud in SOPAS Air, please send screenshots from the SOPAS Air settings, such that we can help with the configuration.

treeandcat, 2023-12-19:
Thank you very much for your reply. Every time i open the browser using IP, i can see some cloud data on the browser.When using the program `minimum_sick_scan_api_client.cpp`, the point cloud data can be obtained in a few cases. If the program ends normally, the point cloud data can also be obtained. However, after exiting the program using `ctrl+c`, the point cloud data cannot be obtained in most cases. 
I tried to use `signal(SIGINT, sigint_handler);` to capture the `ctrl+c` signal and perform the release operation in the corresponding sigint_handler function, but I still cannot obtain the point cloud. No port occupation was found using `netstat -ano | findstr`.
[image]
[image]
[image]
[image]

rostest, 2023-12-19:
Thanks for following up. Just to make sure we have the same understanding of the problem: If you start minimum_sick_scan_api_client.cpp for the very first time, you obtain the pointcloud data via the sick_scan_xd API as expected. After Ctrl-C, program exit and restart, you sometimes get a pointcloud, but mostly you do not. Is this correct?

In this case it looks like the multiScan might not reopen/reset the UDP socket after the TCP connection is lost (i.e. after program exit). Please try the following:
1. Restart the lidar (power down and up) and wait until multiScan is ready.
2. Start minimum_sick_scan_api_client and make sure the callback (i.e. function customizedPointCloudMsgCb) is getting called.
3. Stop minimum_sick_scan_api_client (Ctrl-C) and wait until it has exited.
4. Restart the lidar (power down and up) and wait until multiScan is ready.
5. Restart minimum_sick_scan_api_client.
Does minimum_sick_scan_api_client receive a pointcloud now, i.e. does it work if both multiScan and minimum_sick_scan_api_client are restarted? If so, please retry a few times to make sure the behaviour is reproducable.

If this is the case, please run these steps again, but instead of restarting the multiScan lidar, just Disable and Enable the "Measurement Data Output" in SOPAS Air. Is the minimum_sick_scan_api_client working reproducable after restart, if "Measurement Data Output" has been disabled and enabled before? If so, an additional IP reset might be required between two restarts.

Btw: The implementation in function customizedPointCloudMsgCb appends a shallow copy of the received message in dataQueue. This is unsafe, because allocated message memory is released by sick_scan_xd after executing the callback. This might cause dangling pointer somewhere else in the application. Please make and push a deep copy of the message, e.g.:
```
class PointXYZI
{
public:
  float x;
  float y;
  float z;
  float i;
};
// Get offsets for x, y, z, intensity values
SickScanPointFieldMsg* msg_fields_buffer = msg->fields.buffer;
int field_offset_x = -1, field_offset_y = -1, field_offset_z = -1, field_offset_intensity = -1;
for (int n = 0; n < msg->fields.size; n++)
{
  if (strcmp(msg_fields_buffer[n].name, "x") == 0 && msg_fields_buffer[n].datatype == SICK_SCAN_POINTFIELD_DATATYPE_FLOAT32)
    field_offset_x = msg_fields_buffer[n].offset;
  else if (strcmp(msg_fields_buffer[n].name, "y") == 0 && msg_fields_buffer[n].datatype == SICK_SCAN_POINTFIELD_DATATYPE_FLOAT32)
    field_offset_y = msg_fields_buffer[n].offset;
  else if (strcmp(msg_fields_buffer[n].name, "z") == 0 && msg_fields_buffer[n].datatype == SICK_SCAN_POINTFIELD_DATATYPE_FLOAT32)
    field_offset_z = msg_fields_buffer[n].offset;
  else if (strcmp(msg_fields_buffer[n].name, "intensity") == 0 && msg_fields_buffer[n].datatype == SICK_SCAN_POINTFIELD_DATATYPE_FLOAT32)
    field_offset_intensity = msg_fields_buffer[n].offset;
}
assert(field_offset_x >= 0 && field_offset_y >= 0 && field_offset_z >= 0);
// Concatenate all points to a vector
std::vector<PointXYZI> point_field;
point_field.reserve(msg->width* msg->height);
for (int row_idx = 0; row_idx < (int)msg->height; row_idx++)
{
  for (int col_idx = 0; col_idx < (int)msg->width; col_idx++)
  {
    // Get cartesian point coordinates
    PointXYZI point_xyzi;
    int point_offset = row_idx * msg->row_step + col_idx * msg->point_step;
    point_xyzi.x = *((float*)(msg->data.buffer + point_offset + field_offset_x));
    point_xyzi.y = *((float*)(msg->data.buffer + point_offset + field_offset_y));
    point_xyzi.z = *((float*)(msg->data.buffer + point_offset + field_offset_z));
    point_xyzi.i = 0;
    if (field_offset_intensity >= 0)
      point_xyzi.i = *((float*)(msg->data.buffer + point_offset + field_offset_intensity));
    point_field.push_back(point_xyzi);
  }
}
dataQueue.push(point_field);
```

treeandcat, 2023-12-19:
Thank you very much for your patient reply. According to what you said, I changed the copy method of the `dataQueue` in the callback and performed subsequent processing on it in another thread.
Secondly, as you said, when the `sAN Run 1` prompt appears and the data cannot be obtained, close and reopen the radar or `Disable and Enable the "Measurement Data Output" in SOPAS Air`, and then open the program again ,it can obtain the point cloud normally.
When I think about how to write the program so that I don't have to manually close and reopen the radar or `Disable and Enable the "Measurement Data Output" in SOPAS Air` after each ctrl+c interrupt. I found that after modifying the code as follows, even after ctrl+c interrupts the program, the point cloud data can be obtained normally during the next execution.
Thank you again, you really helped me a lot. You are such a kind and patient person, I wish you happy every day `:D`
```cpp
#include <chrono>
#include <iostream>
#include <thread>
#include <signal.h>
#include <mutex>
#include <time.h>
#include <queue>
#ifndef PCL_NO_PRECOMPILE
#define PCL_NO_PRECOMPILE
#endif
#include <pcl/point_types.h>
#include <pcl/point_cloud.h>
#include <pcl/io/pcd_io.h>
#include "sick_scan_xd_api/sick_scan_api.h"
#include <iomanip>

bool app_stopped = false;
SickScanApiHandle apiHandle1;
pcl::PointCloud<pcl::PointXYZI>::Ptr mycloud (new pcl::PointCloud<pcl::PointXYZI>());
time_t begin,end;
std::mutex mylock;
std::condition_variable mycond;
std::queue<pcl::PointCloud<pcl::PointXYZI>> dataQueue;

std::string getCurrentTime()
{
  auto now = std::chrono::system_clock::now();
  std::time_t now_c = std::chrono::system_clock::to_time_t(now);
  std::tm timeInfo = *std::localtime(&now_c);
  std::stringstream ss;
  ss << std::put_time(&timeInfo, "%Y%m%d%H%M%S");
  return ss.str();
}

// Implement a callback to process pointcloud messages
void customizedPointCloudMsgCb(SickScanApiHandle apiHandle, const SickScanPointCloudMsg* msg)
{
  if (app_stopped)
  {
      return ;
  } 
  std::cout<<"++++++++++++++++++++++++++++++++++++++++++++++++"<<std::endl;
  SickScanPointFieldMsg* msg_fields_buffer = msg->fields.buffer;
  int field_offset_x = -1, field_offset_y = -1, field_offset_z = -1, field_offset_intensity = -1;
  for (int n = 0; n < msg->fields.size; n++)
  {
    if (strcmp(msg_fields_buffer[n].name, "x") == 0 && msg_fields_buffer[n].datatype == SICK_SCAN_POINTFIELD_DATATYPE_FLOAT32)
      field_offset_x = msg_fields_buffer[n].offset;
    else if (strcmp(msg_fields_buffer[n].name, "y") == 0 && msg_fields_buffer[n].datatype == SICK_SCAN_POINTFIELD_DATATYPE_FLOAT32)
      field_offset_y = msg_fields_buffer[n].offset;
    else if (strcmp(msg_fields_buffer[n].name, "z") == 0 && msg_fields_buffer[n].datatype == SICK_SCAN_POINTFIELD_DATATYPE_FLOAT32)
      field_offset_z = msg_fields_buffer[n].offset;
    else if (strcmp(msg_fields_buffer[n].name, "intensity") == 0 && msg_fields_buffer[n].datatype == SICK_SCAN_POINTFIELD_DATATYPE_FLOAT32)
      field_offset_intensity = msg_fields_buffer[n].offset;
  }
  assert(field_offset_x >= 0 && field_offset_y >= 0 && field_offset_z >= 0);
  std::unique_lock<std::mutex> lockGuard(mylock);
  pcl::PointCloud<pcl::PointXYZI> point_field;
  point_field.reserve(msg->width* msg->height);
  for (int row_idx = 0; row_idx < (int)msg->height; row_idx++)
  {
    for (int col_idx = 0; col_idx < (int)msg->width; col_idx++)
    {
      pcl::PointXYZI point_xyzi;
      int point_offset = row_idx * msg->row_step + col_idx * msg->point_step;
[... 54 lines cut ...]
	const char* sick_scan_api_lib = "libsick_scan_xd_shared_lib.so";
#endif
  signal(SIGINT, sigint_handler);
  SickScanApiLoadLibrary(sick_scan_api_lib);
  int argcs=4;
  char* args[] = { "E:/work/jyc/code/sick_scan_ws/build/Release/minimum_sick_scan_api_client.exe", "E:/work/jyc/code/sick_scan_ws/sick_scan_xd/launch/sick_multiscan.launch", "hostname:=192.168.0.237", "udp_receiver_ip:=192.168.0.236"};
  SickScanApiHandle apiHandle = SickScanApiCreate(argcs, args);
  SickScanApiInitByCli(apiHandle, argcs, args);
  apiHandle1=apiHandle;
  std::thread t1(fun);
  SickScanApiRegisterCartesianPointCloudMsg(apiHandle, &customizedPointCloudMsgCb);
  // t1.detach();
  t1.join();
  // std::this_thread::sleep_for(std::chrono::seconds(15));
  // Close lidar and release sick_scan api
  SickScanApiDeregisterCartesianPointCloudMsg(apiHandle, &customizedPointCloudMsgCb);
  SickScanApiClose(apiHandle);
  SickScanApiRelease(apiHandle);
  SickScanApiUnloadLibrary();
}
catch(const std::exception& e) {
    std::cerr << "Exception caught: " << e.what() << std::endl;
}
}
```

rostest, 2023-12-19:
@treeandcat You are welcome! Thank you very much for your kind feedback.

closed 2023-12-19 by rostest