SICKAG/sick_scan_xd #247: Hardcoded/wrong implementation of the scan topic remaps from 3.0.0 onwards

RBinnenmars, 2023-12-05:
From version 3.0.0 onwards, the topic that the laser scan is published changed to the /scan topic. 
For example, I use the `sick_nav_350` lidar, previously this was "/sick_nav_350/scan" (the other topic still has the namespace) but this is hardcoded to the "/scan" topic. This should be done with a launch argument to be remapped to a different topic if wanted, now the topic can only be remapped by also including the namespace (/sick_nav_350/scan:=/wanted_topic).

Also update your package.xml to use the correct version numbers

rostest, 2023-12-11:
Thanks for your feedback. We will add a remap option for ros laserscan messages in the launchfile in the next release. Please use ros option `<remap from="/sick_nav_350/scan" to="/wanted_topic"/>` in the launchfile for remapping laserscan messages.

The version number in package.xml will be updated.

michael1309, 2023-12-13:
# Arbeitspakete sick_scan_xd „Erster Durchstich Automatisiertes Testen“
  * AP1: Docker-Container für automatische V&amp;V-Integrationstests
  * AP2: Erweiterung der Mockup-Testumgebung („Emulator“) für vollautomatische Tests
## AP1: Docker-Container für automatische V&amp;V-Integrationstests
  * Vorbereitung in Abstimmung mit SICK-V&amp;V-Team eines Docker-Containers mit ROS1-Noetic-
Linux-Umgebung, der auf V&amp;V-DIMA-Testrechnern lauffähig ist.
  * Implementierung von Skripts:
    1. zur automatischen Installation von vorcompilierten sick_scan_xd Binaries aus externen
Quellen (zukünftig CI-Pipeline auf gitlab.sick.com) in der Container-Umgebung
    2. zur Ausführung der Mockup-Testumgebung (siehe AP2)
    3. zur Rückgabe der Testberichte und der Statusmeldung der Tests (Pass/Fail)
entsprechend des V&amp;V-Test-Workflows

## AP2: Erweiterung der Mockup-Testumgebung für automatisierte Testberichte
  * Umstellung der aktuellen Mockup-Testumgebung (auch „Emulator“ genannt) von manuellem
Abgleich der ROS-Ausgabedaten auf automatischen Abgleich der ROS-Ausgabedaten
ausschließlich für multiScan-Implementierung unter ROS1-Noetic-Linux in der oben
genannten Container-Umgebung, mit folgenden Tests:
    * Erfolgreiche Parametrierung und Start des Treibers für Compact-Datenausgabe
    * Empfang von Compact-Testdaten aus pcapng-Datei und Ausgabe als ROS
PointCloud2-Daten
    * Stichprobenartige Prüfung von Point-Cloud-Daten auf numerische Korrektheit
- Implementierung der Generierung von Testberichten und Docker-Rückgabewert entsprechend
V&amp;V-Konventionen mit Testergebnis der verschiedenen Einzeltests (Pass/Fail)

Anmerkung Michael: 
* V&amp;V: Interne Abteilung bei SICK für Produkttests
* Unterlagen für  V&amp;V-DIMA-Testrechner liegen (noch) nicht vor

rostest, 2023-12-18:
The ros topic for laserscan messages can now be configured directly by using the new branch [feature/laserscan_topic](https://github.com/SICKAG/sick_scan_xd/tree/feature/laserscan_topic). Use commandline argument `laserscan_topic:=<topic_name>` or launchfile parameter `laserscan_topic` to overwrite the default laserscan topic.

fmessmer, 2023-12-18:
@rostest when will this feature be included in the main branch?
it's still only implemented in a feature branch, right?
it would be easier for the community to follow the changes of the upstream repo if you would also provide PullRequest from the feature branches to the main branches....otherwise it is hard to see when changes become available because you do not get notifications for commits being pushed

I think the issue should stay open until the patch is included in the main branch!

rostest, 2023-12-18:
@fmessmer Thanks for your feedback. This feature is currently only available in branch feature/laserscan_topic. It will be included in the main branch with the next release. We will leave this issue open until then. 
As an alternative to the new laserscan_topic parameter, you can use ros option `<remap from="/sick_nav_350/scan" to="/wanted_topic"/>` in the launchfile for remapping laserscan messages.

fmessmer, 2024-01-18:
what is blocking to merge the feature branch to the main branch?
it is completely backwards-compatible for those that don't want to use it.
thus, it could sit on the main branch being merged while waiting for the next main branch release.

rostest, 2024-01-18:
@fmessmer Thanks for your feedback. We will clarify it with SICK.

fmessmer, 2024-01-26:
I just wanted to create a PullRequest for the mentioned feature branch to increase visibility of getting this merged.
But the creation failed because I'm not a collaborator.

I must say: very strange policy for an open-source repo to not accept contributions from the community.

rostest, 2024-01-26:
@fmessmer Thanks for your feedback. Please contact SICK support for policy requests.

rostest, 2024-01-26:
@fmessmer Pull request #264 is created.

aiplemaSICKAG, 2024-02-01:
@fmessmer We welcome contributions from the community and regularly merge contributions from pull requests. Please use the conventional github process for pull requests consisting of creating a fork of the sick_scan_xd repository, applying changes to a branch in that repository and creating a pull request (see https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request-from-a-fork ). 
Feel free to contact us should you still experience problems when following that process.

fmessmer, 2024-02-19:
Now that the feature/PR made it into `v3.2.0`, do you also plan to trigger a `bloom-release` for this version on the ROS BuildFarm?

rostest, 2024-02-20:
Bloom release for version 3.2.0 is in progress. It has been successfully built on the ros build farm a few days ago (https://build.ros.org/job/Ndev__sick_scan_xd__ubuntu_focal_amd64/4/, git commit https://github.com/SICKAG/sick_scan_xd/commit/abd456030afbdbdc397e17450a99f935014923d2) and should soon be available via apt.

closed 2024-07-08 by rostest