SICKAG/sick_scan_xd #251: can't compile SICK_GENGERIC_CALLER withour ROS in linux gcc4.8

sjeffreydong, 2023-12-09:
[image]

rostest, 2023-12-11:
Thanks for your feedback. gcc 4.8 has been released december 2014 and is no longer supported. Please update gcc and linux. If you  use Ubuntu, please upgrade to Ubuntu 20.04 or newer.

closed 2023-12-11 by rostest

sjeffreydong, 2023-12-11:
Which version can be run on our "Build on Linux generic without ROS"

rostest, 2023-12-11:
sick_scan_xd requires C++ 14, which is supported by gcc 6.1 and newer according to https://stackoverflow.com/questions/47282939/since-which-version-of-gcc-is-c14-supported. We recommend to upgrade to Ubuntu 20.04 with gcc version 9 or newer.