SICKAG/sick_scan_xd #254: 关于 include/sick_scan_xd_api/sick_scan_api.h 中的第142行

BroGua, 2024-01-04:
SickScanPointFieldArray 表明了 field 是一个数组，但是field后面并没有相应的方括号

rostest, 2024-01-04:
Thanks for your message. Can you give some more details and explanations about the problem? 

Google translate suggests: 
> SickScanPointFieldArray indicates that the field is an array, but there is no corresponding square bracket after the field

Note that SickScanPointFieldArray is a plain C struct including a pointer to a data buffer. Please give some more information about your compiler, compiler version and OS, if struct SickScanPointFieldArray causes compiler warnings or errors.

BroGua, 2024-01-05:
当我使用SickScanPointFieldArray 中的 buffer， 只能得到 name  为 x 的数据。那么 y ，z ，intensity 这些数据我该如何拿到呢？
[image]

rostest, 2024-01-05:
Google translates to
> When I use the buffer in the SickScanPointFieldArray, I only get data with name x. So how do I get the data of y, z, and intensity?

The size in struct SickScanPointFieldArray gives you the number of fields:
```
typedef struct SickScanPointFieldArrayType // Array of SickScanPointFieldMsg, which can be serialized and imported in C, C++ or python
{
  uint64_t capacity; // Number of allocated elements, i.e. max. number of elements in buffer, allocated buffer size is capacity * sizeof(SickScanPointFieldMsg)
  uint64_t size;     // Number of currently used elements in the buffer
  SickScanPointFieldMsg* buffer;  // Memory, data in plain order and system endianess (buffer == 0, if size == 0 && capacity == 0, otherwise allocated memory), allocation/deallocation always managed by the caller.
} SickScanPointFieldArray;
```

In this case the fields.size is 4, i.e. fields.buffer contains 4 fields (x, y, z, and intensity), where each field is a SickScanPointFieldMsg. Use the following example to access these fields:
```
// Example callback for cartesian pointcloud messages
static void apiTestCartesianPointCloudMsgCallback(SickScanApiHandle apiHandle, const SickScanPointCloudMsg* msg)
{
    // Get field description for x, y, z, intensity
    SickScanPointFieldMsg* msg_fields_buffer = msg->fields.buffer;
    for(int n = 0; n < msg->fields.size; n++)
    {
        std::cout << std::string(msg_fields_buffer[n].name) << ", " << (int)(msg_fields_buffer[n].datatype) << ", " << (int)(msg_fields_buffer[n].offset) << ", " << (int)(msg_fields_buffer[n].count) << std::endl;
    }
}
```

A complete C/C++ usage example is implemented in https://github.com/SICKAG/sick_scan_xd/blob/master/test/src/sick_scan_xd_api/sick_scan_xd_api_test.cpp, which you can use as a starting point for a customized pointcloud handling. See https://github.com/SICKAG/sick_scan_xd/blob/master/doc/sick_scan_api/sick_scan_api.md#complete-usage-example-in-c for further details.

BroGua, 2024-01-05:
OK，我理解了

closed 2024-02-01 by rostest