SICKAG/sick_scan_xd #256: Better PicoScan lidars support

paulinelvne, 2024-01-08:
We are in the process of integrating the new PicoScan sensors through this library and we did encounter a few issues.

### Performance Profile parameter

We had to modify the library to be able to change the "performance profile" parameter of the lidar. We only partially succeeded as we are only able to change to profiles from 0 to 9, and not to 10. See the pull request we did [here](https://github.com/SICKAG/sick_scan_xd/pull/255).

### Example crashing when deregistering the library

The picoScan cannot stop properly because an issue occurs when calling one of the following 3 functions at the end of the sick_scan_xd_api_test example :
- SickScanApiClose
- SickScanApiRelease
- SickScanApiUnloadLibrary

For now, we haven't found a way to fix this issue.

rostest, 2024-01-08:
Thanks for your feedback. We will discuss support of Profiling Systems with SICK.
For a clean API exit by SickScanApiClose, please checkout branch https://github.com/SICKAG/sick_scan_xd/tree/feature/api_exit and rebuild.

paulinelvne, 2024-01-15:
We still have a problem with the SickScanApiClose function. When it's called in the sick_scan_xd_api_test example, the function is never exited, so the rest, such as SickScanApiRelease, is never called (tested with breakpoints). 

We've noticed that the s_scanner variable is not allocated for picoscans (in sick_generic_laser.cpp file at lines 437-439) and when stopScannerAndExit(bool force_immediate_shutdown) is called in sick_generix_laser.cpp, it never goes through s_scanner->stopScanData(force_immediate_shutdown). 
We then get the impression that we're stuck at the _Thrd_join(_Thr, nullptr) level in the thread file. This may be because the scanner never stops due to s_scanner = nullptr. 

Any ideas? 
Thank you in advance for your time.

rostest, 2024-01-16:
Many thanks for following up! The previous fix was incomplete. Please use the updated version in branch https://github.com/SICKAG/sick_scan_xd/tree/feature/api_exit and rebuild. Note that this version is collected update with several changes incl. cleanup at API exit.

paulinelvne, 2024-01-17:
The problem under Windows for picoScan doesn't seem to have been solved. 
We've noticed that when calling udp_receiver->Close() (line 395 of the scansegment_thread.cpp file), the function gets stuck at m_receiver_thread->join() (line 169 of the udp_receiver file). In fact, the thread doesn't seem to stop and remains blocked on a data reception operation (line 200 in the udp_socket.h file), as the data reception cannot take place because the connection is breaking. 

Thank you for taking the time to answer our questions.

rostest, 2024-01-18:
@paulinelvne Many thanks for your feedback and the detailed error description! Please find an update in branch https://github.com/SICKAG/sick_scan_xd/tree/feature/api_exit, which corrects closing receiver threads at API exit.

closed 2024-01-23 by paulinelvne