SICKAG/sick_scan_xd #260: SickScanApi-Functions error handling

cblesing, 2024-01-17:
Hi everyone,

according to the comments in the header file, there shoul be a possibility to check if the api functions return successfully or not. There is an enum called `SickApiErrorCodes` which should be used to cover that behavior. 

In our application we use the `SickScanApiInitByCli`-Method and the `SickScanApiRegisterCartesianPointCloudMsg`-Method to get the scanner up. But even if I disconnect the scanner (LMS1000) in a physical way the first named Method allways returned 0 which means `SICK_SCAN_API_SUCCESS`. The fact that the callback method could be registered successfully is fine, because the callback is there. 

My question is how to do this right, so that we receive error codes in case that the scanner is not physicaly connected? 

Best regards,
Christian

rostest, 2024-01-17:
Thanks for your feedback. SickScanApiInitByCli starts the lidar initialization, including background threads to handle TCP communication, lidar setup and reconnection after connection failures. In case of SICK_SCAN_API_SUCCESS, this process has started successfully.

Currently the API does not provide functions to query the current lidar state resp. to get error messages and error codes in case of failures. We will provide these functions in a future release.

To detect errors of any kind, you can monitor the receipt of pointcloud messages within your application.

rostest, 2024-01-23:
@cblesing Please find an update in branch https://github.com/SICKAG/sick_scan_xd/tree/feature/api_diagnostics, which exports diagnostic status and messages via the API. You can now register callbacks using SickScanApiRegisterDiagnosticMsg and SickScanApiRegisterLogMsg, or query the current status with function SickScanApiGetStatus. A callback function registered with SickScanApiRegisterDiagnosticMsg will be called, whenever the status changed (INIT, EXIT or OK) or an error occured (WARN or ERROR). See the [diagnostic API description](https://github.com/SICKAG/sick_scan_xd/blob/feature/api_diagnostics/doc/sick_scan_api/sick_scan_api.md#diagnostic) and example [sick_scan_xd_api_test.cpp](https://github.com/SICKAG/sick_scan_xd/blob/feature/api_diagnostics/test/src/sick_scan_xd_api/sick_scan_xd_api_test.cpp) for details.  
Note that this update is a cumulative update with multiple changes.

closed 2024-04-08 by rostest