SICKAG/sick_scan_xd #261: There is not data in topic with linux ROS2 & picoScan

Jonny0816, 2024-01-22:
Dear Sir
we build on linux ROS2 refer to the flow and without any error. 
Seems have data return after launch.
[image]
But not data under RVIZ and echo command. Please help to give some comments what is the reason.
[image]

Thanks

Dynamitans, 2024-01-25:
Hello, my friend. 
Here are my basic configurations。
ubuntu22.04+ROS2_humble+nav2+picoscan150
Regarding the ID of the first laserscan message being "world_1," it is because the program categorizes the radar information source. If you use the default settings, the frame_id is indeed "world." We want to change this ID to a custom name, for example, changing it to "map" results in "map_1."

As for the later issue, I am using the master branch from the source code. We can view radar point cloud information on the webpage. After connecting the radar to our Ubuntu, I only changed the radar's IP and receiver's IP. After compiling and running, in the rviz interface, I can see topic messages, but there is no radar image.

Dynamitans, 2024-01-25:
> Thanks for your feedback and screenshots. Both pointcloud and laserscan messages should be published and visible with rviz and ros2 topic. Note that the frame id of pointcloud messages is "world" and the frame_id of laserscan messages is "world_1" by default.
> 
> Do you use the latest sick_scan_xd release 3.1? Can you see any pointclouds in rviz2 resp. ros2 topic, e.g. on topic "cloud_unstructured_fullframe"?

Hello, my friends. 
Here are my basic configurations.
ubuntu22.04+ROS2_humble+nav2+picoscan150
Regarding the ID of the first laserscan message being "world_1," it is because the program categorizes the radar information source. If you use the default settings, the frame_id is indeed "world." We want to change this ID to a custom name, for example, changing it to "map" results in "map_1."

As for the later issue, I am using the master branch from the source code. We can view radar point cloud information on the webpage. After connecting the radar to our Ubuntu, I only changed the radar's IP and receiver's IP. After compiling and running, in the rviz interface, I can see topic messages, but there is no radar image.

rostest, 2024-01-25:
Thanks for following up. The messages "[sick_scan_xd]: MsgPack/Compact-Exporter: ... messages exported ..." in your screenshot mean that sick_scan_xd receives scan data and publishes ros pointcloud and laserscan messages. These ros messages should be visualized with rviz2 and ros2 topic.

Some example screenshots from Ubuntu 22.04 + ROS2 humble are attached below. Please check your rviz2 settings (topics and frame ids) against the example.

rviz2 with pointcloud messages on topic "cloud_unstructured_fullframe", frame id "world":
[image]

rviz2 with laserscan messages on topic "scan_fullframe", frame id "world_1_0":
[image]

"ros2 topic echo /scan_fullframe":
[image]

saran000123, 2024-01-25:
> world_1
We are also facing the same issue and also requested support from SICK INDIA through Mail
update us with a feasible solution

rostest, 2024-01-25:
@saran000123 If the issue is the configuration of the laserscan topic, there is an update available in branch https://github.com/SICKAG/sick_scan_xd/tree/feature/laserscan_topic. This update adds an option to configure the laserscan topic in the launchfile. See https://github.com/SICKAG/sick_scan_xd/issues/247 for details.

Dynamitans, 2024-01-26:
now,I will detail the entire process I've undergone, using the current github-master section and the Piscoscan150 laser scanner.

After downloading, I changed the IP section as shown in the figure below:
[image]

Compile using the following commands:

1. colcon build --packages-select sick_scan_xd --cmake-args " -DROS_VERSION=2" " -DLDMRS=0" " -DSCANSEGMENT_XD=1" --event-handlers console_direct+

2. source '/home/dyna/sick_scan_ws/install/setup.bash'

3. ros2 launch '/home/dyna/sick_scan_ws/src/sick_scan_xd/launch/sick_picoscan.launch.py'

The results of the run are as shown in the figure below:
[image]
[image]

rostest, 2024-01-26:
@Dynamitans Thanks for details and following up! Your setup looks correct. I can not reproduce the problem, but I do not see any obvious reasons why sick_scan_xd does not publish the expected ros messages. Does sick_scan_xd print any errors or warnings after startup during initialization? Can you restart and send a complete logfile of the console output until "MsgPack/Compact-Exporter" reports about exporting messages?

saran000123, 2024-01-27:
> @saran000123 If the issue is the configuration of the laserscan topic, there is an update available in branch https://github.com/SICKAG/sick_scan_xd/tree/feature/laserscan_topic. This update adds an option to configure the laserscan topic in the launchfile. See #247 for details.

we got multiple frame ID with inconsistent data
we are facing this issue in picoScan150 Core w/o Add On's this model only. 
In picoScan150 Core-1 works fine without this issue

Dynamitans, 2024-01-29:
[sick.log](https://github.com/SICKAG/sick_scan_xd/files/14079477/sick.log)

rostest, 2024-01-29:
@Dynamitans Thanks for the logfile. According to this log, sick_scan_xd tries to connect the lidars ip address 192.168.0.60:
`[sick_generic_caller-1] [INFO] [1706500640.404749584] [sick_scan_xd]: sick_scan_xd: Tcp::open: connecting to 192.168.0.60:2111 ...`

But sick_scan_xd never succeeds, i.e. the driver can not establish a tcp connection to the lidar. The lidar is never initialized, neither is the thread to publish the expected ros messages. That is probably the reason for not publishing pointcloud or laserscan messages.

Nevertheless and unexpectedly, the lidar seems to send udp scan data, which the driver receives. Since the lidar seems to send udp scan data, it should be reachable for tcp connections, too. Therefore, please run the following steps:

1. Kill any sick_scan_xd related processes. Make sure, that neither sick_generic_caller nor any sick_scan_xd proceses are running. If in doubt, restart the system.

2. Restart the lidar (power down and power up). If more than one lidar are connected, make sure that only one lidar is powered up.

3. Check the network connection to the lidar: `ping 192.168.0.60` must be sucessful (assuming 192.168.0.60 is the ip address of your picoScan).

4. Open `//192.168.0.60` in a browser and use SOPAS Air to check the lidar status. If everything works ok, you should see the picoScan pointcloud in SOPAS Air.

5. Turn off measurement mode in SOPAS Air.

6. Restart sick_scan_xd with arguments `hostname:=192.168.0.60` and `udp_receiver_ip:=192.168.0.39`. Make sure that parameter hostname and udp_receiver_ip are identical to the ip addresses shown in SOPAS Air.

7. After sick_scan_xd displays the message `sick_scan_xd: Tcp::open: connecting to 192.168.0.60:2111 ...`, the lidar should be initialized. After succesful initialization, pointcloud and laserscan messages are published. If not, please send a logfile of the new run.

closed 2024-04-08 by rostest